realsenseai/librealsense #9095: Synchronize LiDAR L515 camera with other professional RGB cameras?

DragonDeux, 2021-05-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentation](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is your question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open a new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | Any | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

Was it possible to synchronize the acquisition using the L515 camera with other professional RGB cameras?

sam598, 2021-05-26:
Are you looking to have hardware sync (both cameras are captured at the exact same time) or software sync (know what time the L515 depth image and external RGB image we captured relative to eachother)?

If you are looking for hardware sync, that will likely be difficult with the L515. My understanding is that because of the way it captures depth, the framerate has to be at a very constant rate. It cannot be arbitrarily triggered like some of the D400 cameras can be. This means that the external camera would need to be slaved to the L515, which is not a trivial task either, and depends on the camera.

What kind of professional RGB camera were you planning on using? Some ethernet cameras can be triggered by an arbitrary signal. Other cameras (like genlock film and video cameras) synchronize their clock signals, but do not arbitrarily trigger frame captures.

If you only need software sync, there may be a way to use the sync signal out of the L515 to get a reference time. While the RealSense SDK provides timestamps, you will likely be using it on a non-realtime operating system, which means that you cannot rely on the timestamp to accurately measure a real moment in time. You would need external hardware, like a microcontroller to record the sync signal from the L515, and also record the sync signal from an external camera.

DragonDeux, 2021-05-26:
Hi @sam598, 
Thanks for your reply. 
Im plan to use 2 additional BlackMagic camera plug into a decklink frame grabber. These cameras could be sync by timecode. For software sync alike, was that to sync L515 and Blackmagic camera by using timecode? im also think about to use ROS to synchronize them (L515 and cameras). 

If i well understood your comments, the reliable sync is to trigger L515 and other RGB cameras by Hardware approach, right?

sam598, 2021-05-26:
Let me rephrase my question a bit, and preface it with some information about how video cameras handle synchronization (RealSense cameras handle sync and time stamps slightly differently).

Time code does not actually sync cameras together. It does give each frame a reference time stamp to a shared common clock. But those cameras might be running at slightly different frame rates. Or be running in sync, but be offset by half a frame.

In most cases you probably will not notice the difference, but if you have a fast moving subject it may be misaligned between the different cameras.

Some film and video cameras use a genlock signal in order to precisely synchronize their internal clocks so that every single frame is taken at the exact same time.

In your use case do you just want a time code/timestamp for the L515 camera?

Or do you want the L515 to capture depth at the exact same time as the RGB cameras do?

jieunko, 2021-05-27:
Hello I am doing quite same thing with @DragonDeux however I want use two Azure Kinect instead BlackMagic which means they have their own synchronization method. How should I sync these with L515?? I am trying to use both RGBs and Depth.

DragonDeux, 2021-05-27:
Hi @sam598 

My acquisition scenario is to take dynamic acquisition with L515 and blackmagic RGB cameras(or maybe other brands).depth map should be (ideally) sync prefectly with other RGB cameras. Acquisition should be as long as possible (synchronized)

Actually, decklink (blackmagic cameras) has its own strategy to sync themselves (up to 4 cameras).

The depth maps captured by L515 could is admired, but the quality of color sensor is not conform exactly our requirements. I can images that there are lots similaire demandes about using L515 depth sensor with professional RGB camera scenario have been raising at this time. 

Really appreciate if anyone can give some intuition or solutions.

sam598, 2021-05-27:
@DragonDeux 

This is going to get a bit off topic from RealSense, and you may already know all of this, but I think it might be useful information.

For any camera to get the kind of syncronization you are looking for (where every camera captures a frame at the exact same time) the cameras need to receive a syncronization signal. For film and video cameras this is called a genlock signal. This is a special video signal (either "black-burst or tri-level" format) that tells the camera when it should be capturing frames.

Only some cameras are capable of receiving this signal. For the blackmagic line of cameras, the micro, studio, and ursa cameras can all take black-burst or tri-level genlock signals. The pocket cinema cameras cannot receive a genlock signal. They are meant to be used as a single standalone camera, and there is no way to get precise syncronization between these cameras.

Some blackmagic decklink cards can provide a genlock signal out as a separate cable that needs to go to each camera. But if that signal is not going into the cameras, then they will not be capturing frames at the same time. The card may be aligning the incoming frames, but I do not think that is the kind of syncronization that you are looking for.

Now onto the L515. 

Looking at the multicamera whitepaper.... this isn't really multicamera syncronization.
https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

Apparently you cannot send a signal for each frame to multiple L515 cameras and have them capture at the same time. What you can do is send a signal that starts and stops the camera at a specific time. But the camera needs about 60ms before it can start capturing frames, and then captures a frame every 33ms until it is stopped.

This means if you wanted to have a fully manual sync signal, the best you could do is about 10 frames per second.

But, if you know when your genlocked RGB cameras are going to take a frame, you could time the start of the L515 to match your genlocked cameras.

This dev board can connect to a microcontroller like an Arduino, and give you precise timing on an incoming genlock signal.
https://shieldrock.com/proto.html

You would then program the microcontroller to start the L515 in time with the same genlock signal that is going to your other cameras. This will take a bit of experimentation and trial-and-error.

Also even if you do all of that, there is no guarantee that the cameras will run in sync. All you are doing is having the cameras start at the same time. If the L515 happens to be running at 30.01hz while the RGB cameras are running at 29.97hz, they will drift out of sync over time. I don't know the exact frame rate for the L515, but different D400 cameras run somewhere between 29.9 and 30.1hz.

sam598, 2021-05-27:
@jieunko Unfortunately I think you are going to have a lot of trouble getting that to work, for multiple reasons.

My understanding of how the Azure Kinect handles synchronization is that when they are "synced" it actually offsets the capture time of each depth camera. This is because it is a time of flight camera, and multiple time of flight cameras interfere with eachother.

In the L515 multi camera example Intel actually only has one camera running at a time, because they interfere with eachother. The Kinects are able to run slightly offset within their 33ms capture window. It's not clear if the L515 is able to do that precise of an offset.

If you do want to try it anyway, it looks like the Kinect can take an external synchronization signal.
https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync

My recommendation would be to have a microcontroller that triggers the L515, waits 60ms, then starts sending a signal to the Kinect every 33ms. You will probably need to do a lot of experimentation and refinement to reduce interference, and make sure that the cameras are running at the exact same framerate. Unfortunately there does not seem to be a way to get a direct reference signal out of the L515.

sam598, 2021-05-27:
@DragonDeux A few more notes about genlock.

Some cameras start their exposure right when the black-burst/tri-level sync happens.

Other cameras only use the genlock signal as a reference signal, that the internal hardware and software synchronizes with.

If you have two identical camera models, they will interpret the signal the exact same way and capture frames at the exact same time. So two Blackmagic Micro cameras would run perfectly in sync.

If you have two different camera models or from different manufacturers, they may run at the same rate, but the exposures may be out of phase. So a Blackmagic Micro and a Panasonic BGH1 could have each frame offset by 16ms. 

In most cases the manufacturer won't provide this information, but you can test this with a flash strobe. 

Finally, the D435 and D455 cameras have an experimental "genlock" mode. This is not the same thing as video camera genlock. I would argue it is actually far more useful.

Instead of aligning it's internal clock with an external signal, the camera will capture a frame whenever it receives a sync pulse (as long as it has enough time to process the previous frame). This lets the depth cameras be slaved to any kind of external trigger at arbitrary rates (as long as the pulse is right voltage and duration). Unfortunately the L515 is a different kind of depth camera and does no have this mode.

DragonDeux, 2021-05-27:
Thanks a lot, @sam598, 

So this sync puzzle would be divided into 2 tasks :)

1) As @sam598 well explained above, it is possible to hardware-ly trigger all cameras to start the acquisition at the same time. I have 4 exact same BlackMagic cameras and one LiDAR.
2) Somehow to guarantee all cameras synchronized run during the capture process? Was that possible to use ->
**_ROS: message_filters::TimeSynchronizer_** to software-ly sync these multi-cameras? 

So, (1) hardware triggers start-up of acquisition, then (2) use invoke ROS to synchronize the acquisition system at the software level. was this feasible theoretically?

Regards

sam598, 2021-05-27:
@DragonDeux 

Pretty much.

I can't speak for ROS, but I know that the RealSense SDK has a "syncer" class. In RealSense this tries to align/group frames together based on their sensor timestamps.

Just to be absolutely clear, this class will **ALIGN** frames based on when they **WERE** taken, it does not **SYNCHRONIZE** when they **ARE** taken.

This also only cover RealSense cameras. From the software side I would not know where to start getting ROS and Decklink to work together. I'm sure it's possible, it would just take some figuring out.

jieunko, 2021-05-30:
@sam598  Thank you for wonderful advices. This would be real help.

DragonDeux, 2021-06-08:
Thank you :) @sam598

closed 2021-11-22 by RealSenseCustomerSupport