realsenseai/librealsense #9097: L515: Confusion over reported color camera distortion model (Brown-Conrady) vs. previous versions.

rafaelspring, 2021-05-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L500 | 
| Firmware Version                   | 01.05.04.01 | 
| SDK Version                            |  2.45.0  | 

### Issue Description
The SDK 2.45.0 had some changes w.r.t. Brown-Conrady distortion model (#8759, #8791 and possibly others).

We also noticed that our L515 no longer reports `RS2_DISTORTION_INVERSE_BROWN_CONRADY` as its color camera distortion model but instead `RS2_DISTORTION_BROWN_CONRADY`.

This is a bit confusing. We haven't logged / compared the reported distortion coefficients across different SDK versions. **Do the reported coefficients now have to be interpreted differently** (i.e. has the distortion model *actually* changed), or is libRS just using a different model constant now to describe the same dist. coefficients?
For context: we aren't using the project() / deproject() methods that libRS provides but instead are using our own camera modeling.

rafaelspring, 2021-05-28:
@ev-mp @RealSense-Customer-Engineering @dorodnic

Anyone?

RealSenseSupport, 2021-06-15:
@rafaelspring 

In the previous releases we were reporting the incorrect model and that needed to be fixed which now is Brown-Conrady instead of Inverse-Brown-Conrady.  On top of that, the use of this information in our code with regards to proj and deproj needed to be updated properly.  Do test but I think if you're not using the LRS proj or deproj and using your own camera modeling, there should be no problems with that.

MartyG-RealSense, 2021-09-11:
Hi @rafaelspring  Do you require further assistance with this case, please?  Thanks!

rafaelspring, 2021-09-11:
@MartyG-RealSense I'm good thanks. Feel free to close this ticket.

MartyG-RealSense, 2021-09-11:
Thanks so much, @rafaelspring

closed 2021-09-11 by MartyG-RealSense