realsenseai/librealsense #9098: Dear @RealSense-Customer-Engineering

akshayacharya97, 2021-05-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | / NVIDIA Jetson.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Dear @RealSense-Customer-Engineering 
I am using Realsense L 515. I want to calculate the dimensions of my object by creating a bounding box adn then calculating the real world point of the two extremes of the bounding box and using distance formula. However, when i pass one point to deproject_pixel_to_point it is returning the x y and z values as 0.

```
import darknet
import cv2
import numpy as np
import pyrealsense2 as rs

"""##############. Function definitions. ##################"""

#Define the detection function
def image_detection(image, network, class_names, class_colors, thresh):
    # Darknet doesn't accept numpy images.
    # Create one with image we reuse for each detect
    width = darknet.network_width(network)
    height = darknet.network_height(network)
    darknet_image = darknet.make_image(width, height, 3)

    #image = cv2.imread(image_path)
    image_rgb = cv2.cvtColor(image, cv2.COLOR_BGR2RGB)
    image_resized = cv2.resize(image_rgb, (width, height),interpolation=cv2.INTER_LINEAR)

    darknet.copy_image_from_bytes(darknet_image, image_resized.tobytes())
    detections = darknet.detect_image(network, class_names, darknet_image, thresh=thresh)
    darknet.free_image(darknet_image)
    image = darknet.draw_boxes(detections, image_resized, class_colors)
    return cv2.cvtColor(image, cv2.COLOR_BGR2RGB), detections

# Initialize and declare the neural network along with data files, config files etc 
quantity_apples = []
config_file = "/home/jetson/Desktop/pano_l515/yolov4.cfg"
data_file = "/home/jetson/Desktop/pano_l515/coco.data"
weights = "/home/jetson/Desktop/pano_l515/yolov4.weights"

network, class_names, class_colors = darknet.load_network(
        config_file,
        data_file,
        weights,
        batch_size=1
    )

# Create a pipeline
pipeline = rs.pipeline()

# Create a config and configure the pipeline to stream
#  different resolutions of color and depth streams
config = rs.config()

# Get device product line for setting a supporting resolution
pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)

# Start streaming
profile = pipeline.start(config)

##Alignment begins here
# Getting the depth sensor's depth scale (see rs-align example for explanation)
depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)
# Create an align object
# rs.align allows us to perform alignment of depth frames to others frames
# The "align_to" is the stream type to which we plan to align depth frames.
align_to = rs.stream.color
align = rs.align(align_to)

# Streaming loop
try:
    for i in range(0,2):
        # Get frameset of color and depth
        frames = pipeline.wait_for_frames()
        # frames.get_depth_frame() is a 640x360 depth image

        # Align the depth frame to color frame
        aligned_frames = align.process(frames)

        # Get aligned frames
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        depth_intrin = aligned_depth_frame.profile.as_video_stream_profile().intrinsics
        color_frame = aligned_frames.get_color_frame()
        color_intrin = color_frame.profile.as_video_stream_profile().intrinsics

        # Validate that both frames are valid
        if not aligned_depth_frame or not color_frame:
            continue

        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        dn_frame_width = 416
        dn_frame_height = 416

        frame_width = color_image.shape[1]
        frame_height = color_image.shape[0]

	#### Passing the image to darknet
        image, detections = image_detection(color_image, network, class_names, class_colors, thresh=0.05)

        for i in range(len(detections)):
            xc_percent = detections[i][2][0]/dn_frame_width
            yc_percent = detections[i][2][1]/dn_frame_height 
            w_percent = detections[i][2][2]/dn_frame_width
            h_percent = detections[i][2][3]/dn_frame_height
            xc = xc_percent*frame_width
            yc = yc_percent*frame_height
            w = w_percent*frame_width
            h = h_percent*frame_height
            xmin = xc - w/2.0
            ymin = yc - h/2.0
            xmax = xc + w/2.0
            ymax = yc + h/2.0

            #If object is detected, increase the count of the object in the frame
            if detections[i][0] == "apple":
                cv2.rectangle(color_image, (int(xmin),int(ymin)),(int(xmax),int(ymax)),(0,0,255),2)
                cv2.putText(color_image, "apple", (int(xmin), int(ymin-10)), cv2.FONT_HERSHEY_SIMPLEX, 0.8, (0,0,255), 2)
                print(int(xmin),int(ymin),int(xmax),int(ymax))
                depth_x1 = aligned_depth_frame.get_distance(int(xmin),int(ymin))
                depth_x2 = aligned_depth_frame.get_distance(int(xmin),int(ymax))
                length_x1 = rs.rs2_deproject_pixel_to_point(depth_intrin, [xmin,ymin], depth_x1) 
                length_x2 = rs.rs2_deproject_pixel_to_point(depth_intrin, [xmin,ymax], depth_x2) 	
                print("Point 1 :" + "x - " + str(length_x1[0]) + " y - " + str(length_x1[1]) + " z - " + str(length_x1[2]))
                print("Point 2 :" + "x - " + str(length_x2[0]) + " y - " + str(length_x2[1]) + " z - " + str(length_x2[2]))
        #cv2.imwrite(output_path, frame)            
        # Render images:
        #   depth align to color on left
        #   depth on right
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        images = np.hstack((color_image, depth_colormap))
        cv2.imwrite("test.jpg", color_image)
        cv2.imshow("Hi", color_image)
        key = cv2.waitKey(0)
        # Press esc or 'q' to close the image window
        #if key & 0xFF == ord('q') or key == 27:
        cv2.destroyAllWindows()
finally:
    pipeline.stop()
```

[image]

Output is as follows 
```
Depth Scale is:  0.0002500000118743628
619 499 790 691
Point 1 :x - -0.0 y - 0.0 z - 0.0
Point 2 :x - -0.0 y - 0.0 z - 0.0
620 498 789 692
Point 1 :x - -0.0 y - 0.0 z - 0.0
Point 2 :x - -0.0 y - 0.0 z - 0.0

```

MartyG-RealSense, 2021-07-03:
Case closed due to no further comments received.

closed 2021-07-03 by MartyG-RealSense

lizhiying-2019, 2022-01-10:
Hello, I clone your code to run. Display 'OSError: [WinError 126] Cannot find this  module.' 
[image]
I want to use the L515 camera to detect object. What should I do? Could you please tell me how to do?

MartyG-RealSense, 2022-01-10:
Hi @lizhiying-2019  Someone who experienced the highlighted error suggested making sure that you have got yolo, darknet **and** OpenCV installed.   Have you got all of these installed, please?

lizhiying-2019, 2022-01-17:
Hi, I git clone this code from the address `https://github.com/AlexeyAB/darknet`. @MartyG-RealSense  So I think I have got yolo、darknet installed.

MartyG-RealSense, 2022-01-17:
Thank you for the information.  Is OpenCV installed on your computer too please, as the darknet requirements state that it should be installed separately.

https://github.com/AlexeyAB/darknet#requirements-for-windows-linux-and-macos