realsenseai/librealsense #9099: D435 must be power cycled (disconnect and reconnect cable) after unbinding and rebinding the XHCI driver

v-lopez, 2021-05-26:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 5.12.7.100 | 
| Operating System & Version |   Linux (Ubuntu 18.04)
| Kernel Version (Linux Only)    |   5.6.19                                        | 
| Platform                                 | PC  |
| SDK Version                            | 2.45.0                           | 
| Language                            |  ROS packages                          | 
| Segment			|  Robot                          | 

### Issue Description

D435 must be power cycled (disconnect and reconnect cable) after unbinding and rebinding the XHCI driver.
The T265 reconnects properly.

On some occasions we have to reset the USB subsystem to recover from some abnormal situations. The expected behavior would be that all USB devices reconnect after a few seconds of performing the reset.

But the D435 camera never recovers, and the cable must be removed and reconnected for it to recover.

Instructions on how to reset the XHCI driver can be found here:
http://blog.linuxbits.io/2015/10/01/resetting-the-usb-subsystem-on-linux/

The error the driver displays is:
> [ INFO] [1622044042.142655939]: Loading nodelet /torso_front_camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:
[ INFO] [1622044042.145114493]: waitForService: Service [/torso_front_camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1622044042.244374831]: Initializing nodelet with 16 worker threads.
[ INFO] [1622044042.247347706]: waitForService: Service [/torso_front_camera/realsense2_camera_manager/load_nodelet] is now available.
[ INFO] [1622044042.330714851]: RealSense ROS v2.2.9
[ INFO] [1622044042.330761829]: Running with LibRealSense v2.45.0
[ INFO] [1622044042.344883075]:  
[ INFO] [1622044042.351414434]: Device with serial number 948122071840 was found.
[ INFO] [1622044042.351448708]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/video4linux/video6 was found.
[ INFO] [1622044042.351475638]: Device with name Intel RealSense D435I was found.
[ INFO] [1622044042.351826125]: Device with port number 2-8 was found.
[ INFO] [1622044042.353592896]: getParameters...
[ INFO] [1622044042.411061170]: setupDevice...
[ WARN] [1622044042.411130866]: JSON file provided doesn't exist! (empty)
[ INFO] [1622044042.411180775]: ROS Node Namespace: torso_front_camera
[ INFO] [1622044042.411221412]: Device Name: Intel RealSense D435I
[ INFO] [1622044042.411236965]: Device Serial No: 948122071840
[ INFO] [1622044042.411251218]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/video4linux/video6
[ INFO] [1622044042.411279580]: Device FW version: 05.12.07.100
[ INFO] [1622044042.411295308]: Device Product ID: 0x0B3A
[ INFO] [1622044042.411309068]: Enable PointCloud: On
[ INFO] [1622044042.411322798]: Align Depth: On
[ INFO] [1622044042.411338316]: Sync Mode: On
[ INFO] [1622044042.411378474]: Device Sensors: 
[ INFO] [1622044042.411418998]: Stereo Module was found.
[ INFO] [1622044042.411456379]: RGB Camera was found.
[ INFO] [1622044042.411477245]: Motion Module was found.
[ INFO] [1622044042.411512023]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1622044042.411542531]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1622044042.411554863]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1622044042.411571569]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1622044042.411600132]: Add Filter: pointcloud
[ INFO] [1622044042.412280846]: num_filters: 1
[ INFO] [1622044042.412301975]: Setting Dynamic reconfig parameters.
[ INFO] [1622044042.621794299]: Done Setting Dynamic reconfig parameters.
[ INFO] [1622044042.624690701]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1622044042.624916323]: infra1 stream is enabled - width: 848, height: 480, fps: 30, Format: Y8
[ INFO] [1622044042.625300962]: infra2 stream is enabled - width: 848, height: 480, fps: 30, Format: Y8
 26/05 15:47:22,625 WARNING [139686740637440] (backend-v4l2.cpp:1357) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[ INFO] [1622044042.635780982]: color stream is enabled - width: 848, height: 480, fps: 30, Format: RGB8
[ INFO] [1622044042.639154284]: setupPublishers...
[ INFO] [1622044042.641408298]: Expected frequency for depth = 30.00000
[ INFO] [1622044042.679262972]: Expected frequency for infra1 = 30.00000
[ INFO] [1622044042.699069836]: Expected frequency for aligned_depth_to_infra1 = 30.00000
[ INFO] [1622044042.720794132]: Expected frequency for infra2 = 30.00000
[ INFO] [1622044042.747090549]: Expected frequency for color = 30.00000
[ INFO] [1622044042.771570100]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1622044042.794033564]: setupStreams...
[ INFO] [1622044042.794485897]: insert Depth to Stereo Module
[ INFO] [1622044042.794545566]: insert Color to RGB Camera
[ INFO] [1622044042.794573356]: insert Infrared to Stereo Module
[ INFO] [1622044042.794600160]: insert Infrared to Stereo Module
[ INFO] [1622044042.794632606]: insert Gyro to Motion Module
[ INFO] [1622044042.794664087]: insert Accel to Motion Module
 26/05 15:47:22,795 WARNING [139686740637440] (ds5-motion.cpp:473) IMU Calibration is not available, default intrinsic and extrinsic will be used.
[ERROR] [1622044042.803881056]: An exception has been thrown: xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
terminate called after throwing an instance of 'rs2::backend_error'
  what():  xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
 26/05 15:47:22,797 WARNING [139686446278400] (backend-hid.cpp:715) HID set_power 1 failed for /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.5/0003:8086:0B3A.0008/HID-SENSOR-200073.2.auto/iio:device2/buffer/enable
[torso_front_camera/realsense2_camera_manager-1] process has died [pid 32479, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager ].
[ INFO] [1622044052.442498967]: Bond broken, exiting
[torso_front_camera/realsense2_camera-2] process has finished cleanly
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete

MartyG-RealSense, 2021-05-26:
Hi @v-lopez   I can provide support in this case so long as the discussion is limited to D435 only.  Thanks very much.

Looking through your kindly provided log, there are a number of potentially problematic issues that should be ideally resolved first before further exploring the XHCI reset issue.

-  If you are using SDK version **2.45.0** then RealSense camera firmware version **5.12.13.50** is recommended.  

[image]

-  Librealsense officially supports up to kernel **5.4** at the time of writing this.  Using newer kernel versions such as the **5.6.19** kernel version listed above may have unpredictable results unless librealsense has been built with the **RSUSB backend** method (which is not dependent on Linux versions or kernel versions and does not require patching).

[image]

-  The ROS wrapper should ideally be matched as closely as possible with a particular librealsense SDK version recommended in the release notes of that particular wrapper version.  In the case of llibrealsense SDK 2.45.0, it is recommended that ROS wrapper version **2.3.0** is used with it if you are using ROS1 Melodic.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.3.0

v-lopez, 2021-05-26:
Hi, we also tried with librealsense 2.34.0 and 5.11.6 firmware.

With the same ros wrapper version. Same result

Tomorrow I'll try with the firmware,librealsense and ros wrapper version that you mention. The kernel might be harder to change since this is on a robot and the kernel is tied to other things.

MartyG-RealSense, 2021-05-26:
If you build librealsense with the RSUSB method to bypass the kernel then the rest of your project can continue to use Kernel 5.6 without having to change it.  You would have to build the ROS wrapper from source code rather than using packages though.

v-lopez, 2021-05-27:
Alright, I've compiled librealsense2 with `cmake ../ -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true `, updated firmware to 5.12.13.50, built the 2.3.0 ROS Wrapper after installing the librealsense2 version and I have the same situation, albeit the error messages are different.

One weird thing is that I see is:
>Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/video4linux/video0

Should it still be video4linux with RSUSB, did I do something wrong?

Here are the full logs, it was working fine until I did the USB driver reset.

>process[torso_front_camera/realsense2_camera_manager-1]: started with pid [3721]
process[torso_front_camera/realsense2_camera-2]: started with pid [3727]
[ INFO] [1622133503.176205177]: Loading nodelet /torso_front_camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:
[ INFO] [1622133503.180916262]: waitForService: Service [/torso_front_camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1622133503.351045860]: Initializing nodelet with 16 worker threads.
[ INFO] [1622133503.367173123]: waitForService: Service [/torso_front_camera/realsense2_camera_manager/load_nodelet] is now available.
[ INFO] [1622133503.547639079]: RealSense ROS v2.3.0
[ INFO] [1622133503.547669124]: Built with LibRealSense v2.45.0
[ INFO] [1622133503.547688227]: Running with LibRealSense v2.45.0
[ INFO] [1622133503.573757472]:  
[ INFO] [1622133503.631292193]: Device with serial number 948122071840 was found.
[ INFO] [1622133503.631337643]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/video4linux/video0 was found.
[ INFO] [1622133503.631362276]: Device with name Intel RealSense D435I was found.
[ INFO] [1622133503.632082127]: Device with port number 2-8 was found.
[ INFO] [1622133503.632135227]: Device USB type: 3.2
[ INFO] [1622133503.634053905]: getParameters...
[ INFO] [1622133503.704282151]: setupDevice...
[ WARN] [1622133503.704387634]: JSON file provided doesn't exist! (empty)
[ INFO] [1622133503.704439738]: ROS Node Namespace: torso_front_camera
[ INFO] [1622133503.704462319]: Device Name: Intel RealSense D435I
[ INFO] [1622133503.704482575]: Device Serial No: 948122071840
[ INFO] [1622133503.704500359]: Device physical port: /sys/devices/pci0000:00/0000:00:14.0/usb2/2-8/2-8:1.0/video4linux/video0
[ INFO] [1622133503.704518569]: Device FW version: 05.12.13.50
[ INFO] [1622133503.704564249]: Device Product ID: 0x0B3A
[ INFO] [1622133503.704597465]: Enable PointCloud: On
[ INFO] [1622133503.704624481]: Align Depth: On
[ INFO] [1622133503.704651972]: Sync Mode: On
[ INFO] [1622133503.704747638]: Device Sensors: 
[ INFO] [1622133503.708626798]: Stereo Module was found.
 27/05 16:38:23,708 WARNING [140108555687680] (backend-v4l2.cpp:1357) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
[ INFO] [1622133503.717730544]: RGB Camera was found.
[ INFO] [1622133503.718028177]: Motion Module was found.
[ INFO] [1622133503.718065272]: (Confidence, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1622133503.718561488]: Add Filter: pointcloud
[ INFO] [1622133503.719299084]: num_filters: 2
[ INFO] [1622133503.719327947]: Setting Dynamic reconfig parameters.
[ WARN] [1622133503.737583372]: Param '/torso_front_camera/stereo_module/auto_exposure_limit' has value 0 that is not in range [1, 165000]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1622133503.738121828]: Param '/torso_front_camera/stereo_module/auto_gain_limit' has value 0 that is not in range [16, 248]. Removing this parameter from dynamic reconfigure options.
[ WARN] [1622133503.751680839]: Param '/torso_front_camera/rgb_camera/power_line_frequency' has value 3 that is not in the enum { {50Hz: 1} {60Hz: 2} {Disabled: 0} }. Removing this parameter from dynamic reconfigure options.
[ INFO] [1622133503.765826194]: Done Setting Dynamic reconfig parameters.
[ INFO] [1622133503.766452970]: depth stream is enabled - width: 848, height: 480, fps: 30, Format: Z16
[ INFO] [1622133503.767501371]: color stream is enabled - width: 848, height: 480, fps: 30, Format: RGB8
[ INFO] [1622133503.770203639]: setupPublishers...
[ INFO] [1622133503.773462952]: Expected frequency for depth = 30.00000
[ INFO] [1622133503.805404321]: Expected frequency for color = 30.00000
[ INFO] [1622133503.829850818]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1622133503.848314546]: setupStreams...
[ INFO] [1622133503.848677352]: insert Depth to Stereo Module
[ INFO] [1622133503.848721629]: insert Color to RGB Camera
[ INFO] [1622133503.906023073]: SELECTED BASE:Depth, 0
[ INFO] [1622133503.915397471]: RealSense Node Is Up!
[ WARN] [1622133503.975063506]: 
 27/05 16:39:23,525 ERROR [140108342753024] (types.h:315) acquire: Cannot open '/dev/video0 Last Error: No such file or directory
 27/05 16:39:23,525 ERROR [140108342753024] (global_timestamp_reader.cpp:239) Error during time_diff_keeper polling: acquire: Cannot open '/dev/video0 Last Error: No such file or directory
[ERROR] [1622133563.756518322]: The device has been disconnected!
 27/05 16:39:23,757 ERROR [140108580865792] (types.h:315) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
[... cut ...]
 27/05 16:40:18,723 WARNING [140108191749888] (backend-v4l2.cpp:1122) Frames didn't arrived within 5 seconds
[ WARN] [1622133619.539098324]: Hardware Notification:USB CAM overflow,1.62213e+12,Error,Hardware Error
[ WARN] [1622133620.539515080]: Hardware Notification:USB SCP overflow,1.62213e+12,Error,Hardware Error
[ WARN] [1622133621.539856397]: Hardware Notification:USB CAM overflow,1.62213e+12,Error,Hardware Error
[ WARN] [1622133622.540201879]: Hardware Notification:USB SCP overflow,1.62213e+12,Error,Hardware Error

v-lopez, 2021-05-27:
Alright, I've compiled librealsense2 with `cmake ../ -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true `, updated firmware to 5.12.13.50, built the 2.3.0 ROS Wrapper after installing the librealsense2 version and I have the same situation, albeit the error messages are different.

I do see the physical device detected as:
>[ INFO] [1622134530.622777687]: Device physical port: 2-8-2

Which I believe indicates that it's not using video4linux.

Here are the full logs after launching the appllication after the USB driver reset.
>[ INFO] [1622134589.968304849]: Loading nodelet /torso_front_camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:
[ INFO] [1622134589.973678498]: waitForService: Service [/torso_front_camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1622134590.120621606]: Initializing nodelet with 16 worker threads.
[ INFO] [1622134590.139978782]: waitForService: Service [/torso_front_camera/realsense2_camera_manager/load_nodelet] is now available.
[ INFO] [1622134590.298098631]: RealSense ROS v2.3.0
[ INFO] [1622134590.298136803]: Built with LibRealSense v2.45.0
[ INFO] [1622134590.298159867]: Running with LibRealSense v2.45.0
[ INFO] [1622134590.343411565]:  
[ WARN] [1622134590.366231620]: Device 1/2 failed with exception: failed to set power state
 27/05 16:56:30,365 ERROR [140576603227904] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 27/05 16:56:30,365 ERROR [140576628406016] (sensor.cpp:526) acquire_power failed: failed to set power state
 27/05 16:56:30,366 WARNING [140576628406016] (rs.cpp:306) null pointer passed for argument "device"
[ WARN] [1622134590.376174235]: Device 2/2 failed with exception: Unable to open device interface
[ERROR] [1622134590.376224751]: The requested device with serial number 948122071840 and device name containing d4.5 is NOT found. Will Try again.
 27/05 16:56:30,375 ERROR [140576628406016] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 27/05 16:56:30,375 ERROR [140576628406016] (types.h:315) Unable to open device interface
 27/05 16:56:30,376 WARNING [140576628406016] (rs.cpp:306) null pointer passed for argument "device"
[ INFO] [1622134596.404429253]:  
 27/05 16:56:36,430 WARNING [140576611620608] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/05 16:56:36,487 WARNING [140576611620608] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 27/05 16:56:36,544 WARNING [140576611620608] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
Last message repaeated hundreds of times
[ WARN] [1622134602.134277448]: Device 1/2 failed with exception: set_xu(ctrl=1) failed! Last Error: Resource temporarily unavailable
 27/05 16:56:42,134 WARNING [140576628406016] (rs.cpp:306) null pointer passed for argument "device"
 27/05 16:56:42,142 ERROR [140576628406016] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
[ WARN] [1622134602.143078587]: Device 2/2 failed with exception: Unable to open device interface
[ERROR] [1622134602.143117438]: The requested device with serial number 948122071840 and device name containing d4.5 is NOT found. Will Try again.
 27/05 16:56:42,142 ERROR [140576628406016] (types.h:315) Unable to open device interface
 27/05 16:56:42,143 WARNING [140576628406016] (rs.cpp:306) null pointer passed for argument "device"

v-lopez, 2021-05-28:
I have not, but I will do next Monday when I'm back at the office. Thank you for your support, I'll get back to you.

MartyG-RealSense, 2021-05-28:
You are very welcome @v-lopez - I look forward to your report.  Good luck!

v-lopez, 2021-05-31:
@MartyG-RealSense using `roslaunch realsense2_camera rs_camera.launch initial_reset:=true` does the trick. 

Thanks a lot for your great support!

closed 2021-05-31 by v-lopez

MartyG-RealSense, 2021-05-31:
You are very welcome @v-lopez - it's excellent news that **initial_reset:=true** worked for you.  Thanks for the update!

drapado, 2021-06-07:
Similar problem with the realsense L515

MartyG-RealSense, 2021-06-07:
Hi @drapado  Please create a new case for your L515 issue by visiting the link below and left-clicking on the **New Issue** button.  Thanks!

https://github.com/IntelRealSense/librealsense/issues/