realsenseai/librealsense #9101: C# rosbag is not closed when pipeline is stopped

arturoribessanz, 2021-05-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                                                      | 
| Firmware Version                   |  05.12.13.50 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  2.45.0 | 
| Language                            |  C#  | 

Hi,

I know this has been asked before (for other programming languages) but looks like it is still an unsolved issue. When I record a rosbag and then try to play it, it gives me the following error:

Failed to resolve request. Request to enable_device_from_file("output.bag") was invalid, Reason: Failed to create ros reader: Error opening file: output.bag

If I close the application and try to open it directly, I get this error:

Failed to resolve request. Request to enable_device_from_file("output.bag") was invalid, Reason: Failed to create ros reader: Bag unindexed'

I have stopped the debugger right when the pipeline for playback is started and when I look for the file in the file explorer I see its size is 0 KB, which tells me the file is still not written to disk. When I let the application finish (or close it in the debugger) the file displays its size (it has been efectively closed).

I tried calling Dispose() methods in the pipeline and profile, and also adding a FramesReleaser to the WaitForFrames() loop, but looks like there are some resources that are still allocated that prevent the file from being correctly written to disk.

MartyG-RealSense, 2021-05-26:
Hi @arturoribessanz  The error **bag unindexed** can indicate that the recording process was not closed properly.  Stopping the pipeline with **pipeline.Stop()** can resolve this, as confirmed by a RealSense C# user in the link below who was also experiencing unindexed bags.

https://github.com/IntelRealSense/librealsense/issues/3537#issuecomment-474718891

Can you confirm whether you are usig **pipeline.Stop()** or a similar Stop() instruction when ending recoridng, please?

arturoribessanz, 2021-05-26:
Yes, I am using the pipeline.Stop() function. 

I also tried calling Dispose() on pipleline object  (after Close() ) and didn't work. I also tried wrapping the code inside a function and calling it from the main function, so all the variables would be out of scope, and didn't work either.

MartyG-RealSense, 2021-05-26:
There are bag recording and playback example scripts in the SDK's **C# Cookbook** example page that may be useful references to compare to your own code if you have not seen them already.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#recording

arturoribessanz, 2021-05-26:
Oh! I didn't know the cookbook, it has nice examples. The code in there for recording helped me with this issue. I guess there was still something that was not being disposed (lacking a using keyword).

closed 2021-05-26 by arturoribessanz