realsenseai/librealsense #9104: Python: "RuntimeError: No device connected" after opening/using realsense viewer

drapado, 2021-05-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515, D435} | 
| Firmware Version                   | (01.05.05.00, 05.12.13.50) | 
| Operating System & Version |   Pop OS 20.04 & Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.11.0-7614                                      | 
| Platform                                 | PC |
| SDK Version                            |  2.45                          | 
| Language                            |  Python                         | 
| Segment			|  Others                          | 

### Issue Description
With version 2.45, if you try to run python code to stream from a camera after you've used (and closed afterwards) realsense-viewer, you get the error "RuntimeError: No device connected". This error only occurs after opening realsense-viewer. If you just run python code right after restarting your pc, there are no errors. It seems to be a problem with version 2.45, since I haven't been able to replicate it in 2.44. The problem is consistent in different PCs and cameras (D435, L515).

RealSenseSupport, 2021-05-28:
@drapado I couldn't reproduce the issue on my Ubuntu20.04 setup. Did you get this issue with the example in Python wrapper? And how do you install librealsense and python wrapper? By installing the binary or building from source?  Looking forward to your update. Thanks!

drapado, 2021-05-28:
Indeed, just tested in a different Pc and couldn't replicate it anymore. In the system with the error I was using a conda environment while in the one without the error I use venv. I'll investigate it further. Closing the issue for now because it seems to be something related to my specific system. Thanks for your reply!

closed 2021-05-28 by drapado