realsenseai/librealsense #9105: Apt server is broken/down, gives 403 Forbidden (May 2021)

erelson, 2021-05-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | N/A |
| Firmware Version                   | N/A | 
| Operating System & Version |   Ubuntu 20.04/20.04docker  | 
| Kernel Version (Linux Only)    |  various                                         | 
| Platform                                 | N/A |
| SDK Version                            | N/A |
| Language                            |  N/A |
| Segment			| N/A |

### Issue Description
This may be a reoccurrence of https://github.com/IntelRealSense/librealsense/issues/6611, but appears to have persisted for at least 4 hours so far.  First had a build fail 4 hours ago (4am PST). An earlier build about 7 hours ago worked.

Steps to repro:
```
# on a system with realsense sources already in place
apt update

# or if you don't such sources in place (e.g. your clean build system)... we presently use the following apt sourcing:
apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-key C8B3A55A6F3EFCDE
add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u
apt update
```
Example result:
```
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/bionic/InRelease  403  Forbidden [IP: 52.218.105.153 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

erelson, 2021-05-26:
Really quickly: Workaround may be to use `add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo bionic main" -u` instead, but I've only verified that that source works for `apt update`, not installing packages.

MartyG-RealSense, 2021-05-26:
Hi @erelson   this issue was reported earlier today - the links were updated from http to https in March 2021.

https://github.com/IntelRealSense/librealsense/issues/9102#issuecomment-848822964

erelson, 2021-05-26:
Ahh shoot, thanks! That's what I get for not thinking to look at the closed issues.  Also missed the http/s difference.

rkim87, 2021-05-27:
Hi, I still don't seem to get access to the repo somehow. Any help?

erelson, 2021-05-27:
@rkim87 I should clarify, effectively I ended up doing this:

```
sed 's/http:\/\/realsense-hw-public.s3.amazonaws.com/https:\/\/librealsense.intel.com/' /etc/apt/sources.list
```
(this will print the updated contents; add `-i` after `sed` to actually have this change the file)

aka:
```diff
- deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal main
+ deb https://librealsense.intel.com/Debian/apt-repo focal main
```

My understanding was incomplete earlier, and I thought I needed to just change http to https. It is instead a full domain name change as in the above command:

rkim87, 2021-05-27:
When I try what you suggested, I keep running into the following:

`Err:6 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease`
`403  Forbidden [IP: 52.218.88.250 80]`

Is this a firewall/server issue on my end?

erelson, 2021-05-27:
That error indicates you still have the incorrect source entry somewhere in `/etc/apt/`.  Are you sure you removed it/commented it out?

rkim87, 2021-05-27:
I was able to get it to work by fixing the source entry in `/etc/apt/`. Thank you so much! C:

MartyG-RealSense, 2021-05-27:
Thanks very much @erelson for providing advice to @rkim87   :)

erelson, 2021-05-27:
This might be a thing for a separate issue, but thought I'd mention it here. Another build of ours failed because we pull and build the source from a past release of librealsense (i.e. we pin an older version for stability).  The older version in our case also does a wget from the old URLs of e.g. the firmware binary. Example URL hardcoded in an old release's branch: 

https://github.com/IntelRealSense/librealsense/blob/v2.33.1/common/fw/CMakeLists.txt#L15

The build of librealsense itself succeeds, but it's missing the files that it can't find.

We ended up adding to our build (so that we don't immediately need to upgrade and possibly break other things 🙏 ):

```
 wget -P /usr/share/librealsense2/ https://librealsense.intel.com/Releases/RS4xx/FW/D4XX_FW_Image-${REALSENSE_FW}.bin
```

erelson, 2021-05-28:
And one more detail: If building older source branches, you may also want this cmake flag so it doesn't try to import it in the first place (I guess): `-DIMPORT_DEPTH_CAM_FW=false`

MartyG-RealSense, 2021-05-28:
Thanks very much @erelson for sharing your detailed advice on this subject with the RealSense community!

matanster, 2021-06-01:
Okay so I understand from the later posts on this thread that `librealsense.intel.com/Debian/apt-repo focal main` is now the correct repository thingy to use, whereas the old "aws" name is practically deprecated. Using `sed` to make that switch seems to solve my `apt update` error that showed up few days ago. However I am not actively using RealSense these days, so my verification that this is correct might be void. Let me know if I got it right?

MartyG-RealSense, 2021-06-01:
Hi @matanster   I believe that the changes that you are referring to are the ones documented in the change history of the **distribution_lnux.md** instruction page, with the previous amazonaws links highlighted in red and the current links highlighted in green.

https://github.com/IntelRealSense/librealsense/commit/9ec146bf5a120fd8a0988296fe3e0daad2da73ff

erelson, 2021-06-05:
I think, based on a quick googling, and the reactions to several comments above indicating it's been seen a bunch, that this ticket is being found by those running across the issue.  I'm going to go ahead and close it, since it has a clear solution, is in the release notes, and finding this ticket when googling it is probably the best that anyone can hope for :) (and I assume closing it won't suddenly cause google to stop showing this in results)

closed 2021-06-05 by erelson

MartyG-RealSense, 2021-06-06:
Thanks very much @erelson for the update!

GBT16, 2021-08-28:
Hi @MartyG-RealSense sorry for late reply, I tried in that link and I think I already register the right server's public key since it unchanged when I run your program.
[image]
then I tried the next step and it's still stuck on the same problem
[image]
please help me :')

MartyG-RealSense, 2021-08-28:
@GBT16   Your problem looks like a repeat of one that a RealSense user had earlier in this discussion at https://github.com/IntelRealSense/librealsense/issues/9105#issuecomment-849212977 - the comment below it provided advice about a possible incorrect source entry somewhere in **/etc/apt/**

GBT16, 2021-08-28:
@MartyG-RealSense I already change in my source.list manually and it still back to  
`http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease` 
again like so
[image]
this is my source.list
[image]

GBT16, 2021-08-28:
@MartyG-RealSense I already resolved it thank you for your help :))

MartyG-RealSense, 2021-08-28:
@GBT16  That's great to hear, I was just writing to you now so it's excellent that you found a solution.  Thanks very much for the update!

alk15, 2022-01-21:
Hi! I seem to have a similar issue now in Ubuntu 18.04. Any help ? Thanks :)
```
sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo bionic main" -u

Err:4 https://librealsense.intel.com/Debian/apt-repo bionic Release
  Could not handshake: The TLS connection was non-properly terminated.

E: The repository 'https://librealsense.intel.com/Debian/apt-repo bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

MartyG-RealSense, 2022-01-21:
Hi @alk15  I have not previously seen that handshaking error in relation to librealsense, though my research of the error in general indicates that it can be caused by the presence of a firewall or a network proxy.

JaeYoungLee79, 2022-02-21:
So? Is it solve? 

Err:6 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease                                           
  403  Forbidden [IP: 52.218.1.138 80]

JaeYoungLee79, 2022-02-21:
Err:5 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease             
  403  Forbidden [IP: 52.218.65.17 80]
Hit:6 http://kr.archive.ubuntu.com/ubuntu bionic-updates InRelease                             
Err:7 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease            
  403  Forbidden [IP: 52.218.65.17 80]

MartyG-RealSense, 2022-02-21:
Hi @JaeYoungLee79  I have replied at the new issue that you posted at https://github.com/IntelRealSense/librealsense/issues/10251

swashy69, 2022-05-20:
Hello there
I am also getting the same error.
I am a bit of newbie to Linux, so it will be very helpful of someone who can guide me and solve my problem :)
thanks
P.S. :- My English is bad because it is not my main language so please don't mind it.

MartyG-RealSense, 2022-05-21:
Hi @swashy69  Does librealsense build for you if you use the simple installation method in the link below instead of packages?

https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md