realsenseai/librealsense #9106: Several Fatal Build errors related to libjpeg-turbo ( -DBUILD_NETWORK_DEVICE=ON )

3dsf, 2021-05-26:
f33 x64
gcc 10
sdk v2.45
`cmake ../ -DBUILD_NETWORK_DEVICE=ON`

---  
`make -j`

some examples 

> error: pathspec 'master' did not match any file(s) known to git
> CMake Error at /home/ferg/git/librealsense/build/libjpeg-turbo/tmp/libjpeg-turbo-gitupdate.cmake:201 (message):
>   Failed to checkout tag: 'master'

> librealsense/build/libjpeg-turbo/src/libjpeg-turbo-build" does not appear to contain CMakeLists.txt

---

MartyG-RealSense, 2021-05-26:
Hi @3dsf   There was a case in the past week involving  libjpeg-turbo and DBUILD_NETWORK_DEVICE in which it was believed that the problem was caused by looking for the **master** tag on the libjpeg-turbo GitHub.  The primary branch of that GitHub was now called **main**

https://github.com/IntelRealSense/librealsense/issues/9073

3dsf, 2021-05-26:
I had just commented out the checks before and continued to get errors.  like really bad errors, like crash my internet browser errors as a result of failure.  

---
  
the 2 files that can be switched from master-> main are, and one can be done with a variable called git tag.  
- /librealsense/build/libjpeg-turbo/tmp/libjpeg-turbo-gitclone.cmake
- /librealsense/build/libjpeg-turbo/tmp/libjpeg-turbo-gitupdate.cmake

---

after adjusting to main, here is an example of my really bad errors

> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:238: CMakeFiles/realsense2.dir/src/algo/thermal-loop/l500-thermal-loop.cpp.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:940: CMakeFiles/realsense2.dir/src/proc/disparity-transform.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:1460: CMakeFiles/realsense2.dir/src/device.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:836: CMakeFiles/realsense2.dir/src/proc/synthetic-stream.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:511: CMakeFiles/realsense2.dir/src/l500/l500-depth.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:1057: CMakeFiles/realsense2.dir/src/proc/color-formats-converter.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.

**Do you want a new ticket?**

MartyG-RealSense, 2021-05-26:
I note the references to 'l500' in the errors.  Are you using a RealSense L515, please?  If so, then I would recommend starting a new ticket.

3dsf, 2021-05-26:
This is compiling only, no hardware is plugged in.

- from a clean build folder
   - `cmake ../ -DBUILD_NETWORK_DEVICE=ON -DOpenGL_GL_PREFERENCE="GLVND" -DCMAKE_BUILD_TYPE=release`
   - `make -j`

closed 2021-05-27 by 3dsf

MartyG-RealSense, 2021-05-27:
Thanks very much @3dsf for the update!

I note that you have closed the case.  Have your problems in this particular case been resolved, please?  Thanks!

AndreV84, 2021-05-31:
update? he just said that he tried to build the software without hardware connected