realsenseai/librealsense #9109: Build Fail SDK 2.45 (linux x64)

3dsf, 2021-05-27:
linux x64 [F33]
GCC 10 
SDK 2.45    (SDK 2.44 builds in exact same environment)
---  
libjpeg-turbo  master-->main accommodated w/ hotfixs
---  

- new build folder
   - `cmake ../ -DOpenGL_GL_PREFERENCE="GLVND" -DCMAKE_BUILD_TYPE=release`  
   - `make -j`  

> /usr/include/libusb-1.0/libusb.h:1280:54: note: in expansion of macro ‘ZERO_SIZED_ARRAY’
>  1280 |  struct libusb_iso_packet_descriptor iso_packet_desc[ZERO_SIZED_ARRAY];
>       |                                                      ^~~~~~~~~~~~~~~~
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:329: CMakeFiles/realsense2.dir/src/ds5/ds5-nonmonochrome.cpp.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:836: CMakeFiles/realsense2.dir/src/proc/synthetic-stream.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:368: CMakeFiles/realsense2.dir/src/ds5/ds5-active.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:446: CMakeFiles/realsense2.dir/src/ds5/ds5-auto-calibration.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:550: CMakeFiles/realsense2.dir/src/l500/l500-device.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:732: CMakeFiles/realsense2.dir/src/mock/recorder.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:940: CMakeFiles/realsense2.dir/src/proc/disparity-transform.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:862: CMakeFiles/realsense2.dir/src/proc/decimation-filter.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:1616: CMakeFiles/realsense2.dir/src/rs.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:1083: CMakeFiles/realsense2.dir/src/proc/motion-transform.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:485: CMakeFiles/realsense2.dir/src/ivcam/sr300.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:1460: CMakeFiles/realsense2.dir/src/device.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:680: CMakeFiles/realsense2.dir/src/media/playback/playback_sensor.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:576: CMakeFiles/realsense2.dir/src/l500/l500-factory.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:433: CMakeFiles/realsense2.dir/src/ds5/advanced_mode/advanced_mode.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:1070: CMakeFiles/realsense2.dir/src/proc/depth-formats-converter.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:238: CMakeFiles/realsense2.dir/src/algo/thermal-loop/l500-thermal-loop.cpp.o] Error 1
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:459: CMakeFiles/realsense2.dir/src/ds5/ds5-thermal-monitor.cpp.o] Error 1
> Note: switching to '2f33be817cbce6ad7a36f27dd7ada9219f13584c'.
> 
> You are in 'detached HEAD' state. You can look around, make experimental
> changes and commit them, and you can discard any commits you make in this
> state without impacting any branches by switching back to a branch.
> 
> If you want to create a new branch to retain commits you create, you may
> do so (now or later) by using -c with the switch command. Example:
> 
>   git switch -c <new-branch-name>
> 
> Or undo this operation with:
> 
>   git switch -
> 
> Turn off this advice by setting config variable advice.detachedHead to false
> 
> HEAD is now at 2f33be817 RELEASE-NOTES: synced
> [ 47%] No update step for 'libcurl'
> [ 47%] No patch step for 'libcurl'
> [ 47%] Performing configure step for 'libcurl'
> c++: fatal error: Terminated signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [CMakeFiles/realsense2.dir/build.make:537: CMakeFiles/realsense2.dir/src/l500/l500-color.cpp.o] Error 1
> ...

3dsf, 2021-05-27:
**Additional details** 
[fedora_realsense_sdk_build_from_source_may_2021](https://www.reddit.com/r/realsense/comments/nm1n3a/fedora_realsense_sdk_build_from_source_may_2021/)

RealSenseSupport, 2021-06-04:
I think you have no enough memory in your system.
Please reduce the parallel jobs amount.

3dsf, 2021-06-04:
thanks , **builds with** `make -j 2`  

I'm a little surprised that this takes 30+ GB of ram to compile w/ 4 threads, especially when I did not notice the total system ram usage going over 4gb with 2 threads.

closed 2021-06-04 by 3dsf

3dsf, 2021-06-06:
@RealSenseSupport 

please take note of the answer by **sfx** at this thread [make-j-ram-limits](https://stackoverflow.com/questions/11639524/make-j-ram-limits)  

`make -j 4` builds with peak system ram of ~5.4 gb (more build options used than orig stated in this thread), whereas `make -j` can cause somewhere over 30 gb to be used