realsenseai/librealsense #9112: What is the range value of Short Range Preset in L515?

mogayo, 2021-05-27:
Where can i find the approximate  range of distance suggested for the presets by L515?.  The blog https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/ discusses about the receiver gain and laser power.  For example: For Short Range what is the approximate min and max range that we can expect?

mogayo, 2021-06-01:
can anyone provide a suggestion? Are there any standard procedures to check with preset suits to the required system?.

RealSenseSupport, 2021-06-15:
There is still dependency with regards to min and max range depending on environment, materials of objects in scene, reflectivity, etc.  The document discusses the tradeoffs with regards to various scenarios.  Utilizing the LRS Viewer and also the DQT tool can help in determining what works in your environment.

There is also the datasheet with spec'd min and max information:  https://dev.intelrealsense.com/docs/lidar-camera-l515-datasheet
And Users Guide with information too:  https://support.intelrealsense.com/hc/en-us/articles/360051646094-Intel-RealSense-LiDAR-Camera-L515-User-Guide

closed 2021-07-08 by RealSenseCustomerSupport