realsenseai/librealsense #9113: object detection and localization

yuanyuanhan327, 2021-05-27:
Hello, 

Are there some open-source code about object detection and localization by using D455 and python? Thanks.

MartyG-RealSense, 2021-05-27:
Hi @yuanyuanhan327   The RealSense SDK has a Python-compatible wrapper for the open-source **TensorFlow** platform that provides object detection functionality.  It works with RealSense 400 Series cameras, though I have not seen an example of use that is specific to the D455 model.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/tensorflow
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/tensorflow#part-1---object-detection-and-classification

yuanyuanhan327, 2021-05-27:
Thanks MartyG. This is really helpful. Afte finishing the object detection, for the localization (from pixel (u,v) to (X,Y,Z), do you have some suggestions?

MartyG-RealSense, 2021-05-27:
In regard to localization, do you mean identifying an object from an RGB image, please?  If so, a Deep Neural Network (DNN) based on **MobileNet-SSD** like the Python examples in the link below may meet your needs.

https://github.com/IntelRealSense/librealsense/issues/3086#issuecomment-455181310

Intel also have a Python tutorial on Jupyter Notebook called **distance_to_object** that aligns RGB and depth data and then makes use of MobileNet-SSD to identify an object and display the distance to it.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

yuanyuanhan327, 2021-05-27:
No, what I mean is after finishing the object detection, I will get the pixel coodinates of the object. I want to convert the pixel coodinates(u,v) to real world coordinates(x,y,z). I did some research and it showed I can use the rs2_deproject_pixel_to_point or rs.pointcloud.

MartyG-RealSense, 2021-05-27:
The discussion in the link below may be useful to you.

https://github.com/IntelRealSense/librealsense/issues/8853

Within that discussion is a Python point cloud script that makes use of u,v,

https://github.com/IntelRealSense/librealsense/issues/8853#issuecomment-822031895

MartyG-RealSense, 2021-05-27:
Intel's **opencv_pointcloud_viewer.py** Python example program may be useful to you as it makes use of u,v.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py#L253

yuanyuanhan327, 2021-05-27:
Thanks MartyG. I will have a look.

MartyG-RealSense, 2021-06-09:
Hi @yuanyuanhan327 Can you confirm whether you require further assistance with this case, please? Thanks!

closed 2021-06-14 by MartyG-RealSense