realsenseai/librealsense #9114: RGBD acquisiton gets stuck after 30 mins

AlexandrineRibeiro, 2021-05-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |05.11.01.100                                           | 
| Operating System & Version |  Ubuntu MATE  | 
| Kernel Version (Linux Only)    |  4.14.37 LTS                                        | 
| Platform                                 |  Odroid  X4  |
| Language                            |  Python                          | 

### Issue Description

 I am using the Intel Realsense **D435** camera module with the Odroid-XU4 running Ubuntu 18.04.

The acquisiton of depth and color **gets stuck after 30 minutes** or so.

It would be really helpful if you could share your experience.
Thanks

MartyG-RealSense, 2021-05-27:
Hi @AlexandrineRibeiro   Would it be possible to monitor your Odroid's memory usage in Python during the running of your application to see whether memory is being consumed over time (a 'memory leak').  An example of such a system monitoring tool in Ubuntu is **htop**.

A memory leak would cause an active program's performance to decline over time as available memory is consumed by the leak, to the point where the program may freeze or crash.

AlexandrineRibeiro, 2021-05-27:
Hi @MartyG-RealSense,

Thank you for the fast reply.

I monitored memory usage with psutil library. Memory usage remains stable. It doesn't seem to be a memory leak issue. 

I also tested acquiring the color frame through opencv library and it's the same thing. I can't get more than 30min streaming.

Can it be some camera configuration?

MartyG-RealSense, 2021-05-27:
If you are using the camera firmware driver 5.11.1.100 as indicated in the information at the top of this case then that is a very old firmware dating back to February 2019 and there have been many firmware updates since then at the time of writing this, with numerous improvements and fixes.  

Which RealSense SDK (librealsense) version are you using on your Odroid please?

AlexandrineRibeiro, 2021-05-28:
Hi @MartyG-RealSense,

I have updated the camera's firmware to version 12.100 and the problem continues.
I am using RealSense SDK (librealsense) version  v2.42.0 on my Odroid.

MartyG-RealSense, 2021-05-28:
I note that in the description at the top of the case it is stated that you are using Ubuntu MATE, and then Ubuntu 18.04.  Are you using Ubuntu 18.04 MATE please?

Also, if you built librealsense from source code, did you use the official installation instructions for Odroid that recommend a different patch script to the one in the normal instructions?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_odroid.md

AlexandrineRibeiro, 2021-05-28:
Hi @MartyG-RealSense ,

Yes, I am using Ubuntu 18.04 MATE and I built librealsense from source code and used the official installation instructions for Odroid. 

The installation is correct, otherwise I wouldn't even be able to access the camera images, however I can do RGBD frame acquisition, however after about 30 minutes the acquisition stops.

MartyG-RealSense, 2021-05-28:
I researched your case further and located a past Odroid case on Ubuntu with similar characteristics (starts succesfully, then has **Frames didn't arrived within 5000** after a time and fails immediately on subsequent starts.  In that case though, the problem occurred after a minute on the initial run instead of after 30 minutes.

https://github.com/IntelRealSense/librealsense/issues/3374

A RealSense team member suggested trying a mains powered USB 3 hub to see if it resolved the issue.

MartyG-RealSense, 2021-06-14:
Hi @AlexandrineRibeiro  Do you require further assistance with this case, please?  Thanks!

AlexandrineRibeiro, 2021-06-14:
Hi @MartyG-RealSense , I don't need more assistance with this.

Feel free to close this issue.

The problem was generated by the screensaver on the odroid board which caused the code to stop after running for 30 minutes.

MartyG-RealSense, 2021-06-14:
It's great to hear that you found the cause of your problem @AlexandrineRibeiro - thanks very much for the update!  As suggested, I will close this case as you have found the solution.  Thanks again!

closed 2021-06-14 by MartyG-RealSense