realsenseai/librealsense #9115: Running/Building An Example On Windows

luizaawwolf, 2021-05-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300 | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  legacy / 2.44.0                           | 
| Language                            |  C/C#/python/unity               | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hello! I'm a beginner to IntelRealSense and have been trying to dig through any documentation / issues to figure this out but unfortunately I'm not having much luck! I'm trying to get going by building one of the examples, rs-hello-realsense. I've installed the SDK, viewer, and CMake. The Viewer is working fine for me. I was able to generate using CMake following [this ](https://www.intel.co.uk/content/www/uk/en/support/articles/000030463/emerging-technologies/intel-realsense-technology.html ) tutorial, but when I click "Open Project", it prompts me to select an application, I then select VS Code, and then a .sln file opens. Not quite sure what to do with this but I don't think this is what was supposed to happen.
[image]

I also tried just running the .cpp file from VS Code, but then get an unsupported client issue. Again, I'm not sure if this is an issue relating to VS Code or if this just isn't the correct approach.

Lastly, in an attempt to follow the guidance from this [issue](https://github.com/IntelRealSense/librealsense/issues/6797), I ran the following 
`cd librealsense
mkdir build && cd build
cmake ../ -DBUILD_EXAMPLES=true
cd ./examples/hello-realsense/
./rs-hello-realsense`

It once again prompted me to select an application to open this in, I selected VS Code, and then it just opened the .cpp file in VS Code without actually running anything. 

I'm unsure where to go from here and which of these approaches is the best approach. Any help would be much appreciated! Thank you so much.

MartyG-RealSense, 2021-05-27:
Hi @luizaawwolf  If you installed the RealSense SDK for Windows using the automated installer file in the **Assets** file list of the **Releases** page then a pre-built, immediately launchable set of the examples is provided so that you do not need to build them.

https://github.com/IntelRealSense/librealsense/releases

[image]

An easy way to find the examples in the SDK's **Tools** folder is to right-click on the launch shortcut for the RealSense Viewer on the Windows desktop and select the **Open file location** menu option.  This will take you to the Tools folder where the examples (including **rs-hello-realsense.exe**) are located.

Some of the examples, such as **rs-capture**, can be launched by double-leftclicking on them.  Others need to be launched from the text-based **Command Prompt** interface of Windows.  You can find the Command Prompt by typing **command** into the search box at the bottom of your Windows screen, next to the corner button where you shut down the computer.

An easy method to launch an example in the Tools folder in the Command Prompt is:

1.  Go to the Tools folder on the Windows desktop using the previously mentioned shortcut.

2.  In the address bar at the top of the folder window, left-click on the end of the address text-box to convert it to the Windows-format address **C://Program Files (x86)/Intel RealSense SDK 2.0/Tools**

3.  Copy the contents of the address box with **CTRL-C** and open the Command Prompt window.

4.  Type **cd** followed by a space and use **CTRL-V** to paste the address into the Command Prompt window and press return.  The directory in the Command Prompt will change to the Tools folder.

5.  Input the name of the example file in the Tools folder.  For example, to run rs-hello-realsense.exe, input **rs-hello-realsense**

****************

If you would like to explore building examples again with CMake to update them after modifying their code, the https://github.com/IntelRealSense/librealsense/issues/6797 reference that you linked to is excellent advice.

*************

When you feel ready to start creating your own RealSense project in Visual Studio, a RealSense community member has created a guide to setting up a Visual Studio project with the three RealSense **props** files provided in the SDK folder.

https://github.com/EduardoWang/visual-studio-real-sense-record-and-playback-project

[image]

jmdurant, 2021-05-27:
Hi, beginner here as well.  I'm trying to see if rolling back the driver for the SR300 would help.  Support constantly links to driver 3.3.5718.  All support links for https://downloadmirror.intel.com/25044/eng/intel_rs_dcm_sr300_3.3.27.5718.exe are down.  Does anyone have a mirror or copy they could share?

luizaawwolf, 2021-05-27:
Thank you so much @MartyG-RealSense! Was able to run the examples - really appreciate it

closed 2021-05-27 by luizaawwolf

MartyG-RealSense, 2021-05-27:
Great news @luizaawwolf   :)

Hi @jmdurant   Only version **3.4.98.4970**  (intel_rs_dcm_sr300_3.4.98.4970.exe) of the SR300 DCM (Depth Camera Manager) driver installer program is available from the Intel website.   I would strongly recommend not searching for older DCM versions on internet download sites due to the risk of computer viruses.

https://downloadcenter.intel.com/download/29921/Intel-RealSense-Depth-Camera-Manager