realsenseai/librealsense #9116: 18:36:46 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:315 - MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070005: "Access is denied."

wiliamgray11, 2021-05-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSenseSupport, 2021-06-21:
@wiliamgray11 

Were you able to get past such error?  Have you tried other steps, other techniques to get passed this?  Which version of the SDK are you using?

wiliamgray11, 2021-06-21:
Dear Sir, 
I have not been able to get my Alienware model 17N (Windows 10) to connect to the L515 camera. I have tried all suggested methods to get a connection, however,  I have had no success. I did connect the L515 to different computers without any problem.

Thank you,

William M Gray, Jr.
United States Department of Justice
Drug Enforcement Administration, Retired
🦅🇺🇸

Confidentiality Notice: This e-mail message, including any attachments, is for the sole use of the intended recipient(s) and may contain confidential and privileged information. Any unauthorized review, use, disclosure, or distribution is prohibited. If you are not the intended recipient, please contact the sender by reply e-mail and destroy all copies of the original message.

closed 2021-09-17 by MartyG-RealSense