realsenseai/librealsense #9117: Question about getting Gyro data stream from D435i camera in Unity

huangbj16, 2021-05-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version | Win10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.45.0                          | 
| Language                            |  Unity                          | 
| Segment			|  IMU                    | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi, 

I met a problem when trying to add the IMU stream into one of the unitypackage sample PointCloudDepthAndColor. 

The first thing I tried is to change the configuration, adding the gyro stream.
```
public RsConfiguration DeviceConfiguration = new RsConfiguration
    {
        mode = RsConfiguration.Mode.Live,
        RequestedSerialNumber = string.Empty,
        Profiles = new RsVideoStreamRequest[] {
            new RsVideoStreamRequest {Stream = Stream.Depth, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Z16 , Framerate = 30 },
            new RsVideoStreamRequest {Stream = Stream.Infrared, StreamIndex = -1, Width = 640, Height = 480, Format = Format.Y8 , Framerate = 30 },
            new RsVideoStreamRequest {Stream = Stream.Gyro, StreamIndex = -1, Width = 640, Height = 480, Format = Format.MotionXyz32f , Framerate = 200 },
        }
    };
```
However, nothing happened.

The second thing I tried is to follow this link which I found from previous issues. I added one more profile to these profiles shown under RsDevice Script.
https://qiita.com/NWLab/items/69f1a9f8102096a3d451
It turned out that this time it worked. I could get the gyro data stream smoothly.
[image]

So I was kind of confused. What is the function of the DeviceConfiguration in the code?

I also tried once to leave the config blank like this.
```
public RsConfiguration DeviceConfiguration = new RsConfiguration
    {
        mode = RsConfiguration.Mode.Live,
        RequestedSerialNumber = string.Empty,
        Profiles = new RsVideoStreamRequest[] {
        }
    };
```
The application still worked as usual. So I don't know what exactly is the function of this code?

Thank you so much!

Bingjian,
May 28, 2021

MartyG-RealSense, 2021-05-28:
Hi @huangbj16   It looks as though the code that you were using is almost the same as code already included in the **RsDevice** script, except for the addition of the IMU references.   And you take care of the IMU definition by adding a Gyro profile to RsDevice.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/RsDevice.cs#L51-L60

huangbj16, 2021-05-28:
Hi @MartyG-RealSense! Thanks for the quick answer!

I think what you are saying is that because I added the Gyro profile, the Profiles inside DeviceConfiguration become unnecessary.
But if that is the case, why both these two "profiles" exist in the RsDevice. It may cause confusion.

Moreover, I also tried to do
1. add Gyro to the Device Configuration Profiles.
2. do nothing with the RsDevice profiles.
In this case, the Gyro stream doesn't work correctly, I didn't get any Gyro data.
So it seems like these two "profiles" do have some diffs. Just didn't know the details yet.

Anyway, thanks for the quick answer. Looking forward to more discussion!~

MartyG-RealSense, 2021-05-28:
From your description, it sounds as though you loaded the sample scene **PointCloudDepthAndColor** into Unity and in that scene was included an **RsDevice** object containing an **RsDevice** script.  The default profiles available were Element 0 (Depth) and Element 1 (Color).

At this point, you could increase the profile size from '2' to '3' to create a new element slot.  By left-clicking on the drop-down menu beside the **Stream** option of RsDevice, you could select different streams from a list, including Gyro and Accel.  From your description, it seems that this is what you have already tried.

It should therefore be possible to add Gyro to the new element profile without having to add new program code yourself.  But you found that it didn't work and you were not able to stream gyro data from your D435i after having added the Gyro profile from the menu.

In the **Format** section of the profile, did you set it to **Xyz 32f**, the format in the RealSense SDK for IMU data?

MartyG-RealSense, 2021-05-28:
I got it to work.  Aside from setting the Fornat to 'Motion Xyz 32f', the key was to set the **Framerate** of the IMU profile to a **frequency** supported by the RealSense SDK.  This is 200 or 400 for **Gyro** and 63 or 250 for **Accel**.

It does not matter what values are set for Width and Height, as Gyro and Accel do not use those settings and automatically set them both to '0' when the program is run.

huangbj16, 2021-05-31:
Hi @MartyG-RealSense,

I tried to follow your guidelines above to set the profile size, change the stream option, and set the framerate. Yeah it worked! Awesome. I also found that you can set the framerate at 0, the SDK will automatically determine the available framerate for the sensor. Nice code!

The problem I met previously in my development is that, at the beginning, I assumed that I should be able to add the IMU sensor by changing code profiles in ```DeviceConfiguration``` data structure, because it looks really reasonable to change that part. But it didn't work. It was afterwhile that I found I can change the profiles of the RsDevice. So this might cause a bit confusion for Realsense novice developers like me. 

Still, many thanks for your kind support! This problem is solved.

MartyG-RealSense, 2021-05-31:
You are very welcome @huangbj16 - it is excellent to hear that the IMU profile setup method worked for you.  Thanks for the update!

closed 2021-05-31 by MartyG-RealSense