realsenseai/librealsense #9118: Different result export_to_ply vs np.asanyarray(points.get_vertices())

drapado, 2021-05-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.05.00 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.11.0-7614                                        | 
| Platform                                 | PC |
| SDK Version                            |  2.45                         | 
| Language                            |  python                        | 
| Segment			|  others                          | 

I get different pointclouds when I used `points.export_to_ply` against `np.asanyarray(points.get_vertices()).view(np.float32).reshape(-1, 3)`. I use the following code:

```python
xyz_img = (
    np.asanyarray(points.get_vertices()).view(np.float32).reshape(-1, 3)
)
# Pass xyz to Open3D.o3d.geometry.PointCloud and visualize
pcd_0 = o3d.geometry.PointCloud()
pcd_0.points = o3d.utility.Vector3dVector(xyz_img)
pcd_0.paint_uniform_color([0, 1, 0]) # Green

points.export_to_ply(filename + ".ply", color_frame)
pcd = o3d.io.read_point_cloud(filename + ".ply")
pcd.paint_uniform_color([1, 0, 0]) # Red

o3d.visualization.draw_geometries([pcd, pcd_0])
xyz_load = np.asarray(pcd.points)
print(xyz_img.mean(axis=0), xyz_load.mean(axis=0))
# print output:
# [-0.02074947 -0.00784179  0.6535316 ] [-0.02128658  0.00804392 -0.66972431]
```
Visualization output:
[image]

Why does `points.export_to_ply` kind of invert Y and Z axes?

MartyG-RealSense, 2021-06-27:
Hi @drapado  Do you still require assistance with the problem in this case please?  Thanks!

drapado, 2021-06-29:
Hi @MartyG-RealSense yes if possible! I'd like to understand why the pointclouds are inverted when exporting to ply using pyrealsense2.
Thanks! :D

MartyG-RealSense, 2021-06-29:
Hi @drapado  I located another example in the link below of Y and Z being exported when using **export_to_ply**.  In that case, C++ language and a D435i stereo depth model was being used instead of Python and L515 lidar camera.

https://programmersought.com/article/80301367145/

[image]

It may be related to how RealSense cameras observe a scene by looking outward like a human eye, so the perspective would not be the same as when looking towards the camera from a position in front of it.

There is an alternative method of exporting a ply called **save_to_ply**.  The RealSense SDK has a Python example program called **export_ply_example** that uses this method.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py

save_to_ply also offers a greater range of options for customizing the ply export by enabling and disabling certain features.

https://github.com/IntelRealSense/librealsense/pull/4906

MartyG-RealSense, 2021-07-05:
Hi @drapado  Do you require further assistance with this case, please?  Thanks!

drapado, 2021-07-05:
@MartyG-RealSense Thanks a lot for your answer, quite clarifying! I don't require further assistance. Closing the issue :)

closed 2021-07-05 by drapado

MartyG-RealSense, 2021-07-05:
That's excellent to hear @drapado - thanks for the update!