realsenseai/librealsense #9122: Not work depth sensor without RGB camera, creative blaster X (SR300)

cvazwm, 2021-05-30:
| | |
|---|---|
|**librealsense**|2.45.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!
```

The device in use is a Creative Blaster X Senz3D (with SR300)
This error occurs from disabling RGB camera, Coded-Light Depth Sensor works only for 10 sec. and fail, closing the RealSense Viewer unexpectedly.

beside this, I'm not able to start Coded-Light Depth Sensor if RGB camera has not be starter previously, or restart pc and reconecting  the usb 3 cable to pc.

MartyG-RealSense, 2021-05-31:
Hi @cvazwm   I was able to replicate the **Access Violation** error in the RealSense Viewer on Windows with an official Intel SR300 camera.  In my first test the stream displayed correctly and disabled correctly, but produced the error when disabling the stream on the second test immediately afterwards.  The Viewer subsequently exited after clicking the OK button on the error warning box.

[image]

On subsequent launches of the Viewer, the RGB could be successfully enabled but upon disabling the stream, the **Backlight Compensation** error below appeared each time, followed a moment later by the **Access Violation** error box.  I was able to replicate this error both on the version of the Viewer that is installed by the SDK and by the standalone Viewer **(Intel.RealSense.Viewer.exe**).

[image]

In regard to the issue of the depth stream not starting unless RGB is enabled first, I was not able to replicate this particular problem.

As I was able to replicate the **Access Violation** error, I will seek advice from Intel about it.  Thanks for your patience!

MartyG-RealSense, 2021-06-08:
After discussion with Intel, an official bug report about this issue was filed with Intel.  This case should be kept open whilst that bug report is active.  Thanks very much for the report!

MartyG-RealSense, 2021-11-16:
Case closed due to the SR30x camera models being declared End of Life (retired)

closed 2021-11-22 by RealSenseCustomerSupport