realsenseai/librealsense #9123: Is there any Information about Fixed focus range of RGB sensor?

suho0515, 2021-05-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |   {Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

Hello sir, 
I'm using Realsense D435i for Robotics application.

I'll use it to get variety of information in indoor environment.

and sometimes, I should do very delicate task too, like OCR.

for this purpose, I thought that I need to find out the focus range of RGB sensor, but in every datasheet, it only says It is "Fixed Focus" but what range would be great to observe objects.

so my question is " is there any information about fixed focus range of rgb sensor?"

---

and I tried to calculate appropriate working distance of rgb sensor.

like this

focal length x FOV / sensor size = Working Distance

refferd from [https://www.ni.com/ko-kr/support/documentation/supplemental/18/calculating-camera-sensor-resolution-and-lens-focal-length.html](url)

is it right approch?

I hope to hear your anser soon.

regards.

agrunnet, 2021-05-31:
Actually it is more about what the hyper focal distance is. I don’t have the exact number in front of me but the image is in focus from roughly 30cm to infinity.

MartyG-RealSense, 2021-05-31:
Thanks very much @agrunnet for your advice!

@suho0515  Do you reuire further assistance with your question, please?  Thanks!

suho0515, 2021-06-02:
Thank you @agrunnet :).
 I wanted more detail, but I guess there is no way to figure it out.

so, I'll close this issue.

Thanks, @MartyG-RealSense too.

closed 2021-06-02 by suho0515

MartyG-RealSense, 2021-06-02:
You are very welcome @suho0515 - thanks for the update!

agrunnet, 2021-06-02:
@suho0515 I will give you more details. The way to calculate the hyper focal distance of an imaging system is H= focal length * focal length / (circle of confusion * F-number)

The circle of confusion is how many pixels of blur you can tolerate. Let's set it to 2 pixels.  Since each pixel is 1.4um, this means 0.0028mm. 
 
For D435 the color imager has focal length of 1.91mm. 
Finally the F# is also a lens spec. It is 2. 

With this the hyperfocal distance is 65cm. 

However, the definition of hyper focal distance is that the "image is in focus from HALF hyper focal distance to infinity". 

So that means ~33cm to infinity.

suho0515, 2021-06-29:
> @suho0515 I will give you more details. The way to calculate the hyper focal distance of an imaging system is H= focal length * focal length / (circle of confusion * F-number)
>
> The circle of [...]

sorry for late answer,

and Really appriciate for your details.

It is absolutely helpful, and the answer what I was looking for.

Now the Issue is completely resolved.

Thanks again @agrunnet and @MartyG-RealSense

MartyG-RealSense, 2021-06-29:
You are very welcome @suho0515 - it's great to hear that the advice provided by @agrunnet was helpful to you.  Thanks for the update!

JZhu1208, 2021-07-09:
Hello @MartyG-RealSense @agrunnet,

I want to know what the focal length of the color imager and F# are for L515. Thank you very much.

MartyG-RealSense, 2021-07-09:
Hi @JZhu1208   The table below is taken from the data sheet document for the L515 camera.

[image]