realsenseai/librealsense #9124: [OSX] BUILD_WITH_TM2 requires IMPORT_DEPTH_CAM_FW

PhoebeHui, 2021-05-31:
Hi All,

We built realsense on osx platform and encountered the following error, I took a look at this issue and found the root cause:

From https://github.com/IntelRealSense/librealsense/blob/master/CMakeLists.txt#L14, there is no internet connection, so BUILD_WITH_TM2 was disabled. IMPORT_DEPTH_CAM_FW as well.
However, https://github.com/IntelRealSense/librealsense/blob/master/CMake/unix_config.cmake#L38 directly set the BUILD_WITH_TM2 to ON, it override the previous checks, and it ignore the IMPORT_DEPTH_CAM_FW check, that cause the error.
Could you please take a look?

Failures:
```
CMake Warning at CMakeLists.txt:22 (message):
  No internet connection, disabling IMPORT_DEPTH_CAM_FW

CMake Error at src/tm2/CMakeLists.txt:16 (message):
  BUILD_WITH_TM2 requires IMPORT_DEPTH_CAM_FW
Call Stack (most recent call first):
  src/CMakeLists.txt:57 (include)
  CMakeLists.txt:56 (include)
```

Repro steps:
1. git clone https://github.com/microsoft/vcpkg
2. cd vcpkg
3. ./bootstrap-vcpkg.bat
4. ./vcpkg install realsense2:x64-osx

MartyG-RealSense, 2021-05-31:
Hi @PhoebeHui   Could you provide further details of the method that you are using to build librealsense with vcpkg on MacOS?  Your use of **/bootstrap-vcpkg.bat** makes me think that you may be using the Windows installation instructions for vcpkg, perhaps through a Windows compatibility tool for MacOS such as **Wine**.  Is that what you are doing, please?

The official vcpkg instructions, meanwhile, suggest as a Mac installation method to install **MacOS Developer Tools** as a prerequisite, and also GCC if using a MacOS version that is earlier than 10.15.

https://github.com/microsoft/vcpkg#installing-macos-developer-tools

After that prerequisite is installed for MacOS, it looks as though the **Unix** instructions are recommended for Mac.

https://github.com/microsoft/vcpkg#quick-start-unix

PhoebeHui, 2021-05-31:
@MartyG-RealSense, sorry for the mistake, it should be 'bootstrap-vcpkg.sh'.

MartyG-RealSense, 2021-05-31:
Thanks very much for the clarification @PhoebeHui 

May I ask whether your Mac is a recent model with an **Apple Silicon M1** processor rather than the Intel processor that older Mac models have?

PhoebeHui, 2021-05-31:
We're using the old mac models:

macOS: 11.2.3
Xcode CLTs: 12.4

MartyG-RealSense, 2021-05-31:
There is a past RealSense case on Microsoft's vcpkg support forum with some similarities, where a RealSense user believed that although **BUILD_WITH_TM2** should be set to Off if an internet connection was not present, they were not seeing confirmation of this change in the CMake output.

https://github.com/microsoft/vcpkg/issues/10640

They were advised to edit the vcpkg file **portfile.cmake** to add the instruction **set(BUILD_WITH_TM2 OFF)**

https://github.com/microsoft/vcpkg/issues/10640#issuecomment-607753144

PhoebeHui, 2021-06-01:
Yes, we actually supported it as a feature, it disabled via BUILD_WITH_TM2 by default, you can enabled it by installing the feature tm2, eg: './vcpkg install realsense2[tm2]', however, due to this issue, it doesn't work.

MartyG-RealSense, 2021-06-01:
I will highlight the case to Intel to seek their advice, as I do not have a Mac to test with.  Thanks for your patience!

MartyG-RealSense, 2021-06-01:
If you are building without an internet connection, I wonder if part of the problem is the line that sets **FORCE_RSUSB_BACKEND** to On.  The RSUSB method of building librealsense requires an internet connection to build the dependencies instead of patching them in.  So presumably, building librealsense with RSUSB would fail if there was not an internet connection to install the dpeendencies over.

https://github.com/IntelRealSense/librealsense/blob/master/CMake/unix_config.cmake#L37

PhoebeHui, 2021-06-02:
Well, for 'FORCE_RSUSB_BACKEND' it doesn't check the internet connection, so it doesn't fail here.
I don't think it's right thing to set BUILD_WITH_TM2 to ON in https://github.com/IntelRealSense/librealsense/blob/master/CMake/unix_config.cmake#L38, it should be in control of https://github.com/IntelRealSense/librealsense/blob/master/CMakeLists.txt#L15

MartyG-RealSense, 2021-06-02:
Does your project require you to build librealsense with vcpkg or are you doing so for convenience because it is simpler than the normal MacOS build method on the **installation_osx.md** page?

PhoebeHui, 2021-06-03:
For me It's convenience and simpler to use vcpkg to build librealsense.

MartyG-RealSense, 2021-06-03:
Have you ever been able to build librealsense on Mac with vcpkg without an internet connection or is it only recently that you have encountered this problem, please?

MartyG-RealSense, 2021-06-09:
Hi @PhoebeHui  Do you require further assistance with this case, please?  Thanks!

PhoebeHui, 2021-06-09:
Hi @MartyG-RealSense, yes, I expect this could be fixed in Upstream, I add a patch to fix this in vcpkg, see https://github.com/microsoft/vcpkg/pull/18202.

MartyG-RealSense, 2021-06-09:
Thanks very much @PhoebeHui for the update and for contributing a pull-request fix to Microsoft's vcpkg GitHub!  I see that it has been merged 6 days ago at the time of writing this.

MartyG-RealSense, 2021-06-15:
Hi @PhoebeHui  As your pull request is on the vcpkg GitHub and it has been merged, is it okay to close this case here now, please?  Thanks!

PhoebeHui, 2021-06-15:
@MartyG-RealSense, I expect this could be fixed in librealsense, and then vcpkg would remove the patch in next revision.

MartyG-RealSense, 2021-06-15:
The vcpkg installation information on the librealsense GitHub front page contains the following statement:

"The librealsense port in vcpkg is kept up to date by Microsoft team members and community contributors.  If the version is out of date, please create an issue or pull request on the vcpkg repository".

So a patch on the vcpkg side may be considered to be the appropriate way to handle the situation.

MartyG-RealSense, 2021-06-27:
Case cloed due to no further comments received.

closed 2021-06-27 by MartyG-RealSense