realsenseai/librealsense #9126: Device not connected

whosmyqueen, 2021-05-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | () | 
| Operating System & Version |   ANDROID  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | ANDROID 10 HUAWEI SNE-AL00  |
| SDK Version                            |  { 2.45.0 }                          | 
| Language                            |  {JAVA }                          | 
| Segment			|  {PHONE }                          | 

### Issue Description
`2021-05-31 16:53:29.939 12786-12786/com.intel.realsense.camera W/librs UsbUtilities: getUsbDevice: failed to locate USB device, VID: 0x8086, PID: 0x0000`

MartyG-RealSense, 2021-05-31:
Hi @whosmyqueen  The RealSense SDK's Android wrapper is known to have issues with Android 10 devices for some users.  Could you try setting **targetSdkVersion** to a value of '27' please, as this has been a consistently successful fix for Android 10 users in the past.  I understand though that this method increasingly is difficult to use as time goes by and applications rely on features introduced in Android OS versions newer than 27.

Official support for Android 11 was added to the RealSense SDK in version **2.43.0**.

https://github.com/IntelRealSense/librealsense/pull/8298

whosmyqueen, 2021-05-31:
@MartyG-RealSense ok! let me try

whosmyqueen, 2021-05-31:
@MartyG-RealSense I tried what you said, or the same mistake happened.
And the plug-in device doesn't seem to be recognized！

error info：
`2021-05-31 18:06:15.788 20219-20219/com.intel.realsense.camera W/librs UsbUtilities: getUsbDevice: failed to locate USB device, VID: 0x8086, PID: 0x0000`

MartyG-RealSense, 2021-05-31:
Do you know which firnware version your camera is using?

You can check in the RealSense Viewer on a computer by left-clicking on the **Info** option in the options side-panel.

[image]

whosmyqueen, 2021-05-31:
@MartyG-RealSense 
[image]

MartyG-RealSense, 2021-05-31:
Thanks very much.  I just wanted to confirm that your firmware was a recent one that is compatible with 2.45.0, and it is.

Which installation method did you use to set up the Android wrapper one?  Was it the old **rooted** device method?

https://github.com/IntelRealSense/librealsense/tree/v2.18.1/doc/android

Or was it the current **unrooted** method?

https://github.com/IntelRealSense/librealsense/blob/master/doc/android.md

whosmyqueen, 2021-06-01:
@MartyG-RealSense 
I used unrooted method

whosmyqueen, 2021-06-01:
‘capture’ demo

MartyG-RealSense, 2021-06-01:
Could you download and install Intel's free **RS Camera** RealSense Android app from the Play Store please?  The current version is built around SDK 2.45.0, so if it does not work then it might indicate a problem with the Android device that prevents it from working with the camera, such as an incompatible chipset in the phone.

https://play.google.com/store/apps/details?id=com.intel.realsense.camera

whosmyqueen, 2021-06-02:
@MartyG-RealSense 
I can't download this APK. Can you upload it？

MartyG-RealSense, 2021-06-02:
I do not know of a permitted way to obtain the RS Camera app outside of the Play store, unfortunately.

Where does the **failed to locate USB device** error occur?  For example, does it happen when trying to run the Android wrapper's Java example?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/java_example/readme.md

whosmyqueen, 2021-06-02:
I tried this, but I couldn't detect the camera

MartyG-RealSense, 2021-06-02:
Is the Huawei SNE-AL00 phone model that you are using the one in the link below (also known as Mate 20 Lite).

https://xphone24.com/huawei-mate-20-lite.php

It looks as though this phone model dates back to a time before Huawei stopped using Google apps on their phones.  So you have the Play Store app on your phone but could not download the **RS Camera** app from the Store?

If the camera cannot be detected by any application, whether it is the RealSense Viewer or the Java example, then the problem might be in the USB 3 OTG cable used to connect the camera to the phone.

whosmyqueen, 2021-06-03:
I can connect to the PC version of realsense viewer.

MartyG-RealSense, 2021-06-03:
The camera would be connected to the PC by an ordinary USB cable like the one supplied with it and not a USB3 OTG adapter though, wouldn't it?

MartyG-RealSense, 2021-06-09:
Hi @whosmyqueen  Do you require further assistance with this case, please?  Thanks!

whosmyqueen, 2021-06-09:
now，I used targetSdkVersion 27

MartyG-RealSense, 2021-06-09:
Ok, thanks very much @whosmyqueen for the update.  Can you confirm please that your problem was solved by setting **targetSdkVersion** to 27?

closed 2021-06-15 by MartyG-RealSense