realsenseai/librealsense #9128: SR300 not acquiring depth frames (Linux)

rhaschke, 2021-05-31:
Trying to connect the depth camera with `realsense-viewer`, the camera just disconnects after a few seconds, never showing any frame. Running the basic examples, I get this error:
```
RealSense error calling rs2_pipeline_wait_for_frames(pipe:0x5564abec8e00):
    Frame didn't arrive within 15000
```
Grabbing color frames works as expected - both using `realsense-viewer` and e.g. `rs-color`.
On Windows, everything works fine - including depth. So, the camera is OK.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | **SR300** | 
| Firmware Version                   | 3.21.0.0 | 
| Operating System & Version | Linux Ubuntu 20.04  | 
| Kernel Version (Linux Only)    | 5.4.0-73-generic | 
| Platform                                 | PC  |
| SDK Version                            | 2.45.0 and development branch | 
| Language                            |  C++ |

MartyG-RealSense, 2021-05-31:
Hi @rhaschke   Could you test please whether the **Laser Power** value of the camera is a factor in the SR300 disconnecting when the depth stream is used. 

A method for doing this is to launch the Viewer and expand open the **Controls** category of the Viewer's options side-panel to locate an option called **Laser Power**.  

[image]

Set the **Laser Power** option to zero using the slider before you enable the depth stream.  

[image]

If the camera does not disconnect, slowly move the slider to increase the Laser Power value incrementally.  

[image]

If the camera disconnects after the Laser Power is increased above a certain level then this indicates that the problem is related to the USB port supplying insufficient power to the camera to meet its power needs.

rhaschke, 2021-05-31:
The camera only works with `LaserPower = 0`. I will try on a different machine. Thanks for the hint!

rhaschke, 2021-05-31:
Indeed, moving to a different laptop resolved the issue. However, I noticed that it's not possible to stream both depth and color at 30 Hz. Trying to do so, one stream is lagging a second behind!

closed 2021-05-31 by rhaschke

MartyG-RealSense, 2021-05-31:
Hi @rhaschke  If you are using auto-exposure and are using both depth and color at the same time, you can enforce a constant FPS by going to the **RGB** section of the Viewer's options side-panel and disabling an option called **Auto-Exposure Priority** (a separate option from the Auto-Exposure in that section).