realsenseai/librealsense #9133: Stand Alone Realsense Viewer won't run

KirkPen, 2021-05-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

KirkPen, 2021-05-31:
Is the stand alone realsense viewer supposed to run on old cpu like mine? It does nothing for me, and the one that comes with the SDK give the application was unable to start correctly.  New install of Win10 64, Intel I7 870 CPU, D435I camera
The depth quality tool works.

MartyG-RealSense, 2021-06-01:
Hi @KirkPen   RealSense 400 Series cameras can work with any Intel or Arm processor, and you have found that you can use your i7-870 (which dates back to 2009) with the Depth Quality Tool.

Could you check your Windows user account folder on that computer please to see whether you have a folder within it called **Documents** please?  There have been a couple of past cases where the Viewer could not launch if this folder was absent but the Depth Quality Tool could work.

For example, if your Windows user account name was KirkPen then the Documents folder may be located at **C:\Users\KirkPen\Documents**

KirkPen, 2021-06-01:
Yes. my Documents folder was on my D:\. I make a new one and now it works,
thank you.

MartyG-RealSense, 2021-06-01:
Great news @KirkPen - thanks very much for the update!