realsenseai/librealsense #9134: Will the application of depth camera lead to false detection of yolov5 algorithm？

yangminwei95, 2021-06-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                                                      | 
| Firmware Version                   | opencv                                                    | 
| Operating System & Version |   {Win (10)                                               | 
| Kernel Version (Linux Only)    |                                                                | 
| Platform                                 | PC                                                           |
| SDK Version                            |  pyrealsense2                                        | 
| Language                            |  python                                                     | 
| Segment			|  Robot                                                             | 

### Issue Description
I want to insert the depth camera into yolov5 algorithm, so that the output of yolov5 not only contains confidence, but also can express the three-dimensional coordinates of the center point of the detection box. But the effect is not good. The improved yolov5 algorithm always detects the background as an object when detecting objects. It can't be as accurate as the original color camera detection through opencv. Is it because of the alignment between the depth map and the color map?
In addition, the following problems sometimes occur when the program is running，
I want to insert the depth camera into yolov5 algorithm, so that the output of yolov5 not only contains confidence, but also can express the three-dimensional coordinates of the center point of the detection frame. But the effect is not good. The improved yolov5 algorithm always detects the background as an object when detecting objects. It can't be as accurate as the original color camera detection through opencv. Is it because of the alignment between the depth map and the color map?
In addition, the following problems sometimes occur when the program is running，

 depth = depth_frame.as_depth_frame().get_distance(pixel_x, pixel_y)
RuntimeError: out of range value for argument "y"

I want to know why. The code is attached at the back.
`import pyrealsense2 as rs
import argparse
import time
from pathlib import Path
import numpy as np
import cv2
import torch
import torch.backends.cudnn as cudnn
from collections import defaultdict
from numpy import random
from realsense_device_manager import *
from models.experimental import attempt_load
from utils.datasets import LoadStreams, LoadImages
from utils.general import check_img_size, check_requirements, check_imshow, non_max_suppression, apply_classifier, \
    scale_coords, xyxy2xywh, strip_optimizer, set_logging, increment_path
from utils.plots import plot_one_box
from utils.torch_utils import select_device, load_classifier, time_synchronized
import calibration_kabsch as ck
def detect(save_img=False):
    weights, view_img, save_txt, imgsz = opt.weights, opt.view_img, opt.save_txt, opt.img_size
    # webcam = source.isnumeric() or source.endswith('.txt') or source.lower().startswith(
    # ('rtsp://', 'rtmp://', 'http://'))
    # Directories
    save_dir = Path(increment_path(Path(opt.project) / opt.name, exist_ok=opt.exist_ok))  # increment run
    print('save_dir=', save_dir)
    (save_dir / 'labels' if save_txt else save_dir).mkdir(parents=True, exist_ok=True)  # make dir
    print('weights=', weights, 'view_img=', view_img, 'save_txt=', save_txt, 'imgsz=', imgsz)

    # Initialize
    set_logging()
    device = select_device(opt.device)
    print('device=', device)
    half = device.type != 'cpu'  # half precision only supported on CUDA
    print('half=', half)
    # Load model
    model = attempt_load(weights, map_location=device)  # load FP32 model
    # print('model=', model)
    stride = int(model.stride.max())  # model stride
    print('stride=', stride)
    imgsz = check_img_size(imgsz, s=stride)  # check img_size
    print('imgsz=', imgsz)
    if half:
        model.half()  # to FP16

    # Second-stage classifier
  
    classify = False
[... 193 lines cut ...]
    parser.add_argument('--iou-thres', type=float, default=0.45, help='IOU threshold for NMS')
    parser.add_argument('--device', default='cpu', help='cuda device, i.e. 0 or 0,1,2,3 or cpu')
    parser.add_argument('--view-img', action='store_false', help='display results')
    parser.add_argument('--save-txt', action='store_false', help='save results to *.txt')
    parser.add_argument('--save-conf', action='store_false', help='save confidences in --save-txt labels')
    parser.add_argument('--classes', nargs='+', type=int, help='filter by class: --class 0, or --class 0 2 3'）
    parser.add_argument('--agnostic-nms', action='store_false', help='class-agnostic NMS')
    parser.add_argument('--augment', action='store_true', help='augmented inference')
    parser.add_argument('--update', action='store_true', help='update all models')
    parser.add_argument('--project', default='runs/detect', help='save results to project/name')
    parser.add_argument('--name', default='exp', help='save results to project/name')
    parser.add_argument('--exist-ok', action='store_true', help='existing project/name ok, do not increment')
    opt = parser.parse_args()
    print('opt=', opt)
    check_requirements()

    with torch.no_grad():  
        if opt.update:  # update all models (to fix SourceChangeWarning)
            for opt.weights in ['yolov5s.pt', 'yolov5m.pt', 'yolov5l.pt', 'yolov5x.pt']:
                detect()
                strip_optimizer(opt.weights)  
        else:
            detect()

[image]

MartyG-RealSense, 2021-06-01:
Depth to color alignment should be beneficial when the alignment is accurate, as it can help to distinguish background pixels from foreground pixels.

The get_distance instruction seems to have a problem with the value of **pixel_y** that is being fed to it.  I have seen a similar Python case in the past involving an out of range error when using alignment and get_distance, and storing pixel values in x and y variables.  In that particular case, the error was most likely to occur when reading coordinates near the edge of the image rather than the center of the image.

https://github.com/IntelRealSense/librealsense/issues/7395

The RealSense user in that case concluded that their problem was being caused by the alignment, and they posted an updated script at the end of the case.

https://github.com/IntelRealSense/librealsense/issues/7395#issuecomment-698016942

**********

If you are performing object detection with Python then it may also be worth considering using the RealSense SDK's compatibility wrapper for the open-source **TensorFlow** platform.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/tensorflow

yangminwei95, 2021-06-01:
Thanks for your reply，but But it didn't solve my problem.Is there a way for opencv to open the depth camera and color camera of D435i at the same time?

MartyG-RealSense, 2021-06-01:
If you are aiming for a program for OpenCV that uses Python and displays depth and color as separate panels, the SDK's **opencv_viewer_example** program may meet your needs.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py

The link below describes how to adapt that example for multiple cameras on Windows if you should need to.

https://rahulvishwakarma.wordpress.com/2019/08/17/realsense-435i-depth-rgb-multi-camera-setup-and-opencv-python-wrapper-intel-realsense-sdk-2-0-compiled-from-source-on-win10/

If you need depth to color alignment with Python, the SDK example **align_depth2color.py** imports OpenCV for image rendering.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/align-depth2color.py

If use of Python is not important to you though then the OpenCV-equipped depth to color alignment project in the link below may be helpful.

https://github.com/UnaNancyOwen/RealSense2Sample/tree/master/sample/Align

And Intel provide an OpenCV C++ script for alignment here:

https://dev.intelrealsense.com/docs/opencv-wrapper#section-get-aligned-color-depth

sam598, 2021-06-01:
@yangminwei95 

Do you want yolov5 to use both RGB color and Depth data when it is running inference to better detect an object?

Or do you want to detect an object from just the RGB data, and then find that object's 3D position using the depth data within that bounding box?

yangminwei95, 2021-06-02:
> @yangminwei95
> 
> Do you want yolov5 to use both RGB color and Depth data when it is running inference to better detect an object?
> 
> Or do you want to detect an object from just the RGB data, and then find that object's 3D position using the depth data within that bounding box?

Yes, I want to detect an object from just the RGB data, and then find that object's 3D position using the depth data within that bounding box.However, if the color camera and depth camera are turned on at the same time by using the method provided by pyrealsense, the detection effect will be poor, and the real target object will not be detected, so the background will always be detected as the target.
Do you have any good methods?Thank you in advance.

MartyG-RealSense, 2021-06-09:
Hi @yangminwei95  Do you require further assistance with this case, please?  Thanks!

yangminwei95, 2021-06-12:
> Do you require further assistance with this case, please? Thanks!

I don't need it for the time being. I found the cause of the problem. Because I made a mistake about the dimensions of the data, it led to false detection. Thank you sincerely for your help these days.

MartyG-RealSense, 2021-06-12:
It is great to hear that you found the solution @yangminwei95  - thanks very much for the update!  I will close this case.

closed 2021-06-12 by MartyG-RealSense