realsenseai/librealsense #9138: How to get synchronized imu data without ROS

lbw0502, 2021-06-01:
Hi,

I am using D455 to develop VIO algorithm. 

**if I use ROS**, I can change the `accel_fps`, `accel_fps`, and set `unite_imu_method` as linear_interpolation in launch file to get synchronized imu data.

My question is:
If I only use realsense sdk(**without ROS**) to open the camera, how should I change the imu rate and get such synchronized imu data?
I didn't find it in the sdk.

Thanks for help!

MartyG-RealSense, 2021-06-01:
Hi @lbw0502  The purpose of **unite_imu_method** in ROS is to publish two separate IMU topics (Gyro and Accel) as a single topic called **imu**.   Its **linear interpolation** condition interpolates accel messages to the gyro timestamp.

In librealsense though, Gyro and Accel are published as separate streams.  Sync for gyro and accel in the SDK for the purposes of VIO is a subject that has been discussed in the past in the links below.  I am not aware of a standard built-in function for doing so in librealsense though.

https://github.com/IntelRealSense/librealsense/issues/3921
https://github.com/IntelRealSense/librealsense/issues/5712

In regard to setting a frequency for gyro and accel separately, could you tell me whether you are using C++ or Python please so that I can provide the correct information.  Thanks!

lbw0502, 2021-06-01:
Thanks for your reply!

I am using C++.

MartyG-RealSense, 2021-06-01:
Thanks very much for the confirmation.  The link below provides information about defining a gyro and accel stream in C++, the frequencies that are supported for gyro and accel, and a link to a full C++ IMU script.

https://support.intelrealsense.com/hc/en-us/community/posts/1500001145522/comments/1500000958142

MartyG-RealSense, 2021-06-08:
Hi @lbw0502  Do you require further assistance with this case, please?  Thanks!

lbw0502, 2021-06-08:
Hi MartyG, sorry for the late reply... I have found another way to do the synchronization. Here is the link:

https://github.com/lifelong-robotic-vision/openloris-scene-tools/blob/master/dataprocess/merge_imu_topics.py

I have solved my problem. Thanks for your help :)

closed 2021-06-08 by lbw0502

MartyG-RealSense, 2021-06-08:
No problem at all @lbw0502 - it is excellent news that you found a solution, and thanks very much for sharing it with the RealSense ROS community!   :)