realsenseai/librealsense #9141: memory leak in valgrind report: librealsesne read depth frame from bagfile c++

sanket-mali, 2021-06-02:
### Issue Description
```
rs2::pipeline p;

rs2::config cfg;
cfg.enable_device_from_file("test.bag");

// Configure and start the pipeline
p.start(cfg);
int i = 0;
while (i++<5)
{
   std::cout<<"i: "<<i<<std::endl;
   // Block program until frames arrive
   rs2::frameset frames = p.wait_for_frames();
   if( !frames)
     continue;

   // Try to get a frame of a depth image
   rs2::depth_frame depth = frames.get_depth_frame();

   if (! depth)
          continue;

   // Get the depth frame's dimensions
   auto width = depth.get_width();
   auto height = depth.get_height();

   // Query the distance from the camera to the object in the center of the image
   float dist_to_center = depth.get_distance(width / 2, height / 2);

   // Print the distance
   std::cout << "The camera is facing an object " << dist_to_center << " meters away \r";
 }
 p.stop();
```
Trying to read depth frame from a pre-recorded .bag files. Running with valgrind triggering some memory leake. Is there any problem in the way I am using the API's or it's a known library problem. Is there any known solution.

sanket-mali, 2021-06-02:
valgrind memcheck report
[realsense_memcheck_nofilter_pipestop.log](https://github.com/IntelRealSense/librealsense/files/6584981/realsense_memcheck_nofilter_pipestop.log)

MartyG-RealSense, 2021-06-02:
This is a continuation of a case first posted by @sanket-mali  at the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/1500001255362

Hi @sanket-mali   Continuing onward from the point that the previous discussion ended at: were you able to use a memory monitoring tool such as Task Manager (Windows) or htop (Ubuntu) to monitor memory usage when running your project outside of Valgrind, please?  The purpose of this request was to confirm whether the reported memory leak was only occurring when the project was run with Valgrind.

Hi @rakeshb1977   Could you visit the link above please to read what was previously suggested to @sanket-mali before we continue onwards here on this case please?  Thanks!

rakeshb1977, 2021-06-02:
Thanks @MartyG-RealSense  for a quick response. I went through your suggestion and did the experiment that you have asked for. On the Linux system, I ran the code that I have posted for few iterations and captured the RESIDENT memory usage and %MEM usage. Please find the analysis.

From the data and graph -- we can see a slight upward trend in the memory usage. It's not flat over the period of time.
So, I suspect memory leak here.

Also, you have provided a link that talks about "https://github.com/IntelRealSense/librealsense/issues/3433". Do you know if this known issue of "Valgrind: Conditional jump or move depends on uninitialized value(s)" also leads to Valgrind reporting "Definitely lost/Directly Lost/Indirectly lost" error? I see many of those errors in the Valgrind report?

Based on top output, it looks like the Valgrind is reporting genuine memory leaks?

Thanks for your support
[memory_analysis.xlsx](https://github.com/IntelRealSense/librealsense/files/6586646/memory_analysis.xlsx)

MartyG-RealSense, 2021-06-03:
Hi @rakeshb1977   Which RealSense camera model are you using, please?  There have been past reports by C++ and C# users of memory leaks on models equipped with an IMU (D435i and D455), compared to IMU-less D415 and D435 where memory usage was flat and steady.

https://support.intelrealsense.com/hc/en-us/community/posts/1500000840882-Memory-leak-when-using-context-QueryDevices-C-D455- 

The RealSense user who posted the **Post-processing filters in C: leak and segmentation fault** case also made a post about 'false positives' in Valgrind and dealing with them using a **suppression list.**

https://support.intelrealsense.com/hc/en-us/community/posts/360052556133-Feature-request-Valgrind-suppression-list-for-librealsense

I am not familiar enough with how Valgrind functions to conclude whether conditional jumps are connected directly to memory leaks.

rakeshb1977, 2021-06-03:
Hello @MartyG-RealSense  -  We use Intel RealSense D415

MartyG-RealSense, 2021-06-03:
@rakeshb1977  Your script and that of @sanket-mali are very similar.  So there does seem to be a reproducible issue here, since both are reporting leaks.  

Given that the 'conditional jump' Valgrind issue documented in https://github.com/IntelRealSense/librealsense/issues/3433 has been listed on the 'Known Issues' section of the release notes since SDK version **2.19.2** in March 2019 though, there may not be a fix for this issue in the near future.

MartyG-RealSense, 2021-06-09:
Hi @sanket-mali  Do you require further assistance with this case, please? Thanks!

sanket-mali, 2021-06-09:
Thanks.. we are still investigating the issue. Let's keep it open for now..

MartyG-RealSense, 2021-06-09:
Okay, no problem @sanket-mali - thanks very much for the update!

MartyG-RealSense, 2021-06-28:
Hi @sanket-mali   If we do not hear from you then this case will be closed after 7 days from the time of writing this.  Thanks very much.

closed 2021-07-05 by MartyG-RealSense