realsenseai/librealsense #9144: Splitting multicam rs2::syncer frameset into one frameset per device

roelofvandijk, 2021-06-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.10.00 | 
| Operating System & Version |   Win 10 | | 
| Platform                                 | PC |
| SDK Version                            | 2.38.1                     | 
| Language                            |  C++            | 
| Segment			| Robot                          | 

### Issue Description

I am using an `rs2::syncer` object to receive synchronized frames from two D415s (similar to this comment https://github.com/IntelRealSense/librealsense/issues/4158#issuecomment-505559462). 

However, this gives me a single frameset for all images. I would like to split this frameset into two framesets, one per sensor, to be able to align the color frame to the depth frame per sensor (rs2::align::process requires a frameset).

Splitting frames by serial into e.g. arrays is no problem, however, re-creating two framesets seems difficult. Another theoretical option would be to clone the frameset, and remove (e.g. swap out) the frames from one sensor from each of the framesets.

Any suggestions how I could split the frameset into two, create new framesets from rs2::frame objects or align frames without a frameset?

A cumbersome solution seems to temporarily swap out the double depth/color frames (using rs2::frame::swap), so the alignment works, but I hope there is a better way.

MartyG-RealSense, 2021-06-02:
Hi @roelofvandijk  I should emphasize from the start that I do not have specialist knowledge of this particular qspect of librealsense programming, so thanks in advance for your patience as I work through the problem with you.

In a case in the link below, advice was given that if you need to listen to multiple sensors then you should use one of the following methods:

-  Create an rs2::syncer object (as you have already done) and pass it as a callback to multiple sensors; or 

-  Create a frame queue per sensor and poll each one.

https://github.com/IntelRealSense/librealsense/issues/1238#issuecomment-368793933

A RealSense user who was having problems with syncer instead created a C++ script with a technique that they described as using a frame queue with a custom bundler as a processing block.

https://support.intelrealsense.com/hc/en-us/community/posts/360037090133/comments/1500000449662

```
//Prepare frame-set getter (the frame queue!)
 rs2::frame_queue q(2);
 std::vector<rs2::frame> bundle;

  rs2::processing_block bundler([&](rs2::frame f, rs2::frame_source& src) {
    bundle.push_back(f);     //add frame to vector
    if (bundle.size() == 2){ //or however many when full
        auto fs = src.allocate_composite_frame(bundle);
        src.frame_ready(fs);
        bundle.clear();
     }
  });
  bundler.start(q); //when enough frames are loaded, the queue is ready

  depth_sensor.start([&](rs2::frame f){
    bundler.invoke(f);  //add frame to bundler
  });
  color_sensor.start([&](rs2::frame f){
    bundler.invoke(f);
  });
```

***************

If the above information is not helpful to your particular goals then please let me know and I will do my best to suggest other potential solutions.

roelofvandijk, 2021-06-03:
Hello @MartyG-RealSense, thank you for the quick reply, it is much appreciated! The links you shared were spot on, and I can split the frameset into two frameset per device using the custom processing block, thank you!

For posterity, the route is: 
1. Split single syncer frameset with synchronized frames from two sensors into two frame vectors by serial
2. Convert the frame vectors to framesets using the processing block from here: https://github.com/IntelRealSense/librealsense/issues/5847#issuecomment-586718261
3. Align each individual frameset.

MartyG-RealSense, 2021-06-03:
That's great to hear @roelofvandijk - thanks so much for sharing information about your method with the RealSense community!   :)

MartyG-RealSense, 2021-06-09:
Hi @roelofvandijk  Do you require further assistance with this case, please? Thanks!

roelofvandijk, 2021-06-09:
Hello @MartyG-RealSense, no thank you.

closed 2021-06-09 by roelofvandijk