realsenseai/librealsense #9146: D455 returns Bayer-like infrared images with Y16 format

newpavlov, 2021-06-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.13.50 | 
| Operating System & Version |   Debian 10 | 
| Kernel Version (Linux Only)    |  4.19.0-16                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.45.0                          | 
| Language                            |  Rust/C                       | 
| Segment			|  Robot | 

### Issue Description
When infrared images are captured with Y16 pixel format from a D455 sensor on maximum resolution (1280x800), I get images like this:
[image]

By zooming-in we can easily see a Bayer-like pattern. Strangely enough when the images get captured with Y8 format and the same resolution. The result can be also seen in `realsense-viewer`.

is this behavior intentional? if yes, can it be turned off?

newpavlov, 2021-06-03:
@MartyG-RealSense 
Please, read the issue text more carefully. My question is not about rectification, but about the Bayer-like pattern, i.e. about greatly varying measured intensity of neighboring pixels. Such effect was not present on D435 sensors.

Is this effect caused by applying a Bayer-like pattern of filters on the IR matrix level or can it be disabled somehow?

MartyG-RealSense, 2021-06-03:
It is important in my opinion to make clear the distinction between Y8 (rectified) and Y16 (unrectified) when considering the difference between these two formats.

I will refer your question above to @agrunnet

agrunnet, 2021-06-03:
Yes the unrectified rgb has a Bayer pattern. This is the 1280x800 resolution. The D455 has a color Bayer pattern filter. The d435 does not. 
You cannot turn it off. 
The other resolutions are rectified and have been de-Bayered so you won’t see the effect you mention on them.

MartyG-RealSense, 2021-06-03:
Thanks so much for the information @agrunnet

newpavlov, 2021-06-03:
@agrunnet 
Can you please provide more details about filters used in the Bayer pattern for IR matrices? Do they simply dampen incoming light to extend dynamic range or do they filter out certain spectral ranges? Is it correct to assume that the same Bayer pattern is also used in the [D450 modules](https://store.intelrealsense.com/buy-intel-realsense-depth-module-d450-and-d4-board-bundle.html)?

agrunnet, 2021-06-03:
The RGB Bayer pattern is a spectral filter red, green, green, blue. So it attenuates colors differently like a normal Bayer pattern would. It is the same in the D450 module.

newpavlov, 2021-06-03:
Let me clarify: are you telling me that **infrared** cameras in D455 have different spectral sensitivity for different pixels?? It's **very** surprising, to say the least. Can you please point me to any official documentation which would support this statement?

agrunnet, 2021-06-03:
That is correct. They are color cameras on the D455. The term “infrared” is more related to fact that we don’t block the inferred. Plus it is a legacy term when cameras used to be ONLY infrared. 
Our stereo algo uses all colors to help determine depth. The same is true for the D415 camera.

newpavlov, 2021-06-03:
Thank you! I guess we can close this issue then.

closed 2021-06-03 by newpavlov

agrunnet, 2021-06-03:
@newpavlov  I will see if we can do a better job in pointing this out in the documentation. We certainly don’t intend any confusion. Thanks for pointing it out.