realsenseai/librealsense #9147: Cannot detect camera devices (/dev/video*)

rkim87, 2021-06-02:
Hi,

I can't seem to detect any camera devices under /dev after I've installed the ros integrated library of librealsense. I tried sudo apt update/upgrade to update any potential drivers/firmwares and encountered the following:

```
rkim87@ME02L0464GRD09L:~$ sudo apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  librealsense2 librealsense2-dbg librealsense2-dev librealsense2-gl
  librealsense2-gl-dbg librealsense2-gl-dev librealsense2-net
  librealsense2-net-dbg librealsense2-net-dev librealsense2-udev-rules
  librealsense2-utils
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.

```

I tried manually installing one of the packages and encountered the following:

```
rkim87@ME02L0464GRD09L:~$ sudo apt install librealsense2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 librealsense2-dbg : Depends: librealsense2 (= 2.40.0-0~realsense0.3814) but 2.45.0-0~realsense0.4551 is to be installed
 librealsense2-dev : Depends: librealsense2 (= 2.40.0-0~realsense0.3814) but 2.45.0-0~realsense0.4551 is to be installed
 librealsense2-gl : Depends: librealsense2 (= 2.40.0-0~realsense0.3814) but 2.45.0-0~realsense0.4551 is to be installed
 librealsense2-net : Depends: librealsense2 (= 2.40.0-0~realsense0.3814) but 2.45.0-0~realsense0.4551 is to be installed
 librealsense2-utils : Depends: librealsense2 (= 2.40.0-0~realsense0.3814) but 2.45.0-0~realsense0.4551 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
```

Has anyone encountered this issue or have any suggestions on what I should do to resolve this issue?

MartyG-RealSense, 2021-06-03:
Hi @rkim87   Could you provide the following details please:

-  The RealSense camera model that you are using
-  The computer / computing device (PC, Mac, Jetson, Raspberry Pi, etc)
- The operating system (Windows, Ubuntu, Raspbian, etc)

rkim87, 2021-06-03:
@MartyG-RealSense 

Hi! I'm using the RealSense D435i camera with Dell Precision 5530/Dell Precision 3541 laptops on Ubuntu 18.04. I seem to have the same issue on both laptops.

When I remove all RealSense™ SDK-related packages with:
```
dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge
```
I am able to detect any usb-cameras that I connect to the laptops on the /dev/video* directories.

MartyG-RealSense, 2021-06-03:
I am reminded of a case in the link below, where USB cameras were no longer detected after installing librealsense from packages and there was no dev/video.  Uninstalling librealsense-dkms caused the USB cameras to work again.

https://github.com/IntelRealSense/librealsense/issues/7287

The investigation in that case indicated that the kernel was responsible, and installing **HWE** and kernel 4.15.0-120-generic resolved that RealSense user's problems.

https://github.com/IntelRealSense/librealsense/issues/7287#issuecomment-708479006

rkim87, 2021-06-03:
Booting to previous version of kernel solved this issue! Thanks so much for linking the solution! 😄

closed 2021-06-03 by rkim87

MartyG-RealSense, 2021-06-03:
Excellent news @rkim87  - thanks very much for the update!