realsenseai/librealsense #9148: librealsense load offline *.bag will crash

HelloCentaurus, 2021-06-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

HelloCentaurus, 2021-06-03:
When pipe is about to stop, some sensors  haven't stopped？

MartyG-RealSense, 2021-06-03:
Hi @HelloCentaurus  I would recommend preventing the bag from looping so that playback stops when the end frame of the bag is reached.  This is accomplished by inserting the instruction **repeat_playback=False** into the bracket of cfg.enable_device_from_file:

**cfg.enable_device_from_file(filename, repeat_playback=False);**

HelloCentaurus, 2021-06-03:
> Hi @MartyG-RealSense ,Thanks!
change my code to this below, can avoid this problem!

void test_playback_problem()
{

	string filename = "./20210602_rectmirror.bag";
	
	rs2::pipeline pipe;
	rs2::config cfg;

	cfg.enable_device_from_file(filename, false);
	auto profile = pipe.start(cfg); // Load from file

	bool ret=true;
	while(ret)
	{
		rs2::frameset frames;
		ret=pipe.try_wait_for_frames(&frames,500);
	}
		
	pipe.stop();

}

MartyG-RealSense, 2021-06-03:
Thanks very much @HelloCentaurus for sharing your solution with the RealSense community!

keithahern, 2021-06-03:
I was about to file this as a new bug but I think its the same.  Here's now to reproduce:

```
import pyrealsense2 as rs
import faulthandler; faulthandler.enable()

for i in range(0, 1000):
    print(i)
    config = rs.config()
    config.enable_device_from_file("test.bag")
    pipeline = rs.pipeline()
    pipeline.start(config)
    pipeline.wait_for_frames(timeout_ms=1000)
    pipeline.stop()
```
NOTES
repeat_playback=False does not fix
pipeline.try_wait_for_frames does not fix

It DOES work with a multi frame bag as saved from RealSense viewer, it DOES NOT work with a single frame bag as saved from the realsense api using rs.save_single_frameset().  I have attached a sample bag here.

For us it fails quite quickly.  This code is a simplified version of what we have in a big project which now fails within seconds of sequentially loading a dozen or so bag files.

[test.bag.zip](https://github.com/IntelRealSense/librealsense/files/6592946/test.bag.zip)

It fails on Linux, macOS and Windows.

macOS mostly points me to the following:
```Assertion failed: (m_is_started == false) librealsense/src/media/playback/playback_device.cpp, line 499```
but sometimes to:
```Assertion failed: (0), function librealsense/src/media/playback/playback_device.cpp, line 158```

MartyG-RealSense, 2021-06-09:
Hi @keithahern  Do you still require assistance with your own problem above, please? Thanks!

keithahern, 2021-06-09:
HI @MartyG-RealSense, yes I do.

HelloCentaurus, 2021-06-09:
@keithahern You have to just finish all frames。If counts of frames is larger than 1000, your must finish all frames,then stop.
I wirte this code below in C++,just for finish all frames.

/*this is my code*/

bool ret=true;
while(ret)
{
	rs2::frameset frames;
	ret=pipe.try_wait_for_frames(&frames,500);
}	
pipe.stop();

MartyG-RealSense, 2021-06-09:
Thanks very much @HelloCentaurus for offering help to @keithahern   :)

keithahern, 2021-06-10:
Thanks @HelloCentaurus.  It does seem to reduce the frequency of the crashes with each subsequent run, but not completely eliminate them.  I think its an artefact of the single frame bag file since it doesn't happen with multi-frame bag files. This change does seem to really slow things down too.

When I modify my code as follows:

```
import pyrealsense2 as rs
import faulthandler; faulthandler.enable()

bag_filename = "test.bag"

for i in range(0, 1000):
    print(i)
    config = rs.config()
    config.enable_device_from_file(bag_filename)
    pipeline = rs.pipeline()
    pipeline.start(config)
    ret = True
    while ret:
        ret, frame = pipeline.try_wait_for_frames(500)

    pipeline.stop()
```
Heres the output of running it 3 times.
```
% python bulk_bagfile_fail.py 
0
1
2
3
Assertion failed: (m_is_started == false), function operator(), file /Users/code/librealsense/src/media/playback/playback_device.cpp, line 499.
Fatal Python error: Aborted

Thread 0x0000000114438e00 (most recent call first):
  File "bulk_bagfile_fail.py", line 9 in <module>
zsh: abort      python bulk_bagfile_fail.py
% python bulk_bagfile_fail.py
0
1
2
3
...clipped..
17
18
19
Fatal Python error: Segmentation fault

Thread 0x0000000107a36e00 (most recent call first):
  File "bulk_bagfile_fail.py", line 9 in <module>
zsh: segmentation fault  python bulk_bagfile_fail.py

% python bulk_bagfile_fail.py
0
1
2
3
...clipped...
17
18
19
20
Assertion failed: (m_is_started == false), function operator(), file /Users/code/librealsense/src/media/playback/playback_device.cpp, line 499.
Fatal Python error: Aborted

Thread 0x000000011565ce00 (most recent call first):
  File "bulk_bagfile_fail.py", line 9 in <module>
zsh: abort      python bulk_bagfile_fail.py

```

MartyG-RealSense, 2021-06-10:
Hi @HelloCentaurus   Does the speed improve if you remove **500** from the wait_for_frames() bracket?

I would also suggest putting the **repeat_playback=False** instruction back into the bracket of **config.enable_device_from_file** to ensure that bag looping cannot take place.  Segmentation errors when reading bags have been associated in the past with the loop prevention instruction being absent.

keithahern, 2021-06-10:
@MartyG-RealSense - yes both those changes (repeat_playback=False  and removing 500) seem to have stabilised the code.  Both are needed.

The code does seem to be slow however, taking 3-4 seconds for each loop.

Ultimately we found a workaround for this by not saving single frame bag files, instead we converted single frame depth data to a numpy array and serialised it to disk as a pickled file.  Reading of these files, to estimate, is 50 times faster than the bag files.

MartyG-RealSense, 2021-06-10:
Thanks so much @keithahern for sharing the method that worked for you - great news that you had a positive outcome!

HelloCentaurus, 2021-06-11:
@keithahern I think I have understood the problem, but my English is too poor to express. I recommend the following code：

#code below
import pyrealsense2 as rs
import faulthandler; faulthandler.enable()

bag_filename = "test.bag"

ret = True
i=0
while ret:
    print(i)
    i=i+1
    config = rs.config()
    config.enable_device_from_file(bag_filename)
    pipeline = rs.pipeline()
    pipeline.start(config)
    ret, frame = pipeline.try_wait_for_frames(500)

pipeline.stop()

closed 2021-06-22 by MartyG-RealSense