realsenseai/librealsense #9149: ColorFrame data in Align to Depth

anhTuan0712, 2021-06-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 / D400 } | 
| Firmware Version                   | | 
| Operating System & Version |   {Win (8.1/10)   | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.43. }                          | 
| Language                            |  {C/C# }                          | 
| Segment			|                    | 

## Hello everyone!
### Issue Description

+ I am building project in c sharp language.
+ I used Align to depth (described here: https://dev.intelrealsense.com/docs/rs-align)
+ After align to Depth I get a smaller size RGB image, and around the image there is a black "BORDER".
[image]

###Question
1) Is there any "Option" in the API (realsense API) that can adjust the "BORDER" color above?
Thanks everyone.

anhTuan0712, 2021-06-03:
Hi @MartyG-RealSense , Hi @RealSenseCustomerSupport , Hi @RealSenseSupport.
Please confirm this problem for me !

RealSenseSupport, 2021-06-07:
@anhTuan0712 Sorry, we don't have such API to adjust this "BORDER" color.

anhTuan0712, 2021-06-07:
Hi @RealSenseSupport,
Thank you for your response. sorry, I want to ask one more question, is there any link between RGB size after align and point cloud.
e.g: x, y coordinates with edge or angle RGB

RealSenseSupport, 2021-06-10:
@anhTuan0712 Sorry, there's no such link as you mentioned.

RealSenseSupport, 2021-07-21:
@anhTuan0712 Any other questions about this? Looking forward to your reply. Thanks!

RealSenseSupport, 2021-07-28:
@anhTuan0712 Could you please update? Please note that the issue will be closed if we don't hear from you for another 7 days. Thanks!

RealSenseSupport, 2021-09-17:
@anhTuan0712 Sorry that we didn't hear from you for several weeks. Will close the ticket at this point. Please feel free to create new one if you still have any other questions or issues. Thanks!

closed 2021-09-17 by RealSenseSupport