realsenseai/librealsense #9150: Avoiding post-processing and smoothing of data

mocialov, 2021-06-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |   Windows 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |                           | 
| Language                            |  python                          | 
| Segment			|  others                          | 

### Issue Description
I am using the RealSense for inspection purposes and I need raw 3D data without any post-processing. Right now, I am looking at a patch on a wall from ~1m distance and I see 3D measuments fluctuate from frame to wave, which looks like a wave. Is this behaviour normal? I was expecting noise of different distribution

[image]

My setup is as follows (taken from examples):

```
import pyrealsense2 as rs
pipeline = rs.pipeline()
config = rs.config()

pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))
depth_sensor = device.first_depth_sensor()
depth_sensor.set_option(rs.option.emitter_enabled, 1.0)

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

if device_product_line == 'L500':
    print ('device L500')
    config.enable_stream(rs.stream.color, 960, 540, rs.format.bgr8, framerate=30)
else:
    print ('device not L500')
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, framerate=30)

profile = pipeline.start(config)

align_to = rs.stream.color
align = rs.align(align_to)

while True:
    frames = pipeline.wait_for_frames()
    aligned_frames = align.process(frames)
    aligned_depth_frame = aligned_frames.get_depth_frame()
    if not aligned_depth_frame or not color_frame:
        continue

    depth_image = np.asanyarray(aligned_depth_frame.get_data())
```

MartyG-RealSense, 2021-06-03:
Hi @mocialov  The optimal depth accuracy resolution of D455 is **848x480**, so it may aid the accuracy of your results if you are able to use that resolution instead of 640x480.

It looks as though the z-depth measurement on your chart is fluctuating between a range of 1100 and 1160 on average.  Is there stable air temperature in the location that the camera is being used, or can it change (for example, from 25 degrees C to 35 degrees C).

mocialov, 2021-06-03:
> Hi @mocialov The optimal depth accuracy resolution of D455 is **848x480**, so it may aid the accuracy of your results if you are able to use that resolution instead of 640x480.
> 
> It looks as though the z-depth measurement on your chart is fluctuating between a range of 1100 and 1160 on average. Is there stable temperature in the environment that the camera is being used in (the external air temperature, not the internal camera temperature), or can it change over time - for example, from 25 degrees C to 35 degrees C and back again.

@MartyG-RealSense the temperature is stable. I have changed the resolution as you suggested, but the results are very similar

mocialov, 2021-06-03:
@MartyG-RealSense 
[image]

The surface of the wall is quite "bumpy" with bumps of about 1-2mm deep

[image]

MartyG-RealSense, 2021-06-03:
There does not seem to be anything on that wall that would obviously contribute to image quality loss.  Having a bumpy surface could be a positive factor that provides more texture for the camera to analyze than a plain-colored flat wall would.

Do the fluctuations reduce if you move the camera further away from the wall by a small amount?

mocialov, 2021-06-03:
@MartyG-RealSense here I have plotted std of the min and max distances of 10 consecutive frames at 3 distances

[image]

MartyG-RealSense, 2021-06-03:
Does the chart signify that the measurements become less accurate as the camera is moved further away from the wall?

mocialov, 2021-06-03:
@MartyG-RealSense Yes, I would say so. We are moving further from the original question, I believe. I understand that the noise is going to be present. I am simply not sure whether the noise distribution that I am getting is the real noise or smoothed noise based on my initial GIF in the question

MartyG-RealSense, 2021-06-03:
Whilst the RealSense Viewer tool applies a range of post-processing filters by default, scripts that you create yourself have no post-processing and have to have the filters deliberately programmed into them.  So without filters, your application would be rendering the raw depth data.

mocialov, 2021-06-03:
@MartyG-RealSense this is what I wanted to either confirm or dismiss. Thank you for the feedback!

closed 2021-06-03 by mocialov

MartyG-RealSense, 2021-06-03:
You are very welcome @mocialov   :)