realsenseai/librealsense #9151: Intel Realsense and Flask on Raspberry pi4 - RuntimeError: failed to set power state

peppe89ct, 2021-06-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | | 
| Operating System & Version | Raspberry OS| 
| Kernel Version (Linux Only)    | 5.10.17-v7l+ | 
| Platform                                 | Raspberry Pi/ |
| SDK Version                            |                           | 
| Language                            | python | 
| Segment			|  Robot                         | 

### Issue Description
Hy all, i'am tryng to have the payload from antennas and the image from relasense d435i connected to a raspberry pi 4.

My code is the following:

```
import pyrealsense2.pyrealsense2 as rs
import cv2
import numpy as np
import datetime
from app import app
from flask import request,  jsonify
import io, json

resp = None
pipeline = None

#Gestiamo solo il POST per la ricezione dei Payloads
@app.route("/file-upload", methods=["POST"])
def upload_file():

    #if pipeline in None:
        #print("Create pipeline")
        #pipeline = create_pipeline()
    pipeline = rs.pipeline()

    # Streams
    cfg = rs.config()
    cfg.enable_stream(rs.stream.color,640 , 480, rs.format.bgr8, 30)

    pipeline.start(cfg)

    string = request.data
    string = string.decode('utf-8')
    frames = pipeline.wait_for_frames()
    color_frame = frames.get_color_frame()
    Current_Time = datetime.datetime.now()  # .time() or .date()
    timestampStr = Current_Time.strftime("%d-%b-%Y -- %H-%M-%S.%f")

    if not color_frame:
        print("Color frame non presente")

    color_image = np.asanyarray(color_frame.get_data())

    if string is not None:
        payload = json.loads(strinhttps://stackoverflow.com/questions/67790851/intel-realsense-and-flask-on-raspberry-pi4-runtimeerror-failed-to-set-power-sg)
    else:
        resp = jsonify({'message': "Il payload � vuoto"})
        resp.status_code = 404

    #Scriviamo sul disco il json
    try:
        with io.open("./DatasetProva/Json/" + timestampStr+'.json', 'w', encoding='utf8') as outfile:
            # Scriviamo leimmagini sul disco nel path
            str_ = json.dumps(payload,indent=4, separators=(',', ': '), ensure_ascii=False)
            outfile.write(str_)
            #i+=1
            resp = jsonify({'message': "E' stato salvato il json"})
            resp.status_code = 201
        cv2.imwrite("./DatasetProva/Immagini/" + timestampStr + ".jpg", color_image)

        return resp
    except Exception as ex:
        print("Errore sulla scrittura sul disco ", ex)
        #i-=1

    pipeline.stop()
    return resp

if __name__ == "__main__":
    app.run(debug=True, host ="192.168.1.191",  port="8080")

```
Whit this error:

```
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/flask/app.py", line 2309, in __call__
    return self.wsgi_app(environ, start_response)
  File "/usr/lib/python3/dist-packages/flask/app.py", line 2295, in wsgi_app
    response = self.handle_exception(e)
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1741, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/lib/python3/dist-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/usr/lib/python3/dist-packages/flask/app.py", line 2292, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1815, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1718, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/lib/python3/dist-packages/flask/_compat.py", line 35, in reraise
    raise value
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1813, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/lib/python3/dist-packages/flask/app.py", line 1799, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Desktop/CreateDataset/main.py", line 36, in upload_file
    pipeline.start(cfg)
RuntimeError: failed to set power state

  
```

When i run on terminal "rs-enumerate-devices" i see that when rs call the stream color i have this error:

``` 
    pi@raspberrypi:~ $ rs-enumerate-devices
 01/06 16:30:19,693 ERROR [3012822080] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
Could not create device - failed to set power state . Check SDK logs for details
 01/06 16:30:19,693 ERROR [3069284368] (sensor.cpp:526) acquire_power failed: failed to set power state

```
How can i solve the error?

(I runned on a raspberry pi3b too and the error is the same)

MartyG-RealSense, 2021-06-03:
Hi @peppe89ct   There is a previous case with this **failed to claim usb interface ... RS2_USB_STATUS_BUSY** error when using Flask.  That case was on an Nvidia Jetson board, which like Raspberry Pi is Arm-based.

https://github.com/IntelRealSense/librealsense/issues/6551

 A RealSense user in the link below shared information on the Flask approach that worked for them with a Raspberry Pi 4 and D435.

https://github.com/IntelRealSense/librealsense/issues/6475#issuecomment-635990864

peppe89ct, 2021-06-03:
> #6475 (comment)

Hi @MartyG-RealSense, thanks for the reply and for applying the labels :)
For the first  link that you gave me the author unfortunately didn't solve it.
For the second link i need to read better the project linked in the comment.
In all of this i didn't understand the cause of this problem, and why the runtime error is an error such as a power failure?
Furthermore, sorry for the ignorance, what is a pickable function?

Thank you so much for your patience.

MartyG-RealSense, 2021-06-03:
If you mean 'picklable', I recommend the Python tutorial about pickling in the link below.

https://www.datacamp.com/community/tutorials/pickle-python-tutorial

In regard to the **Failed to set power state** error, this can refer to an attempt to access a particular camera that fails for some reason.

MartyG-RealSense, 2021-06-14:
Hi @peppe89ct  Do you require further assistance with this case, please?  Thanks!

peppe89ct, 2021-06-14:
> Hi @peppe89ct Do you require further assistance with this case, please? Thanks!

No i solved, thanks :)

MartyG-RealSense, 2021-06-14:
That's great news @peppe89ct - thanks very much for the update!  As you had a successful outcome, I will close this case.  Thanks again!

closed 2021-06-14 by MartyG-RealSense