realsenseai/librealsense #9152: How to turn off auto-exposure and auto-white balance?

JimXu1989, 2021-06-03:
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17)  | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<45>.<0> }                          | 
| Language                            |  {python}                          | 

### Issue Description
I want to turn off auto exposure as well as auto white balance, I have tried the following way:
rs.option.enable_auto_exposure = False
rs.option.enable_auto_white_balance = False
But it seems does not work.

Thanks a Lot!!

MartyG-RealSense, 2021-06-03:
Hi @JimXu1989   The script reference in the link below should be helpful for disabling auto-exposure.  Set the sensor index number to '0' to disable depth auto-exposure and set it to '1' to disable color auto-exposure.

https://github.com/IntelRealSense/librealsense/issues/3558#issuecomment-548743478

In regard to disabling auto white balance in Python, I do not have an example script for this, though my understanding is that it is controlled in Python with the color sensor option **rs.option.enable_auto_white_balance** - which would be set to '0' to disable it.

JimXu1989, 2021-06-03:
> Hi @JimXu1989 The script reference in the link below should be helpful for disabling auto-exposure. Set the sensor index number to '0' to disable depth auto-exposure and set it to '1' to disable color auto-exposure.
> 
> [#3558 (comment)](https://github.com/IntelRealSense/librealsense/issues/3558#issuecomment-548743478)
> 
> In regard to disabling auto white balance in Python, I do not have an example script for this, though my understanding is that it is controlled in Python with the color sensor option **rs.option.enable_auto_white_balance** - which would be set to False to disable it.

Thanks a lot! It worked

closed 2021-06-03 by JimXu1989