realsenseai/librealsense #9156: Android: LIDAR L515: VGA and XGA options are not available

Shanki070, 2021-06-03:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.05.00 | 
| Operating System & Version |   Android 11 | 
| Platform                                 | Android  |
| SDK Version                            | 2.45.0 | 
| Language                            |  Java | 
| Segment			|  Smartphone                         | 

### Issue Description
In "Depth Resolution" for L515, the highest option available is XGA(1024x768). But in the android device, it is showing me **only one option, i.e. QVGA(320x240)**.

[image]

---------------------------------------------------------------------------------------------------

I am using cable USB-Type-C to USB-Type-C for a phone to L515 connection. But here it is showing **USB 2.1**

[image]

RealSenseSupport, 2021-06-07:
@Shanki070  Yes, only 320x240 is supported under USB2 mode. Please note that the USB connection type not only depends on the connection cable but also depends on the  capability of the typeC port on the phone. Please double check the cable and port. When you get USB3 connection, then you can get VGA and XGA options. Thanks!

RealSenseSupport, 2021-07-13:
@Shanki070 Did you get through this issue? Looking forward to your update. Thanks!

RealSenseSupport, 2021-07-21:
@Shanki070 Any update from your side? Please note that the issue will be closed if we don't hear from you for another 7 days. Thanks!

Shanki070, 2021-07-21:
Yes, this issue has been fixed after changing the cable to USB3 type.

RealSenseSupport, 2021-07-21:
@Shanki070 Great! Then close the issue accordingly. Thanks!

closed 2021-07-21 by RealSenseSupport