realsenseai/librealsense #9157: Limitations of the RSUSB backend

puzzlepaint, 2021-06-03:
In the documentation on the RSUSB backend, it is mentioned that it has some limitations. For example, [this page](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md) states:

> As a general rule it is recommended to use the native kernel drivers, especially in production environment. The second method [RSUSB] augments the native installation and allows for a fully-functional SDK deployment at the expense of certain performance and functional limitations (e.g. multi-cam).

Similarly, the documentation on the old FORCE_LIBUVC flag [here](https://dev.intelrealsense.com/docs/build-configuration), which now activates the RSUSB backend, mentions that "it has well known limitations".

Despite supposedly being well known, unfortunately I could not find a list of these limitations. Maybe I am overlooking it? Or if there is no such list, would it be possible to set one up?

The reason I am asking is that I have a case where something works with the V4L2 backend, but not with the RSUSB backend, and I don't know whether that is a bug in the RSUSB backend or a known limitation. Specifically, with RSUSB I get an error "failed to set power state" when trying to change camera settings such as exposure time while the camera is currently streaming, while it works with V4L2. However, the V4L2 backend has the disadvantage that it sometimes drops frames, while the RSUSB backend seems to deliver them more reliably. Thus I would prefer using RSUSB.

MartyG-RealSense, 2021-06-03:
Hi @puzzlepaint   For a comparison of RSUSB backend (previously also known as libuvc backend) with a patch-based librealsense build, please visit the link below and scroll down through the linked-to comment to the section headed **What are the advantages and disadvantages of using libuvc vs patched kernel modules?**

https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604

puzzlepaint, 2021-06-03:
Thank you for the quick and helpful reply! This resolves the question for me, but given that the information seems to be buried somewhere in a GitHub issue conversation, it seems to be hard to find in general. Maybe this could be described or linked for example [in the list of config options](https://dev.intelrealsense.com/docs/build-configuration) or [in the libuvc installation instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md)? Apart from that, this issue can be closed, thanks again.

MartyG-RealSense, 2021-06-04:
You are very welcome.  I have added a **Documentation** tag to this case so that your suggestion for official documentation on the subject of RSUSB vs patched kernels can be considered by the RealSense team at a future date.  This case should be kept open in the meantime, though you do not need to do anything further yourself.  Thanks!

puzzlepaint, 2021-06-04:
I would just like to add that I found the solution to my specific issue I mentioned above:

> Specifically, with RSUSB I get an error "failed to set power state" when trying to change camera settings such as exposure time while the camera is currently streaming, while it works with V4L2.

It was not a limitation of the RSUSB backend, but an issue with my code. The code queried the librealsense device, started streaming, and then continued to use the device object queried at the start, using code like this:
```c++
rs2::context ctx;

rs2::device_list devices = ctx.query_devices();
rs2::device device = devices.front();

std::unique_ptr<rs2::color_sensor> colorSensor;
colorSensor.reset(new rs2::color_sensor(device.first<rs2::color_sensor>()));

rs2::config cfg;
// Some code to set up cfg here ...

rs2::pipeline pipeline;
pipeline.start(cfg);

// At this point, trying to do something like:
colorSensor->set_option(RS2_OPTION_EXPOSURE, 10 * colorExposureTimeMS);
// ... works with the V4L2 backend, but causes "failed to set power state" with the RSUSB backend.
```

The issue with this is that apparently, starting the pipeline internally creates new `uvc_sensor` objects with the RSUSB backend, and trying to use the old `uvc_sensor` objects afterwards conflicts with the new ones. The correct approach is to re-query the objects from the `rs2::pipeline_profile` after starting streaming, by modifying the code to start the pipeline like this:

```c++
rs2::pipeline pipeline;
rs2::pipeline_profile pipeProfile = pipeline.start(cfg);

// Re-query the device and sensor objects after starting the pipeline
device = pipeProfile.get_device();
colorSensor.reset(new rs2::color_sensor(device.first<rs2::color_sensor>()));
```

With this, everything seems to work as expected.

MartyG-RealSense, 2021-06-04:
Excellent news that you found a confirmed solution to your project's problem @puzzlepaint    :)

ev-mp, 2021-07-07:
@puzzlepaint hello, having multiple instances to the same (shared) resource is the fundamental limitation with RSUSB backend. so indeed after opening the `pipeline` , the original handles are being invalidated.
Additionally, while the originally-posted code is working with V4L driver's it is still not correct to use it that way as it may introduce side-effect on GW-Host sync level, and thus the approach you've implement (to re-acquire the sensor's handle after opening `pipe`) is the proper one not only for RSUSB but also for native (V4L/MediaFoundation) backends.

puzzlepaint, 2021-07-07:
Thanks for confirming that this is the right approach. It is however extremely easy for users of librealsense to do this incorrectly, since it appears to be working just fine with the V4L2 backend, and the need to re-acquire the handles is thus very easily overlooked.

From my point of view, it would thus be strongly advisable to:
* Document this very clearly, with a clear warning, and/or
* Print verbose, descriptive warnings at runtime if multiple handles to the same resource are attempted to be used (even with the V4L2 backend), and/or
* Change the library design such that opening a pipeline does not invalidate the original handles.

hg-zt, 2021-12-03:
@MartyG-RealSense Thanks for your explanation. My understanding is that RSUSB backend on Jetson devices does not provide extended frame metadata and hardware timestamp (all jetpack versions are based on kernel v4.9).

Could you elaborate a bit more on "performance"? As you said in #5212

> The kernels are fast and (mostly) stable. Adding patched modules into kernel tree in controlled and pier-reviewed manner allows to get most of the benefits and also get the additional features.

and https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md#3-choose-librealsense-sdk-backend
> ... at the expense of certain performance and functional limitations

How is native backend faster than RSUSB backend (less memory copies / lower latency / less cpu usage)? How did you do the benchmark / test?

Because deploying patched L4T for jetson devices is painful and harder to maintain, I want to know if it worth the effort.

MartyG-RealSense, 2021-12-03:
Hi @hg-zt  The advice about libuvc vs patched kernels in https://github.com/IntelRealSense/librealsense/issues/5212 was written by RealSense team member @ev-mp (who also provides advice in this case), who has specialist knowledge of that subject that I unfortunately do not have. 

An RSUSB build of librealsense includes support for hardware metadata.  It is not dependent on Linux versions or kernel versions and does not require patching.

MartyG-RealSense, 2026-10-23:
This open GitHub issue is being closed due to its significant age.

closed 2026-10-23 by MartyG-RealSense