realsenseai/librealsense #9158: D435 camera won't publish infra2 topic

Russ76, 2021-06-04:
An update or a wrong setting seems to have stopped infra2 ROS topic from being published. This is to work with rtabmap software. At first the camera worked perfectly with the mapping software, with infra1 and infra2 stereo image streams. Now I can't get camera to produce infra2 at any time. Using the parameters I can reliably turn on and off the infra1 steam, but infra2 is gone! Is this from a software or firmware update? I am using the Nvidia Jetson TX1 board, Ubuntu 18.04, and ROS Melodic.

MartyG-RealSense, 2021-06-04:
Hi @Russ76  Infra2 would be unavailable if the camera was on a USB 2 connection, as that mode requires USB 3.  If your camera is plugged into a USB 3 port, could you check please the USB type that your ROS launch process is detecting by checking the **Device USB type** line of the launch log, as described in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1907#issuecomment-851961352

Russ76, 2021-06-04:
[ INFO] [1622814592.984338256]: Device with name Intel RealSense D435 was found.
[ INFO] [1622814592.985620981]: Device with port number 1-3 was found.
[ INFO] [1622814592.985719524]: Device USB type: 2.1
[ WARN] [1622814592.985786712]: Device 819112071357 is connected using a 2.1 port. Reduced performance is expected.

Yes, this indicates a problem. The port is type 3, I understand. At some times I have used a hub so that the keyboard and mouse may be also connected. That device says it is type 3. Somewhere the type number got changed, for it was working with both stereo streams on an earlier day. Is this a ROS problem or a Linux problem?

MartyG-RealSense, 2021-06-04:
If you launch the RealSense Viewer program at look at the number beside the camera name in the options side-panel (2.1 or 3.2), this should indicate whether the detection issue is only in the ROS wrapper or in both RealSense ROS and librealsense.

Russ76, 2021-06-04:
Realsense viewer reports USB port is type 2.1

It also reports this error at startup:

Timestamp: 1622821858845.864258
Severity: Warn
Description: Multiple realsense udev-rules were found! :
1:/etc/udev/rules.d/99-realsense-libusb.rules
2: /lib/udev/rules.d/60-librealsense2-udev-rules.rules
Make sure to remove redundancies!

MartyG-RealSense, 2021-06-04:
The **Multiple realsense udev-rules were found** warning typically occurs if librealsense has been built with both the source-code and Debian packages on the same computer.

If this is the case, the warning should disappear if the version of the rules in the **etc/udev** folder is removed.  

https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605687547

https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605662460

The rule in the **lib/udev** folder is the one that should not be deleted, as doing so may break linkages to Debian packages.

Russ76, 2021-06-04:
OK, thanks. Is there any way to correct this USB misidentification?

MartyG-RealSense, 2021-06-04:
I would recommend first dealing with the multiple udev rule warning to see if that resolves the USB 2.1 issue.

You could also try adding the command **initial_reset:=true** to the end of your ROS roslaunch instruction to reset the camera at launch if you are not doing so already.  

A camera reset can also be performed in the Viewer by going to the **More** option near the top of the side-panel and selecting **Reset Hardware** from the drop-down menu.

Russ76, 2021-06-04:
I have done all you suggested. Even installed another USB 3.0 powered hub. ROS and Realsense-viewer still report port is USB 2.1.
These errors come up repeatedly:
 04/06 12:00:05,665 WARNING [547331498368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/06 12:00:05,751 WARNING [547331498368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

MartyG-RealSense, 2021-06-04:
If the **control_transfer returned** warning generates continously then it can indicate that there is a serious communication problem with the device.

We can test whether the USB system on that particular computer is being over-burdened by launching ROS with a custom camera configuration that only streams low-resolution depth.

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30.0 initial_reset:=true**

Russ76, 2021-06-04:
$ roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30.0 initial_reset:=true
... logging to /home/thomas/.ros/log/ab5e3b92-c55c-11eb-ada1-00044b79c483/roslaunch-thomas-tx1-14685.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: while processing /opt/ros/melodic/share/realsense2_camera/launch/includes/nodelet.launch.xml:
Invalid <node> tag: Invalid <param> tag: invalid literal for int() with base 10: '30.0'. 

Param xml is <param name="depth_fps" type="int" value="$(arg depth_fps)"/>. 

Node xml is <node args="load realsense2_camera/RealSenseNodeFactory $(arg manager)" name="realsense2_camera" pkg="nodelet" required="$(arg required)" type="nodelet">
    <param name="serial_no" type="str" value="$(arg serial_no)"/>
    <param name="usb_port_id" type="str" value="$(arg usb_port_id)"/>
    <param name="device_type" type="str" value="$(arg device_type)"/>
    <param name="json_file_path" type="str" value="$(arg json_file_path)"/>
    <param name="rosbag_filename" type="str" value="$(arg rosbag_filename)"/>

    <param name="enable_pointcloud" type="bool" value="$(arg enable_pointcloud)"/>
    <param name="pointcloud_texture_stream" type="str" value="$(arg pointcloud_texture_stream)"/>
    <param name="pointcloud_texture_index" type="int" value="$(arg pointcloud_texture_index)"/>
    <param name="allow_no_texture_points" type="bool" value="$(arg allow_no_texture_points)"/>
    <param name="ordered_pc" type="bool" value="$(arg ordered_pc)"/>

    <param name="enable_sync" type="bool" value="$(arg enable_sync)"/>
    <param name="align_depth" type="bool" value="$(arg align_depth)"/>

    <param name="fisheye_width" type="int" value="$(arg fisheye_width)"/>
    <param name="fisheye_height" type="int" value="$(arg fisheye_height)"/>
    <param name="enable_fisheye" type="bool" value="$(arg enable_fisheye)"/>
    <param name="enable_fisheye1" type="bool" value="$(arg enable_fisheye1)"/>
    <param name="enable_fisheye2" type="bool" value="$(arg enable_fisheye2)"/>

    <param name="depth_width" type="int" value="$(arg depth_width)"/>
    <param name="depth_height" type="int" value="$(arg depth_height)"/>
    <param name="enable_depth" type="bool" value="$(arg enable_depth)"/>

    <param name="confidence_width" type="int" value="$(arg confidence_width)"/>
    <param name="confidence_height" type="int" value="$(arg confidence_height)"/>
    <param name="enable_confidence" type="bool" value="$(arg enable_confidence)"/>
    <param name="confidence_fps" type="int" value="$(arg confidence_fps)"/>

    <param name="color_width" type="int" value="$(arg color_width)"/>
    <param name="color_height" type="int" value="$(arg color_height)"/>
    <param name="enable_color" type="bool" value="$(arg enable_color)"/>

    <param name="infra_width" type="int" value="$(arg infra_width)"/>
    <param name="infra_height" type="int" value="$(arg infra_height)"/>
    <param name="enable_infra" type="bool" value="$(arg enable_infra)"/>
    <param name="enable_infra1" type="bool" value="$(arg enable_infra1)"/>
    <param name="enable_infra2" type="bool" value="$(arg enable_infra2)"/>
    <param name="infra_rgb" type="bool" value="$(arg infra_rgb)"/>

    <param name="fisheye_fps" type="int" value="$(arg fisheye_fps)"/>
    <param name="depth_fps" type="int" value="$(arg depth_fps)"/>
    <param name="infra_fps" type="int" value="$(arg infra_fps)"/>
    <param name="color_fps" type="int" value="$(arg color_fps)"/>
    <param name="gyro_fps" type="int" value="$(arg gyro_fps)"/>
    <param name="accel_fps" type="int" value="$(arg accel_fps)"/>
    <param name="enable_gyro" type="bool" value="$(arg enable_gyro)"/>
    <param name="enable_accel" type="bool" value="$(arg enable_accel)"/>
    <param name="enable_pose" type="bool" value="$(arg enable_pose)"/>

    <param name="base_frame_id" type="str" value="$(arg base_frame_id)"/>
    <param name="depth_frame_id" type="str" value="$(arg depth_frame_id)"/>
    <param name="infra1_frame_id" type="str" value="$(arg infra1_frame_id)"/>
    <param name="infra2_frame_id" type="str" value="$(arg infra2_frame_id)"/>
    <param name="color_frame_id" type="str" value="$(arg color_frame_id)"/>
    <param name="fisheye_frame_id" type="str" value="$(arg fisheye_frame_id)"/>
    <param name="fisheye1_frame_id" type="str" value="$(arg fisheye1_frame_id)"/>
[... 19 lines cut ...]
    <param name="aligned_depth_to_infra2_frame_id" type="str" value="$(arg aligned_depth_to_infra2_frame_id)"/>
    <param name="aligned_depth_to_fisheye_frame_id" type="str" value="$(arg aligned_depth_to_fisheye_frame_id)"/>
    <param name="aligned_depth_to_fisheye1_frame_id" type="str" value="$(arg aligned_depth_to_fisheye1_frame_id)"/>
    <param name="aligned_depth_to_fisheye2_frame_id" type="str" value="$(arg aligned_depth_to_fisheye2_frame_id)"/>

    <param name="publish_tf" type="bool" value="$(arg publish_tf)"/>
    <param name="tf_publish_rate" type="double" value="$(arg tf_publish_rate)"/>

    <param name="odom_frame_id" type="str" value="$(arg odom_frame_id)"/>
    <param name="topic_odom_in" type="str" value="$(arg topic_odom_in)"/>
    <param name="calib_odom_file" type="str" value="$(arg calib_odom_file)"/>
    <param name="publish_odom_tf" type="bool" value="$(arg publish_odom_tf)"/>
[... cut ...]
    <param name="stereo_module/gain/2" type="int" value="$(arg stereo_module/gain/2)"/>

    <param name="filters" type="str" value="$(arg filters)"/>
    <param name="clip_distance" type="double" value="$(arg clip_distance)"/>
    <param name="linear_accel_cov" type="double" value="$(arg linear_accel_cov)"/>
    <param name="initial_reset" type="bool" value="$(arg initial_reset)"/>
    <param name="unite_imu_method" type="str" value="$(arg unite_imu_method)"/>

  </node>
The traceback for the exception was written to the log file

Russ76, 2021-06-04:
$ roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 initial_reset:=true
... logging to /home/thomas/.ros/log/1bb7ca52-c55d-11eb-8907-00044b79c483/roslaunch-thomas-tx1-14894.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://thomas-tx1:41437/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 87 lines cut ...]
[ INFO] [1622828936.090110560]: color stream is enabled - width: 640, height: 480, fps: 15, Format: RGB8
[ INFO] [1622828936.090288630]: setupPublishers...
[ INFO] [1622828936.103773761]: Expected frequency for depth = 30.00000
[ INFO] [1622828936.240078736]: Expected frequency for color = 15.00000
[ INFO] [1622828936.354829972]: setupStreams...
[ INFO] [1622828936.380543119]: insert Depth to Stereo Module
[ INFO] [1622828936.381308626]: insert Color to RGB Camera
 04/06 12:48:56,458 WARNING [547185447296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1622828936.509927381]: SELECTED BASE:Depth, 0
[ INFO] [1622828936.550762401]: RealSense Node Is Up!
[ WARN] [1622828936.598675891]: 
 04/06 12:48:56,605 WARNING [547185447296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/06 12:48:56,656 WARNING [547185447296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/06 12:48:56,707 WARNING [547185447296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/06 12:48:56,859 WARNING [547185447296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/06 12:48:56,962 WARNING [547185447296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/06 12:48:57,015 WARNING [547185447296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 04/06 12:48:57,081 WARNING [547185447296] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2021-06-04:
Can you run it again with **depth_fps:=30** instead of 30.0 please?  Sorry about that!

Russ76, 2021-06-04:
Yes, the second one listed above was with number corrected.

MartyG-RealSense, 2021-06-04:
It is unusual that a depth-only configuration was defined in the link below:

**roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 initial_reset:=true**

But in the launch log, the color stream seems to be being published too at 640x480 and 15 FPS.

Please add **enable_color:=false** to the above roslaunch instruction to see whether that provides a depth-only launch.

Russ76, 2021-06-04:
thomas@thomas-tx1:~$ roslaunch realsense2_camera rs_camera.launch depth_width:=640 depth_height:=480 depth_fps:=30 initial_reset:=true enable_color:=false
... logging to /home/thomas/.ros/log/79474998-c564-11eb-9d8a-00044b79c483/roslaunch-thomas-tx1-19931.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://thomas-tx1:39687/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: False
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: -1
 * /camera/realsense2_camera/gyro_fps: -1
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 110 lines cut ...]

rostopic list
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/image_rect_raw/compressed
/camera/depth/image_rect_raw/compressed/parameter_descriptions
/camera/depth/image_rect_raw/compressed/parameter_updates
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/camera/depth/image_rect_raw/compressedDepth/parameter_updates
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/theora/parameter_descriptions
/camera/depth/image_rect_raw/theora/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static

Russ76, 2021-06-04:
Stop the presses! It was my cable. It was not 3.0 type. I'm sorry, I didn't know to watch for that. A number of my cables are type 2.0 even though they are new. It must have the blue plastic inside, then it has more pins for the connections. My bad

MartyG-RealSense, 2021-06-04:
Which firmware driver version is your D435 camera using please?

If you do not know the firmware version, you can find it in the RealSense Viewer by left-clicking on the **Info** option near the top of the options side-panel

Russ76, 2021-06-04:
05.12.13.50

MartyG-RealSense, 2021-06-04:
Okay, that firmware is correct for SDK 2.45.0.  Thanks very much.

Please do post an update when you are able to test with a USB 3 cable.  Good luck!

Russ76, 2021-06-04:
Yes, I had one 3.0 cable, and there is the topic infra2, as before! Sorry, I must have missed the memo about blue plugs on USB 3.0 cables. There are more wires inside! Of course the others can't work. I thought the difference was only in the jacks. Thanks for your patient help. I now have 5 new cables on order!

MartyG-RealSense, 2021-06-04:
You are very welcome - I'm pleased that you had a successful outcome. :)

Russ76, 2021-06-04:
This issue can bite a fellow. In fact, in an article on the _Intel website_, there is a link to buy a cable for this _very camera,_ and the cable is not type 3.0
https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

closed 2021-06-04 by Russ76