realsenseai/librealsense #9160: Wrongful access of opened USB handles results in "failed to set power state"

NOx-z, 2021-06-04:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 05.12.13.50 | 
| Operating System & Version | Win7  | 
| Platform                                 | PC  |
| SDK Version                            |  2.45.0  | 
| Language                            |  {C++  | 

"failed to set power state"
This was a deep dive of ~3 days.. because it wasn't working...
This problem has been brought up several times before and wrongly addressed. Notably one solution was to initialize a second detected camera before the first, and although that is a valid workaround the problem persisted.

Code to reproduce the problem:
```
        rs2::config cfg;
        rs2::pipeline pipe(ctx);

        // The color stream is the only setting that will pass, this will be explored in the description below further
        //cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);

        // All of these others will fail
        //cfg.enable_stream(RS2_STREAM_INFRARED, 1, 1280, 720, RS2_FORMAT_Y8, 30);
        cfg.enable_stream(RS2_STREAM_INFRARED, 2, 1280, 720, RS2_FORMAT_Y8, 30);
        //cfg.enable_stream(RS2_STREAM_DEPTH, 0, 1280, 720, RS2_FORMAT_Z16, 30);
        //cfg.enable_all_streams();
        
        auto profile = cfg.resolve(pipe);
        profile = pipe.start(cfg);
        
        auto devices = ctx.query_devices();
        rs2::device rsdevice = devices[0]; // crashes on the access (because of an init) of the element
```

The root cause of it, at least in Windows (and specifically with RSUSB??), is that the File Handle of the USB is already opened which causes an Access Violation, but it is not properly handled downstream..

All [depth] devices (in my case, the d435_device) inherit ds5_device and appear to add a sensor, followed by initing..
```
[constructor]
    {
        _depth_device_idx = add_sensor(create_depth_device(ctx, group.uvc_devices));
        init(ctx, group);
    }
```
We follow std::shared_ptr<synthetic_sensor> ds5_device::create_depth_device and come acroos..
```
        for (auto&& info : filter_by_mi(all_device_infos, 0)) // Filter just mi=0, DEPTH
            depth_devices.push_back(backend.create_uvc_device(info));
```
Hereafter the issue is actually handled..
traversing through
rs_backend::create_uvc_device
If the return of create_rsuvc_device is null, then return a nullptr.. but the fact that in create_depth_device we push the nullptr into the list without checking is a fault.
But if we continue to look through create_uvc_device -> create_rsuvc_device, again there is a check for (essentially for not null) and will return the nullptr.. But let's dive deeper..
working through more we go through usb_enumerator::create_usb_device(usb_info)
Here we have our first safety net,
```
            try
            {
                return std::make_shared<usb_device_winusb>(info, devices_path);
            }
            catch (std::exception e)
            {
                LOG_WARNING("failed to create usb device: " << info.id << ", error: " << e.what());
            }
            return nullptr;
```
I have never enabled the logging - but stepping through debug this is certainly caught and reported correctly as an Access Violation, because we are attempting to open up a File Handle to a USB device that is already open..

One more layer.. usb_device_winusb::usb_device_winusb calls query_device_interfaces..
```
            handle_winusb handle;
            auto sts = handle.open(path);
            if (sts != RS2_USB_STATUS_SUCCESS)
                throw winapi_error("WinUsb action failed, " + GetLastError());
```
We are clearly throwing an error here..
the problem, as stated before is that as all of these nullptr's are returned we push back a null uvc_device into depth_devices - so it has a size of 1, but upon accessing it (attempting to power on and send a command) it will fail. This access starts from ds5_device::init -> get_raw_depth_sensor -> get_sensor(_depth_device_idx=0) so it assumes there is a depth sensor, attempts to access it by first powering it on - and this is where it breaks for everyone..

My temporary fixes reside in create_depth_device & the ds5_device constructor, where I check if the returned uvc_dev or depth sensor is nullptr.. Currently I am just passing a non-init'd d435_device around, but I wrap the code that may access it (such as rs2::device rsdevice = ctx.query_devices()[0]; ) in a try/catch - but this is not very elegant and more proper solution by someone more familiar in the code base is requested.

Simply wrapping a try/catch without modifications to the sdk/DLL does not work.

The color channel has no effect, why? Well, because I think these are "depth" cameras so the software always looks for access at depth[0]. 

I am unsure if methods other than RSUSB have this issue, and based on other solutions to this similar issue I would like to say it doesn't, but I have not checked it.

Please let me know if any additional information is requested, but I believe this was enough of a deep-dive after only having the cameras for one week.

MartyG-RealSense, 2021-06-04:
Hi @NOx-z   Windows 7 may not be the most suitable platform for a multiple camera project, as the RealSense SDK installation documentation page for Windows 7 in the link below states in its **Known Limitations** section that "multiple cameras are likely not to work at the moment (not validated)".

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_win7.md#known-limitations

RSUSB in general is suited to single-camera applications rather than multiple-camera projects.

NOx-z, 2021-06-04:
Thanks for the quick reply, Marty..

Upgrading to Win10 has been on my To Do List for a while as Win7 is EOL.. and I am aware issues may be remediated upon upgrading.

The example I give and tested on is with a single camera.. Although my hacks do enable multiple infrared/depths to work as expected (beyond performance issues, they do function), the problem of pushing a nullptr into the depth_devices still exists in the current codebase for uvc_device.

MartyG-RealSense, 2021-06-04:
There was a past case involving a RealSense user who reported problems with nullptr when using RSUSB, and they stated that it did not occur when RSUSB was off.

https://github.com/IntelRealSense/librealsense/issues/7860

They submitted a pull-request fix but it has not been merged into the SDK yet as it is awaiting review.

https://github.com/IntelRealSense/librealsense/pull/8046

NOx-z, 2021-06-04:
I, admittedly, did not search with the new information I uncovered from the deep dive (after the "failed to set power state" which was my initial search terms).. And this is an exact duplicate, including all the supporting information (I wish I had found that thread before)

As stated by MojamojaK, it is still an active bug in the code.. although I will be upgrading in the near future and hopefully not need RSUSB which is a valid work around.

MartyG-RealSense, 2021-06-04:
Thanks very much @NOx-z - I will highlight the pull request to Intel.

MartyG-RealSense, 2021-06-16:
Hi @NOx-z   As I have already highlighted the pull request https://github.com/IntelRealSense/librealsense/pull/8046 to Intel, do you require further assistance with this case please?  Thanks!

MartyG-RealSense, 2021-06-22:
Case closed due to no further comments received.

closed 2021-06-22 by MartyG-RealSense

NOx-z, 2021-06-22:
Apologies.. I was out of town for a few days then I did a fresh install of win10 yesterday and I am working towards getting my system set up again.. But considering I am now using windows 10 because windows 7 is EOL I dont believe I this issue has much merit, for me or anyone else.

MartyG-RealSense, 2021-06-22:
No problem at all, @NOx-z - thanks very much for the update!