realsenseai/librealsense #9166: Get texture co-ordinates from new depthmap and point cloud

jaimeet64, 2021-06-06:
Hi there.

I am using Intelrealsense D435 for a robotics application.  Lets's say, I have RGB, Depth-map, and Point-Cloud (basically depth vertexes) for a single capture of a flat surface (for eg wall). When I get a texture map corresponding to the single capture, texture coordinates corresponding to holes in the depth map would be zero which is quite obvious.

But when I project all the depth points to a plane I will not have any missing depth point or depth with zero value. Now, given a new depth map and depth vertexes (point cloud), how can I regenerate the texture map?

MartyG-RealSense, 2021-06-06:
Hi @jaimeet64   It sounds as though you are initially creating a **textured point cloud** (RGB coordinates mapped to depth coordinates on the 3D point cloud).

When you say that you are projecting all the depth points to a plane, do you mean that you are converting the point cloud's 3D 'world' points to 2D 'pixel' coordinates (perhaps using a function such as **rs2_project_point_to_pixel**) and therefore losing the ability to analyze z-depth values and find which ones are zero as xyz becomes xy?

If the above description is correct, could you provide information please about what you are trying to achieve with the zero-depth or missing pixels.  Are you trying to identify where there are holes, or to fill the holes in?

jaimeet64, 2021-06-06:
HI @MartyG-RealSense, 

Thank you for your quick response.  Let me clarify your doubts about projecting Point cloud to a plane. 

Assume that D435 is fronto-parallel to the wall which is at a Z distance of 0.8 m. Lets me show you an example of how four 3D vertexes of Intel's real sense are projected to a plane. 

(X,Y,Z) Intel realsense point cloud  -> (X,Y,Z)  proejcted to a plane

i)  (0.018,-0.030,0.834) -> (0.019, -0.031, 0.8)   
ii) (0.045,0.014,0.791) ->  (0.044, 0.013, 0.8)  
iii)  (-0.011, 0.028, 0.821) -> (0.019, 0.027, 0.8)
iv)  (0.078,  0.011,  0) - > (0.079, 0.010, 0.8)  // 4th point does not have Z value as the depth not generated from Intel Realsense. 
                                                                     Point generated on the Plane is calculated using Depth Intrinsics. 

As you may notice, all the points projected to a plane have varying X & Y values, but the Z value is constant as the plane is front-parallel. 

Hence, my question is how to re-calculate Texture coordinates for the newly generated 3D points (for example 4th point in the example shown above)? 

FYI, my understanding is that texture coordinates are calculated from RGB image and Point cloud. Correct me if I am missing something.

MartyG-RealSense, 2021-06-06:
Thank you for the additional information.  I am not familiar with the term **fronto-parallel**.  From researching the term, it appears to relate to how the human eye-line looks at a scene.  So would it be accurate to say that the camera is facing a wall at human eye height?

jaimeet64, 2021-06-06:
Yes, you are right. 

Front-o-parallel w.r.t to camera center meaning, assume a rectangle that can be drawn on the plane and let the corner vertex be (X,Y,0),(-X,Y,0),(X,-Y,0),(-X,-Y,0) ; where X and Y are constant and these points are in the plane coordinate system.

 By fronto-parallel, I meant to say that all the four vertexes will have the same distance from the camera center and the normal of the plane estimated will pass the camera center and the center of the rectangle.
 
 I hope this helps.

MartyG-RealSense, 2021-06-06:
Okay, I can visualize the positive and negative XY vertices of your rectangle if thinking of them in terms of the camera's coordinate system (where the origin is the centerline of the left imager).   So when you are projecting the depth points to a plane, you are referencing the coodinates in terms of their position left/right and up/down from the center of the rectangle?  So a coordinate that is to the left and above the center may be (-X, -Y, 0)

Thanks very much for your patience as I visualize your project.

jaimeet64, 2021-06-06:
@MartyG-RealSense  Thank you for spending time over the weekend to visualize my project.

Yes, you are right. Point (-X. -Y, 0) would be the top left of the rectangle  (in the Plane coordinate system), and the same point if visualized from the camera coordinate system, it would be somewhat like (-0.023, -0.078, 0.8).   

Please let me know if you have more questions.

MartyG-RealSense, 2021-06-06:
You are very welcome!

If you are converting a 3D point cloud into a 2D plane, could you describe the method that you are using to generate the plane please and post your script code in a comment if possible.  Thanks very much.

jaimeet64, 2021-06-07:
@MartyG-RealSense  My question is not in converting the 3D point cloud to the 2D plane, but I am interested in converting Color Texture for the newly generated Point Cloud from the Plane. Can you please point me to a documentation or example script ? 

Thanks

MartyG-RealSense, 2021-06-07:
Instead of aligning depth to color, you could possibly use the less common technique of **color to depth** alignment by applying a UV map obtained from texture mapping.  This is described in the section of Intel's **Projection, Texture-Mapping and Occlusion** white-paper document linked to below.

https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras#section-3-3-the-uv-map

jaimeet64, 2021-06-07:
Thanks for sharing the links. I have already gone through the above mentioned link and was not quite useful except rs-measure. Please share rs-measure example python as I was not able to find one. 

I will reiterate my question so that it's clear about what I am looking for. 

I have RGB, Depth-map and Point cloud from intel real sense capture. Intelrealsense also provides texture coordinates along with the texture coordinates [u,v; range: 0 to 1] calculated along with the point cloud vertex. 

Later, I post-process depth and point cloud and now I do not have any missing depth values or vertex with Z = 0. Hence, I would like to calculate a new texture map for the newly generated depth/point cloud along with the RGB image. 

Note: My post-processing on depth map is performed on OpenCV format depth and not depth frame from intelrealsense. Converting NumPy matrix or OpenCV matrix back to depth frame is still an open issue (#5784) .

MartyG-RealSense, 2021-06-07:
Soarwing52's Python conversion of **rs-measure** is at the link below.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

I researched your new information very carefully but did not find much that matched your project description unfortunately. The pyrealsense2 script in the link below looks as though it may fit some of what you are describing though in regard to generating texture mapping for a point cloud.

https://www.programmersought.com/article/12275489223/

jaimeet64, 2021-06-07:
@MartyG-RealSense  #2560 was a bit helpful. But I faced the same issue trying the script. I have posted my results on that issue.

MartyG-RealSense, 2021-06-08:
I looked at the results that you posted at https://github.com/IntelRealSense/librealsense/issues/2560#issuecomment-856310057 - if you are aligning **color to depth** (which the example that Dorodnic provided does - then I wonder if the black areas on the depth image are invalid (**nan**) values rather than coordinates where the depth is zero.

In the past I have seen a couple of cases where RealSense users dealt with invalid nan values by converting them all to zero so that they could be rendered on the image.  

The lRealSense Python example program **opencv_pointcloud_viewer** demonstrates code for dealing with nan values.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py#L157-L185

And here is a non-RealSense example of a nan to zero conversion in Python:

https://www.geeksforgeeks.org/numpy-nan_to_num-in-python/

jaimeet64, 2021-06-08:
> I looked at the results that you posted at [#2560 (comment)](https://github.com/IntelRealSense/librealsense/issues/2560#issuecomment-856310057) - if you are aligning **color to depth** (which the example that Dorodnic provided does - then I wonder if the black areas on the depth image are invalid (**nan**) values rather than coordinates where the depth is zero.
> 
> In the past I have seen a couple of cases where RealSense users dealt with invalid nan values by converting them all to zero so that they could be rendered on the image.
> 
> The RealSense Python example program **opencv_pointcloud_viewer** demonstrates code for dealing with nan values.
> 
> https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_pointcloud_viewer.py#L157-L185
> 
> And here is a non-RealSense example of a nan to zero conversion in Python:
> 
> https://www.geeksforgeeks.org/numpy-nan_to_num-in-python/
> 
> As you are using an example script from Dorodnic that was based on a Jupyter Notebook example, it may alternatively be worth applying a **Hole Filling Filter** post-processing filter on the depth gaps that is described in another Jupyter example by Intel below.
> 
> https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb

Please have a second look at my results. The second image pair (RGB stacked with depth) does not have NAN value associated with the depth map. The whole depth map is replaced with the mean value of the matrix (which approx 127). Hence, why there are holes in the color image. (https://user-images.githubusercontent.com/16480820/121096369-f131bf80-c7a6-11eb-9057-8611c7db5d48.PNG)

I will try out the script that you suggested in the previous comment.

MartyG-RealSense, 2021-06-08:
Your conversion to a mean value of the matrix is outside of my programming knowledge unfortunately, so I do not have any advice that I can offer on that particular aspect of your project.  I do apologize.

jaimeet64, 2021-06-09:
Mean value  of 3x3 matrix [[2,3,1],[3,3,1],[1,2,2]] is 2. Basically, the sum of all components divided by the total number of components will give you the mean value of the matrix. This documentation will help you understand mean  of a matrix(https://numpy.org/doc/stable/reference/generated/numpy.mean.html)

MartyG-RealSense, 2021-06-10:
Thanks very much for the link but I still do not have any programming advice that I can offer on that subject.

You mentioned that you were going to test a script.  If you have done so, what was the outcome of that test please?

closed 2021-06-16 by jaimeet64