realsenseai/librealsense #9172: timestamp_domain

mengxz0203, 2021-06-07:
Hello,I want to know how to record a bag file with system_time but not global_time? I have tried several times but failed.

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  1909                                    | 
| Platform                                 | PC  |
| SDK Version                           | pyrealsense2                       | 
| Language                              | python                          | 
| Segment			|  Robot                          | 

### Issue Description
I find `timestamp_domain` in the doc, but I don't know how to set  system_time in viewer or with python code.
[image]
[image]

mengxz0203, 2021-06-07:
After I uncheck the Global Time Enabled, clock domain becomes hardware
[image]
[image]

MartyG-RealSense, 2021-06-07:
Hi @mengxz0203  The link below contains Python scripting for setting the status of global time.

https://github.com/IntelRealSense/librealsense/issues/8112

[image]

Instead of setting **rs.option.global_time_enabled** to '1' (true) like the script does, use '0' instead to disable global time.

mengxz0203, 2021-06-07:
@MartyG-RealSense I tried your code but the clock was hardware
[image]
[image]
so how to get system_time

mengxz0203, 2021-06-07:
@MartyG-RealSense And I also want to know the difference between global_time and system_time, thanks a lot.

MartyG-RealSense, 2021-06-07:
It looks as though your hardware clock image is from the RealSense Viewer.  A Python program will not change any settings in the RealSense Viewer.

The global time setting alternates between Global Time when it is true and Hardware Clock (if supported) when it is false.  Device time and system time are aligned when Global Time is enabled.  The link below is a good reference about the workings of global time.

https://github.com/IntelRealSense/librealsense/issues/4505#issuecomment-553590316

System Time is a **software timestamp** that is used when hardware timestamps are not available (for example, if **metadata** support is not installed).  If hardware timestamps are available, librealsense will take advantage of them.

https://github.com/IntelRealSense/librealsense/issues/1294

More information about how different timestamps function is here:

https://github.com/IntelRealSense/librealsense/issues/2188#issuecomment-409985803

mengxz0203, 2021-06-07:
@MartyG-RealSense Thanks very much!

closed 2021-06-07 by mengxz0203