realsenseai/librealsense #9173: dontnet c# Intel.Realsense assembly bad format exception

jonaumann, 2021-06-07:
Hi,
I have implemented a realsense image capture program under windows 10 with Visual Studio 2019.
I want to run that program on a raspberry pi with 64 bit PiOS.
I could publish my program on windows for the linux-arm64 target runtime and could run that program on the Raspi, but I get an exception when the Intel.Realsense assembly is used:

Unhandled exception. System.BadImageFormatException: Could not load file or assembly 'Intel.RealSense, Version=2.42.0.0, Culture=neutral, PublicKeyToken=null'. An attempt was made to load a program with an incorrect format.

File name: 'Intel.RealSense, Version=2.42.0.0, Culture=neutral, PublicKeyToken=null'

It seems that the Intel.Realsense assembly is for 32bit only, because on a Raspberry Pi 32bit OS it works.

How can I get the assembly for 64 bit und use that for my windows development and publish the correct version?
Thanks for any help!

MartyG-RealSense, 2021-06-07:
Hi @jonaumann   This error can occur with C# when there is an architecture mismatch with the target platform, such as when a program is targeting **AnyCPU** instead of 'x86' or 'x64'.

https://github.com/IntelRealSense/librealsense/issues/2125#issuecomment-407186203

If you are building librealsense with CMake then you could target x64 by adding the build flag below to your CMake build instruction:

**-DCMAKE_GENERATOR_PLATFORM=x64**

The RealSense user in the above link solved the problem using this command and shared the full CMake instruction that they used.

Another example of the command's use is in the build instructions for the RealSense Unity wrapper (Unity is C#-based).

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity#step-1---build-dependencies-and-open-the-unity-project

https://github.com/IntelRealSense/librealsense/issues/2125#issuecomment-407189662

jonaumann, 2021-06-08:
thanks, that worked :-)

closed 2021-06-08 by jonaumann

MartyG-RealSense, 2021-06-08:
Great news @jonaumann - thanks very much for the update!