realsenseai/librealsense #9174: How to get supported resolutions and options programmatically in python?

mabouseif-riact, 2021-06-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400| 
| Firmware Version                   | 0.5.12.13.50 | 
| Operating System & Version |  Linux (Ubuntu 120)| 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |    2.45.0                           | 
| Language                            |  python                          | 
| Segment			|  robot  | 

### Issue Description
How to get supported resolutions and options programmatically by realsense device (in this case D435) in Python? More importantly resolutions.
Thanks in advance!

MartyG-RealSense, 2021-06-07:
Hi @mabouseif-riact   The RealSense Python wrapper supports the following instructions for retrieving stream profiles and supported options, similar to the C++ **rs-enumerate-devices** tool.

**get_stream_profiles()**
Retrieves the list of stream profiles (resolution, format and frames per second) supported by the sensor

**get_supported_options()**
Retrieve list of supported options

A simple example of using **get_stream_profiles()** to inspect stream profiles with Python can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/2652#issuecomment-435056679

The RealSense Python script in the link below, meanwhile, provides an example of inspecting supported options with **get_supported_options()**

https://www.programmersought.com/article/70864312709/

[image]

mabouseif-riact, 2021-06-08:
Hi @MartyG-RealSense, 

That solved the problem. Thanks! : )
It just seems I have to do some extra steps in order to extract the resolutions themselves that are supported.

MartyG-RealSense, 2021-06-08:
Great to hear that the information was helpful in finding a solution @mabouseif-riact - thanks for the update!   :)

closed 2021-06-09 by mabouseif-riact