realsenseai/librealsense #9176: Can not start Real Sense Viewer again

Carp-catcher, 2021-06-07:
Hello, 
today I installed the real sense Viewer and it worked .... until I cloesed it.
Now  I can't open it.
The following appears.
Could anybody help me please ?
Thank you

 | | |
|---|---|
|**librealsense**|2.45.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
GL_INVALID_VALUE - C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\common\opengl3.cpp:569

```

MartyG-RealSense, 2021-06-07:
Hi @Carp-catcher   SDK **2.47.0** has just been released and includes a fix for a crash when the RealSense Viewer is closed.  Could you try installing 2.47.0 and see whether it corrects your problem please?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.47.0

Carp-catcher, 2021-06-07:
Hi MartyG,
yes, I remeber ... when I started the app for the first time it asked for a update and I took it.
I tried your suggestion... unfortunately it did't help.
[image]

MartyG-RealSense, 2021-06-07:
Thanks very much @Carp-catcher for trying the new SDK version.

Have you installed the full RealSense SDK using the file **Intel.RealSense.SDK-WIN10-2.47.0.3309.exe** please?

Carp-catcher, 2021-06-07:
No, I only installed the viewer.

MartyG-RealSense, 2021-06-07:
Okay, would it be possible to install the full SDK and re-test please?

Carp-catcher, 2021-06-07:
I just installed this: Intel.RealSense.SDK-WIN10-2.47.0.3309.exe
After installation it starts automatically the real sense Viewer ... but again the same error :(

MartyG-RealSense, 2021-06-07:
Is the Viewer able to launch again if you reboot the computer, at least until the next time that you close the Viewer?

Carp-catcher, 2021-06-07:
I have reboot the computer.
Now I get an other error ... see attached

MartyG-RealSense, 2021-06-07:
If you attached a picture or file, it is missing.  Please attach it on the web-page instead of sending it by email when you have the opportunity to do so.  Thanks!

Carp-catcher, 2021-06-08:
Hello, 
oh you are right, it isn't possible to attach files by mail ... sorry.
So that came up when I started the viewer yesterday:
[image]
Today (so after restarting again) that already knwon error comes up:
[image]

It drives me crazy....

MartyG-RealSense, 2021-06-08:
Thank you very much for the update.  As you installed the full SDK yesterday, you should be able to access its pre-made example programs on Windows.  Could you use the instructions in the link below to run an example called **rs-capture** and let me know whether it works, please?

https://github.com/IntelRealSense/librealsense/issues/8024#issuecomment-747630656

Carp-catcher, 2021-06-08:
okay I will do so. 
Thanks for your try.

closed 2021-06-08 by Carp-catcher