realsenseai/librealsense #9181: Apt server is broken, gives 403 Forbidden

hyuDev, 2021-06-08:
sudo apt-get update
받기:1 file:/var/cuda-repo-10-2-local-10.2.89-440.33.01  InRelease
무시:1 file:/var/cuda-repo-10-2-local-10.2.89-440.33.01  InRelease
받기:2 file:/var/cuda-repo-10-2-local-10.2.89-440.33.01  Release [574 B]
받기:2 file:/var/cuda-repo-10-2-local-10.2.89-440.33.01  Release [574 B]
기존:3 http://mirror.kakao.com/ubuntu bionic InRelease                                                                                          
기존:5 http://mirror.kakao.com/ubuntu bionic-updates InRelease                                                                                  
기존:6 http://mirror.kakao.com/ubuntu bionic-backports InRelease                                                                                
기존:7 http://mirror.kakao.com/ubuntu bionic-security InRelease                                                                                 
기존:8 https://linux.teamviewer.com/deb stable InRelease                                                                                        
기존:9 https://packages.microsoft.com/repos/vscode stable InRelease                                                                             
무시:10 http://dl.google.com/linux/chrome-remote-desktop/deb stable InRelease                                                                   
기존:11 http://dl.google.com/linux/chrome/deb stable InRelease                                                                                  
기존:12 http://dl.google.com/linux/chrome-remote-desktop/deb stable Release                                                                     
기존:13 http://packages.ros.org/ros/ubuntu bionic InRelease                                                                                     
오류:14 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease                                                            
  403  Forbidden [IP: 52.218.104.41 80]
기존:15 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu bionic InRelease                             
기존:17 http://ppa.launchpad.net/jonathonf/ffmpeg-4/ubuntu bionic InRelease
패키지 목록을 읽는 중입니다... 완료     
E: http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease 파일을 받는데 실패했습니다  403  Forbidden [IP: 52.218.104.41 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

I think your server is broken, can you fix it??

MartyG-RealSense, 2021-06-08:
Hi @hyuDev   The **http** addresses at the bottom of your list suggest that your problem is occurring because Intel updated the package links to **https** in March 2021.  The package installation instructions were updated to reflect this change.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

MartyG-RealSense, 2021-06-15:
Hi @hyuDev   Do you require further assistance with this case, please?  Thanks!

korosava, 2021-06-15:
> Hi @hyuDev The **http** addresses at the bottom of your list suggest that your problem is occurring because Intel updated the package links to **https** in March 2021. The package installation [...]

I can't add the realsense server to the list of repositories, following this tutorial.
```bash
sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
```
``` bash
Hit:1 http://ua.archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://ua.archive.ubuntu.com/ubuntu focal-updates InRelease                                                               
Hit:3 http://ua.archive.ubuntu.com/ubuntu focal-backports InRelease                                                                                                                                        
Err:4 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease                                                                                                                         
  403  Forbidden [IP: 52.218.89.162 80]
Hit:5 https://librealsense.intel.com/Debian/apt-repo focal InRelease                                                   
Hit:6 http://packages.ros.org/ros2/ubuntu focal InRelease                                        
Hit:7 http://security.ubuntu.com/ubuntu focal-security InRelease
Reading package lists... Done
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/bionic/InRelease  403  Forbidden [IP: 52.218.89.162 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

korosava, 2021-06-15:
> > Hi @hyuDev The **http** addresses at the bottom of your list suggest that your problem is occurring because Intel updated the package links to **https** in March 2021. The package installation instructions were updated to reflect this change.
> > https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages
> 
> I can't add the realsense server to the list of repositories, following this tutorial.
> 
> ```shell
> sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
> ```
> 
> ```shell
> Hit:1 http://ua.archive.ubuntu.com/ubuntu focal InRelease
> Hit:2 http://ua.archive.ubuntu.com/ubuntu focal-updates InRelease                                                               
> Hit:3 http://ua.archive.ubuntu.com/ubuntu focal-backports InRelease                                                                                                                                        
> Err:4 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease                                                                                                                         
>   403  Forbidden [IP: 52.218.89.162 80]
> Hit:5 https://librealsense.intel.com/Debian/apt-repo focal InRelease                                                   
> Hit:6 http://packages.ros.org/ros2/ubuntu focal InRelease                                        
> Hit:7 http://security.ubuntu.com/ubuntu focal-security InRelease
> Reading package lists... Done
> E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/bionic/InRelease  403  Forbidden [IP: 52.218.89.162 80]
> E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' is not signed.
> N: Updating from such a repository can't be done securely, and is therefore disabled by default.
> N: See apt-secure(8) manpage for repository creation and user configuration details.
> ```

Oh, I just forget to delete old realsense link form /etc/apt/sources.list, all works now

MartyG-RealSense, 2021-06-15:
Thanks very much for the update @korosava - I'm very glad that you found a solution  :)

UserFLO, 2021-06-16:
Hi, i have the same issue. Any idea ? Thanks

[image]

MartyG-RealSense, 2021-06-16:
Hi @UserFLO   I note that you are using VirtualBox (a VM).  Whilst i have not personally seen this issue in a librealsense case, my research indicates that this 'handshake' error may be caused by using a VM.  An example of a VM user who had the same problem in a non-RealSense case is in the link below.

https://askubuntu.com/questions/1095266/apt-get-update-failed-because-certificate-verification-failed-because-handshake/1210812#1210812

MartyG-RealSense, 2021-06-22:
Hi @UserFLO  Do you still require assistance with your problem, please?  Thanks!

UserFLO, 2021-06-22:
Hi @MartyG-RealSense, i used the method : https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md 
It works ! Thanks

MartyG-RealSense, 2021-06-22:
Great news @UserFLO - thanks very much for the update!

@hyuDev - as the creator of this case, have you solved your problem, please?

MartyG-RealSense, 2021-07-03:
Case closed due to no further comments received.

closed 2021-07-03 by MartyG-RealSense

hyuDev, 2021-07-05:
Hi 
I'm sorry about late feedback.
I tried as you suggested, but the result was the same. Here's the comment screen I've done. What more should I do? I don't even listen to the erasing orders you gave me.

sudo apt update
기존:1 http://mirror.kakao.com/ubuntu bionic InRelease
받기:3 http://mirror.kakao.com/ubuntu bionic-backports InRelease [74.6 kB]     
받기:4 http://mirror.kakao.com/ubuntu bionic-security InRelease [88.7 kB]      
받기:2 http://mirror.kakao.com/ubuntu bionic-updates InRelease [88.7 kB]       
기존:5 https://linux.teamviewer.com/deb stable InRelease                       
받기:6 http://mirror.kakao.com/ubuntu bionic-backports/universe amd64 DEP-11 Metadata [9,272 B]
무시:7 http://dl.google.com/linux/chrome-remote-desktop/deb stable InRelease   
기존:8 http://packages.ros.org/ros/ubuntu bionic InRelease                     
기존:9 https://packages.microsoft.com/repos/vscode stable InRelease            
받기:10 http://mirror.kakao.com/ubuntu bionic-security/main amd64 DEP-11 Metadata [48.5 kB]
받기:11 http://mirror.kakao.com/ubuntu bionic-security/universe amd64 DEP-11 Metadata [60.5 kB]
받기:12 http://mirror.kakao.com/ubuntu bionic-security/multiverse amd64 DEP-11 Metadata [2,468 B]
기존:13 http://dl.google.com/linux/chrome/deb stable InRelease                 
기존:14 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu bionic InRelease  
받기:16 http://mirror.kakao.com/ubuntu bionic-updates/main i386 Packages [1,311 kB]
받기:16 http://mirror.kakao.com/ubuntu bionic-updates/main i386 Packages [1,311 kB]
기존:19 http://dl.google.com/linux/chrome-remote-desktop/deb stable Release    
받기:22 http://mirror.kakao.com/ubuntu bionic-updates/multiverse amd64 DEP-11 Metadata [2,468 B]
받기:15 http://mirror.kakao.com/ubuntu bionic-updates/main amd64 Packages [2,130 kB]
오류:23 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease
  403  Forbidden [IP: 52.218.62.242 80]
받기:16 http://mirror.kakao.com/ubuntu bionic-updates/main i386 Packages [1,311 kB]
받기:20 http://mirror.kakao.com/ubuntu bionic-updates/main amd64 DEP-11 Metadata [294 kB]
받기:21 http://mirror.kakao.com/ubuntu bionic-updates/universe amd64 DEP-11 Metadata [290 kB]
기존:24 http://ppa.launchpad.net/jonathonf/ffmpeg-4/ubuntu bionic InRelease    
패키지 목록을 읽는 중입니다... 완료                                            
E: http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease 파일을 받는데 실패했습니다  403  Forbidden [IP: 52.218.62.242 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
(base) user1@user:~$ ^C
(base) user1@user:~$ sudo degit /etc/apt/sources.list
sudo: degit: command not found
(base) user1@user:~$ sudo gedit /etc/apt/sources.list
^C
(base) user1@user:~$ sudo gedit /etc/apt/sources.list
^C
(base) user1@user:~$ sudo gedit /etc/apt/sources.list
^C
(base) user1@user:~$ sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo $(lsb_release -cs) main" -u
기존:1 http://mirror.kakao.com/ubuntu bionic InRelease
기존:2 http://mirror.kakao.com/ubuntu bionic-updates InRelease                 
기존:3 http://mirror.kakao.com/ubuntu bionic-backports InRelease               
기존:4 http://mirror.kakao.com/ubuntu bionic-security InRelease                
기존:5 https://linux.teamviewer.com/deb stable InRelease                       
무시:6 http://dl.google.com/linux/chrome-remote-desktop/deb stable InRelease   
기존:7 https://packages.microsoft.com/repos/vscode stable InRelease            
기존:8 http://dl.google.com/linux/chrome/deb stable InRelease                  
기존:9 http://dl.google.com/linux/chrome-remote-desktop/deb stable Release     
오류:10 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease
  403  Forbidden [IP: 52.218.108.34 80]
기존:11 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu bionic InRelease  
기존:12 http://packages.ros.org/ros/ubuntu bionic InRelease                    
기존:14 http://ppa.launchpad.net/jonathonf/ffmpeg-4/ubuntu bionic InRelease    
받기:15 https://librealsense.intel.com/Debian/apt-repo bionic InRelease [3,251 B]
받기:16 https://librealsense.intel.com/Debian/apt-repo bionic/main amd64 Packages [118 kB]
패키지 목록을 읽는 중입니다... 완료      
E: http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease 파일을 받는데 실패했습니다  403  Forbidden [IP: 52.218.108.34 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
(base) user1@user:~$ sudo apt-get update
기존:1 http://mirror.kakao.com/ubuntu bionic InRelease
기존:2 http://mirror.kakao.com/ubuntu bionic-updates InRelease                 
기존:3 http://mirror.kakao.com/ubuntu bionic-backports InRelease               
[... 83 lines cut ...]
Codename:	bionic
++ uname -r
+ echo 'Kernel version 4.15.0-23-generic'
Kernel version 4.15.0-23-generic
+ sudo apt-get update
기존:1 http://mirror.kakao.com/ubuntu bionic InRelease
기존:2 http://mirror.kakao.com/ubuntu bionic-updates InRelease                 
기존:3 http://mirror.kakao.com/ubuntu bionic-backports InRelease               
[... cut ...]
기존:14 https://librealsense.intel.com/Debian/apt-repo bionic InRelease        
패키지 목록을 읽는 중입니다... 완료                                 
E: http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/xenial/InRelease 파일을 받는데 실패했습니다  403  Forbidden [IP: 52.218.120.49 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

MartyG-RealSense, 2021-07-05:
Hi @hyuDev  Did you use the command **sudo apt-key adv --keyserver keys.gnupg.net --recv-key** to access the key server to install packages, please?  If so, there was a recently identified issue with the **keys.gnupg.net** keyserver address that can be resolved by using a different address.

https://github.com/IntelRealSense/librealsense/issues/9289#issuecomment-873457374

If you did attempt to install packages using **sudo apt-key adv --keyserver keys.gnupg.net --recv-key** then would it be possible to try an installation using an alternative keyserver address such as the one in the link above and see whether it makes a positive difference?

MartyG-RealSense, 2021-07-07:
It is confirmed in https://github.com/IntelRealSense/librealsense/issues/9289#issuecomment-874920076 that using the full keyserver command in the SDK installation instructions that provides an alternative server address via the '||' symbol should correct the problem.