realsenseai/librealsense #9182: L515 Problem with running the RealSense Viewer

Carp-catcher, 2021-06-08:
Hello, 

i have connected my new L515 with my Comupter and downloaded/installed this:
Intel.RealSense.SDK-WIN10-2.45.0.3212.exe
After successful installation I want to start the Viewer and that error came up.
I tried the previous Version (Intel.RealSense.SDK-WIN10-2.47.0.3309) too... but with the same error report.

... when I start the rs-capture the following comes up: 
[image]

My graphics driver is up-to-date.

Could anybody help me ? 
Thanks a lot.
  

| | |
|---|---|
|**librealsense**|2.45.0 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

RealSenseSupport, 2021-06-18:
Hi @Carp-catcher 
there is similar issue reported before 
https://github.com/IntelRealSense/librealsense/issues/7666
when streaming using viewer, is everything ok and the camera is enumerated as USB 3 device

MartyG-RealSense, 2021-06-27:
Hi @Carp-catcher  Do you require further assistance with this case, please?  Thanks!

Carp-catcher, 2021-06-28:
Hi,
oh sorry, I forgot to answer.
No, thank you.
I don't need further assistance.

Regards

MartyG-RealSense, 2021-06-28:
Thanks very much @Carp-catcher for the update!  I will close this case as you do not require further assistance.

closed 2021-06-28 by MartyG-RealSense