realsenseai/librealsense #9187: How to get tracking data with D455?

Noctstar, 2021-06-09:
| Camera | D455 | 
| OS | Win 10 |

Hi,

I want traking data like T265 with RealSenseD455. 
Is it possible? Also how can I get them in python code?

If I can't get the tracking data, can I derive them from IMU data?
Thank you.

MartyG-RealSense, 2021-06-09:
Hi @Noctstar  The 400 Series depth camera models that have an IMU do not have built-in support for 6DOF pose data but there are ways that pose of the camera or observed objects can be obtained.  

The discussion in the link below provides information resources for obtaining the pose of objects.

https://support.intelrealsense.com/hc/en-us/community/posts/1500001220042-D415-Projection-Matrix

Alternatively, objects can be tracked by applying **ArUco** image tags to them.

https://github.com/IntelRealSense/librealsense/issues/8961

In regard to obtaining the current angle of the camera itself instead of an observed object, the Python discussion in the link below looks at obtaining yaw, pitch and roll from the IMU.

https://github.com/IntelRealSense/librealsense/issues/4391

Angle can also be obtained without the IMU by using a **plane-fit** algorithm.

https://support.intelrealsense.com/hc/en-us/community/posts/360050894154/comments/360013322694

MartyG-RealSense, 2021-06-14:
Hi @Noctstar  Do you require further assistance with this case, please?  Thanks!

Noctstar, 2021-06-14:
Thank you! It's solved!

closed 2021-06-14 by Noctstar

MartyG-RealSense, 2021-06-14:
Excellent news that you obtained a solution @Noctstar - thanks very much for the update!