realsenseai/librealsense #9191: Camera image freezes randomly

AndyKohr, 2021-06-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.11.01.100 / 05.12.07.100 / 05.12.13.50 | 
| Operating System & Version |   Linux (Raspbian 10 Lite)| 
| Kernel Version (Linux Only)    |  5.10.17 32 bit                                         | 
| Platform                                 | Raspberry Pi 4 2GB  |
| SDK Version                            |  2.38.1 / 2.45.0                          | 
| Language                            |  C++                          | 
| Segment			|  others                          | 

### Issue Description
Hi,

we recognized a problem with the Realsense D435 or D430 module, together with a Raspberry Pi 4 2 GB like issue #8750 in our application. Sometimes, the camera image freezes and we can see the following error in syslog:
usb 2-1: Process x (y) called USBDEVFS_CLEAR_HALT for active endpoint 0x82.
The problem occurs randomly and with different devices within minutes or once a month. This problem is new for us but it looks like more and more of or devices have that issue.
It is also critical, because in most cases, the system seems to be running normal because we still can receive 3d data of the camera. The problem can only be bypassed with a restart of the application / reboot of the device.

It would be very helpful for us if someone can lead us in the right direction. We cannot reproduce this at the moment with different software and hardware combinations.  

Some additional information:
* We only use infrared stream and depth stream, not Rob camera
* We use and have to use default resolution 848x480 @30fps  
* I want to exclude a cable/connector problem because we have the issue for different devices with  D435 and also D430 module and different cables.
* We use Librealsense 2.38.1 with firmware 05.12.07.100 for a very long time now and had no issues so far (or didn't notice) but problems are increasing now.
* We have tested kernel update (raspberrypi-kernel from 1.20210303-1 to 1.20210527-1), Librealsense and FW update (2.45.0) without any difference. 
* With one device, the Issue occurs a few seconds / minutes after restart of our application, for another device only once a month.
* Cpu usage depends a bit on the usage scenario but in most cases it is around 80- 110%.
* Memory usage is round about 1/4 of the available memory.

* We compile and use Librealsense with the following flags:
```
cmake -DCMAKE_BUILD_TYPE=Release \
      -DBUILD_EXAMPLES=ON \
      -DBUILD_GRAPHICAL_EXAMPLES=OFF \
      -DFORCE_RSUSB_BACKEND=ON \
      -DENFORCE_METADATA=OFF \
      -DGLFW_INSTALL=OFF \
      -DBUILD_WITH_TM2=OFF \
```

MartyG-RealSense, 2021-06-09:
Hi @AndyKohr   Thanks very much for the detailed description of your setup.  The main item that stands out is the reported CPU usage of between 80 and 110 percent depending on the usage scenario.  I can envision a scenario where problems may occur when reaching and exceeding 100% of usage, as the Pi may simply run out of resources to process the incoming data and jam up.  This is described in the link below.

https://github.com/IntelRealSense/librealsense/issues/5412#issuecomment-564122881

What kinds of application are you running on your Pi please?

AndyKohr, 2021-06-09:
Information update:
We tested with reduced frame rate (15 fps) and reduced resolution (640x480). The problem is still there. The CPU load was below 50%.

MartyG-RealSense, 2021-06-09:
Do you have auto-exposure enabled, please?  If you do, next time the problem occurs could you test whether the problem automatically corrects itself if you cover the camera lenses with your hand?

AndyKohr, 2021-06-09:
The Problem exists with or without auto exposure

MartyG-RealSense, 2021-06-09:
A scenario that could cause the depth image to severely deteriorate whilst the IR image changed little would be if the value of the **Laser Power** option dropped to minimum from its default of '150'.

If the exposure was minimizing then I would expect the IR image to go completely dark.

Laser Power could be automatically shut off by the camera's laser safety mechanism if the Projector temperature of the camera was first detected to exceed 60 degrees C for several seconds (causing Laser Power to be halved) and then the temperature continued to exceed 60 degrees C.  The shut-off can also be triggered if the detected temperature falls below 0 degrees C.

AndyKohr, 2021-06-09:
I think it is very unlikely that laser power has to do with it. The ir-image did not change anymore (no noise visible, not a single bit has changed compared to the previous image), but the depth image still changes compared to the previous frame.

MartyG-RealSense, 2021-06-09:
How does the depth stream behave when the camera is used on your Pi with the **RealSense Viewer** tool instead of your own application?

AndyKohr, 2021-06-09:
The system is headless, so i can't test this.

MartyG-RealSense, 2021-06-09:
Are you using a **Virtual Machine (VM)** setup with your headless system?  If so, this may not provide performance that is comparable to a regular setup where the camera is connected directly to a computer.  If a VM is being used, Intel recommends using **VMWare Workstation Player** in order to provide USB 3 emulation.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#linux-ubuntu-installation

AndyKohr, 2021-06-09:
It's running on native Raspi-4 hardware with raspbian OS. CPU-Load is not very high.

MartyG-RealSense, 2021-06-10:
I have extensively researched your case.  I believe that USBDEVFS_CLEAR_HALT is not going to be a useful line of investigation as there are so few references about it in regard to librealsense specifically.

Your case may be similar to https://github.com/IntelRealSense/librealsense/issues/8880 - a case where the infrared stream was occasionally going wrong on Raspberry Pi 4 with Raspbian 10 and causing a bad depth image.  A test for whether it is the same situation may be to stream RGB too, as in https://github.com/IntelRealSense/librealsense/issues/8880 the RGB image went completely black when the problem with the infrared occurred.

https://github.com/IntelRealSense/librealsense/issues/8880 was tested with a static scene of a store-room with nobody inside it, so I cannot tell if the infrared stream is experiencing erratic motion like the gates in https://github.com/IntelRealSense/librealsense/issues/8750

The RealSense user in the storeroom case found that the camera could run for several weeks without problems if using the official RealSense 1 meter cable instead of their own choice of cable

AndyKohr, 2021-06-10:
I did some further investigations and did find out that the frame numbers i get for the images via the function get_frame_number() start to differ when the USB syslog kenel message occurs. The depth frame count number increases as expected but the ir frame counter did not change any more.

infrared frame number = 2098, depth frame number = 1289
infrared frame number = 2098, depth frame number = 1290
infrared frame number = 2098, depth frame number = 1291
infrared frame number = 2098, depth frame number = 1292
infrared frame number = 2098, depth frame number = 1293
infrared frame number = 2098, depth frame number = 1294
infrared frame number = 2098, depth frame number = 1295
infrared frame number = 2098, depth frame number = 1296
infrared frame number = 2098, depth frame number = 1297
infrared frame number = 2098, depth frame number = 1298
infrared frame number = 2098, depth frame number = 1299
infrared frame number = 2098, depth frame number = 1300
infrared frame number = 2098, depth frame number = 1301

MartyG-RealSense, 2021-06-10:
Thanks very much for the new data.  And there is no multiple camera hardware sync involved, with sync cabling connecting more than one camera together?

AndyKohr, 2021-06-11:
No, nothing special. Just one camera.

MartyG-RealSense, 2021-06-11:
A past solution to the **USBDEVFS_CLEAR_HALT** issue has been to reduce resolution, but you have to use **848x480 at 30FPS** and cannot alter that.

Would it therefore be an option to reduce depth detail without changing the resolution setting by applying a post-processing **Decimation Filter**?

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#decimation-filter  

https://github.com/IntelRealSense/librealsense/tree/master/examples/post-processing

AndyKohr, 2021-06-11:
As mentioned before we already reduced resolution and frame rate (640x480 @ 15 fps). But that did not solve the problem.

MartyG-RealSense, 2021-06-11:
I researched the case from the beginning to seek new insights.  As it is the infrared stream that is apparently freezing, I believe that there are a limited number of possibilities to investigate as it would be more common for depth to freeze than infrared.  I would expect an infrared freeze to result from the IR sensor being saturated due to directly facing a bright light source such as the sun.

Are you performing any stream alignment, such as depth to IR?

Thanks very much for your patience during this investigative process!

AndyKohr, 2021-06-11:
In principal i do this:

rs2::config cfg;
cfg.enable_stream(RS2_STREAM_INFRARED, 1, width, height, RS2_FORMAT_Y8, fps);
cfg.enable_stream(RS2_STREAM_DEPTH, width, height, RS2_FORMAT_Z16, fps);
rs2::pipeline pipe;
pipe.start(cfg)

while(true) {
  rs2::frameset frames = pipe.wait_for_frames(5000);
  rs2::depth_frame depth_frame = frames.get_depth_frame();
  print depth_frame.get_frame_number();
  rs2::video_frame ir_frame = frames.get_infrared_frame(1);
  print ir_frame.get_frame_number();
}

AndyKohr, 2021-06-11:
The freeze of ir frame has nothing to do with exposure or sunlight, etc. it can happen anytime, also at night. We've tried auto expsure and manual expsure. No success.

MartyG-RealSense, 2021-06-11:
As you are not using the right-side infrared stream, it may not be necessary to use a stream index number.

cfg.enable_stream(RS2_STREAM_INFRARED, width, height, RS2_FORMAT_Y8, fps);

rs2::video_frame ir_frame = frames.get_infrared_frame();

AndyKohr, 2021-06-12:
Is it a problem to use the stream index in enable_stream if i only want to use the left imager? Then the documentation dhout be clearer about this.

MartyG-RealSense, 2021-06-12:
Python examples of defining left and right IR streams typically use the index number '1' for the left stream and '2' for the right.  The SDK's C++ example code for doing the same uses '0' and '1' though.  So I just want to eliminate the possibility that the camera is streaming the right IR instead of the left.  This could cause a stream failure if a USB event occurred that caused the camera to go into USB 2 mode instead of USB 3, since Infrared 2 (right infrared) is not supported on USB 2.

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#start-streaming-left-and-right-infrared-imagers

AndyKohr, 2021-06-12:
Just tested it only Index 1 and 2 are working. When I give 0 as index in enable_stream() an error message occurs.

MartyG-RealSense, 2021-06-12:
And does it work with no index number, on the assumption that it defaults to the left infrared without an index.

AndyKohr, 2021-06-12:
Will try this tomorow, but it is unlikely that this will make a difference.

MartyG-RealSense, 2021-06-12:
If something can be checked quickly then it can be worth doing so in order to help to eliminate it as a possibility, even if it seems an unlikely possibility.  Thanks very much for your patience during this investigative prices.

AndyKohr, 2021-06-13:
Still freezes when i use -1 in enable_stream(RS2_STREAM_INFRARED, -1, width, height, RS2_FORMAT_Y8, fps);

MartyG-RealSense, 2021-06-13:
And what is the result please when using no index number at all?

**cfg.enable_stream(RS2_STREAM_INFRARED, width, height, RS2_FORMAT_Y8, fps);**

AndyKohr, 2021-06-13:
Same problem. Still freezes.

MartyG-RealSense, 2021-06-13:
If your application is capable of running for a month without a problem then there is likely not a problem with the CMake build settings, or a significant issue with the resultant librealsense build.

 You could try increasing the size of your Pi's swapfile to provide it with more virtual memory to use in the event that the 2GB of real memory is used up.  Given that the problem can occur seconds after start, running out of memory also seems unlikely to be a cause.

Regarding your earlier statement of "With one device, the Issue occurs a few seconds / minutes after restart of our application, for another device only once a month" ... is it always the same device that has a particular begaviour?  For example, Camera A always stops after seconds and Camera B always runs for a month before stopping.

AndyKohr, 2021-06-13:
Most Sensors run without any problem. Only a few are having this poblem. I just want to understand how th frame counter woks. Why do they start fom beginning? When are they incremented? Can i loog into the sourcecode where this is done?

MartyG-RealSense, 2021-06-13:
There is a scenario where the frame counter of the depth stream can reset if there is an electrostatic discharge (ESD) event, though that typically occurs with multiple-camera sync cabling that do not have ESD protection components.   This phenomenon is described in the section of Intel's multiple camera white paper linked to below.

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration#section-1-connecting-the-cameras

That was why I asked earlier if you were using sync cabling when the subject of the frame counter returning to zero came up.  But you have said that you aren't using such cabling.

Advice about other types of event that can cause the counter to reset is provided in the link below.

https://github.com/IntelRealSense/librealsense/issues/7819#issuecomment-733006800

AndyKohr, 2021-06-15:
This is a test program which causes the issue:

`// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

/* Include the librealsense C header files */
#include <librealsense2/rs.h>
#include <librealsense2/h/rs_pipeline.h>
#include <librealsense2/h/rs_frame.h>

#include <stdlib.h>
#include <stdint.h>
#include <stdio.h>
#include "example.h"

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//                                     These parameters are reconfigurable                                        //
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#define WIDTH           848               // Defines the number of columns for each frame                         //
#define HEIGHT          480               // Defines the number of lines for each frame                           //
#define FPS             30                // Defines the rate of frames per second                                //
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////

int main()
{
    rs2_error* e = 0;

    // Create a context object. This object owns the handles to all connected realsense devices.
    // The returned object should be released with rs2_delete_context(...)
    rs2_context* ctx = rs2_create_context(RS2_API_VERSION, &e);
    check_error(e);

    /* Get a list of all the connected devices. */
    // The returned object should be released with rs2_delete_device_list(...)
    rs2_device_list* device_list = rs2_query_devices(ctx, &e);
    check_error(e);

    int dev_count = rs2_get_device_count(device_list, &e);
    check_error(e);
    printf("There are %d connected RealSense devices.\n", dev_count);
    if (0 == dev_count)
        return EXIT_FAILURE;

    // Get the first connected device
    // The returned object should be released with rs2_delete_device(...)
    rs2_device* dev = rs2_create_device(device_list, 0, &e);
    check_error(e);

    print_device_info(dev);

    // Create a pipeline to configure, start and stop camera streaming
    // The returned object should be released with rs2_delete_pipeline(...)
    rs2_pipeline* pipeline =  rs2_create_pipeline(ctx, &e);
    check_error(e);

    // Create a config instance, used to specify hardware configuration
    // The retunred object should be released with rs2_delete_config(...)
    rs2_config* config = rs2_create_config(&e);
    check_error(e);

    // Request a specific configuration
    rs2_config_enable_stream(config, RS2_STREAM_DEPTH, 0, WIDTH, HEIGHT, RS2_FORMAT_Z16, FPS, &e);
    rs2_config_enable_stream(config, RS2_STREAM_INFRARED, 1, WIDTH, HEIGHT, RS2_FORMAT_Y8, FPS, &e);
    check_error(e);

    // Start the pipeline streaming
    // The retunred object should be released with rs2_delete_pipeline_profile(...)
    rs2_pipeline_profile* pipeline_profile = rs2_pipeline_start_with_config(pipeline, config, &e);
    if (e)
    {
        printf("The connected device doesn't support streaming!\n");
        exit(EXIT_FAILURE);
    }

    while (1)
    {
        // This call waits until a new composite_frame is available
        // composite_frame holds a set of frames. It is used to prevent frame drops
        // The returned object should be released with rs2_release_frame(...)
        rs2_frame* frames = rs2_pipeline_wait_for_frames(pipeline, RS2_DEFAULT_TIMEOUT, &e);
        check_error(e);

        // Returns the number of frames embedded within the composite frame
        int num_of_frames = rs2_embedded_frames_count(frames, &e);
        check_error(e);

        int i;
        for (i = 0; i < num_of_frames; ++i)
        {
            // The retunred object should be released with rs2_release_frame(...)
            rs2_frame* frame = rs2_extract_frame(frames, i, &e);
            check_error(e);

            const uint8_t* frame_data = (const uint8_t*)(rs2_get_frame_data(frame, &e));
            check_error(e);

            unsigned long long frame_number = rs2_get_frame_number(frame, &e);
            check_error(e);

            rs2_time_t frame_timestamp = rs2_get_frame_timestamp(frame, &e);
            check_error(e);

            // Specifies the clock in relation to which the frame timestamp was measured
            rs2_timestamp_domain frame_timestamp_domain = rs2_get_frame_timestamp_domain(frame, &e);
            check_error(e);
            const char* frame_timestamp_domain_str = rs2_timestamp_domain_to_string(frame_timestamp_domain);

            rs2_metadata_type frame_metadata_time_of_arrival = rs2_get_frame_metadata(frame, RS2_FRAME_METADATA_TIME_OF_ARRIVAL, &e);
            check_error(e);

            printf("Frame %d arrived.\n", i);
            printf("First 10 bytes: ");
            int i;
            for(i=0; i < 10; ++i)
                printf("%02x ", frame_data[i]);

            printf("\nFrame No: %llu\n", frame_number);
            printf("Timestamp: %f\n", frame_timestamp);
            printf("Timestamp domain: %s\n", frame_timestamp_domain_str);
            printf("Time of arrival: %lld\n\n", frame_metadata_time_of_arrival);
            rs2_release_frame(frame);
        }

        rs2_release_frame(frames);
    }

    // Stop the pipeline streaming
    rs2_pipeline_stop(pipeline, &e);
    check_error(e);

    // Release resources
    rs2_delete_pipeline_profile(pipeline_profile);
    rs2_delete_config(config);
    rs2_delete_pipeline(pipeline);
    rs2_delete_device(dev);
    rs2_delete_device_list(device_list);
    rs2_delete_context(ctx);

    return EXIT_SUCCESS;
}
`

MartyG-RealSense, 2021-06-15:
My main thought about the script is that there may be an excessive number of **check_error(e);** instructions when a single instance of it in the line before the **pipeline.stop()** instruction may be sufficient.

Likewise, the pipeline.stop() instruction should release device resources used by the pipeline, according to the SDK documentation for the **Stop()** instruction.  So the section of release instructions after the Stop() instruction may be unnecessary.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pipeline.html#a142ee1adb798f9a03f86eb90895dd8a5

MartyG-RealSense, 2021-06-15:
The **rs-color.c** script is a **C** language script, which is not quite the same as a **C++** librealsense script.  So if you are developing in C++ language (as stated in the information at the top of this case) then a C script would not be the best one to use for tests.

**rs-hello-realsense** may be an appropriate C++ example program to use if you want to run a test that provides a realtime-updating text information display.

AndyKohr, 2021-06-16:
Just tested with newest librealsense 2.47.0  and firmware 05.12.14.50. But sadly no progress.
For a test I also tried to get left ir, right ir and depth stream in one frameset. With this setup it happened that the depth stream was hanging.
Maybe we contact the intel support directly to achieve progress or at least find out if it is a hardware or software problem.
Thank you very much for your kind help.

MartyG-RealSense, 2021-06-16:
You are very welcome.  )

If you have performance problems when using multiple streams then you could use **callbacks** in C++.   The SDK's **rs-callback** C++ example program demonstrates doing so using the default stream configuration of a particular camera model.

https://github.com/IntelRealSense/librealsense/tree/master/examples/callback

MartyG-RealSense, 2021-06-22:
Hi @AndyKohr  Do you require further assistance with this case , please?  Thanks!

AndyKohr, 2021-06-22:
Hi @MartyG-RealSense 
I think, that we are stuck with it. I would like to come back to this if we have any further information.
Thank you very much for your help and quick replies!

closed 2021-06-22 by AndyKohr

MartyG-RealSense, 2021-06-22:
You are very welcome @AndyKohr - thanks for the update!