realsenseai/librealsense #9193: L515 Intel Real Sense Viewer Depth Sensing - No stream available when depth module is enabled

tkwane, 2021-06-09:
| | |
|---|---|
|**librealsense**|2.45.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0265839|
|**Firmware Version**|01.05.05.00|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003b530fa67|
|**Firmware Update Id**|0003b530fa67|

The Depth Sensor doesn't seem to be initializing. IntelRealSenseViewer.exe reports that no stream in available from the depth sensing module with the following error:

***** Backend in rs2_set_option(options:000002430BB2B580, option:Visual Preset,
value:5):
hr returned: HResult 0x8007001f: "A device attached to the system is not
functioning." *******

Nonetheless, the RGB sensor seems to be working fine.

Please see attached screenshots. [
[image]
[image]
[image]
[image]

](url)

MartyG-RealSense, 2021-06-27:
Hi @tkwane  Do you still require assistance with the problem described in this case, please?  Thanks!

tkwane, 2021-06-27:
Yes assistance still needed. Thank you.

MartyG-RealSense, 2021-06-27:
Could you try downloading the standalone version of the Viewer called **Intel.RealSense.Viewer.exe** from the 'Assets' file list on the SDK Releases page and see whether the error still occurs on that Viewer version, please? 

This Viewer should be launched with a double-leftclick from the folder that it is downloaded to, and not with the Viewer launch icon on the Windows desktop.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.45.0

[image]

tkwane, 2021-06-27:
Already tried that as a troubleshoot option. Proved futile for all versions of realSenseVewier.exe from 2.3.5 - 2.4.7. 

The issue seems to be an hardware issue because the light on the sensor is not coming on when you toggle the depth module on and off.
Since then, I tried downgrading the camera's firmware and it failed. After fail, it is no longer detectable by windows and invisible to rs-enumerate devices and rs-fw-update.exe. 
Many users experiencing this death state and there seems to be no solution online throughout the length and breath of the community. 
The last attempt I made was setting up the software on Ubuntu. And the issue was same. And the device continued to be invisible to rs-enumerate-devices and rs-fw-update while connected to compatible USB 3 ports in ubuntu. After this proved futile I have since then give up, as further troubleshooting continues to consume too much time without results. 
I think we should create a new thread or topic. L515 death state. Where it is no longer detectable by windows nor the firmware update tool and enumerate tool across different and be sure to make a note this issue is OS independent. This new topic is not to be confused with the tethered state I was able to resolve before in a previous post. The tethered state is when the device is not detectable by windows but still visible to rs-fw-update tool. That would have allow for update the firmware and recovering the device. 

https://github.com/IntelRealSense/librealsense/releases/tag/v2.45.0

MartyG-RealSense, 2021-06-27:
Next, could you test whether the depth and infrared can be displayed in the **rs-capture** example program if you have not done so already, please.

If you have installed the full Windows version of the RealSense SDK, you can find a pre-built executable version of this program by right-clicking on the RealSense Viewer launch icon on the Windows desktop and selecting the **Open file location** menu option to open the SDK's **Tools** folder.  Then locate in this folder the file **rs-capture.exe** and launch it with a double-leftclick whilst the L515 is plugged in.

[image]

[image]

tkwane, 2021-06-27:
The camera is currently is a death state. Not detectable to windows and not visible to the rs-fw-update.exe or rs-enumerate-devices.exe

MartyG-RealSense, 2021-06-27:
I will seek advice from Intel tomorrow about your case and brief them about your situation.  Thanks very much for your patience.

tkwane, 2021-06-27:
Much appreciated. I look forward to further correspondence.

MartyG-RealSense, 2021-06-28:
Hi @tkwane  Before I contact Intel, could you perform a quick test of your L515 for me please.  Unplug the end of the USB cable that is inserted in the side of the camera, flip it around to the opposite side and re-insert it (this end of the cable is two-way insertion).  Is the camera detected after doing so?

tkwane, 2021-06-28:
No that does not work

MartyG-RealSense, 2021-06-28:
Okay, thanks for testing.  I will submit a report to Intel about your case.

tkwane, 2021-06-28:
Ok thank you. This is becoming a prominent issue as many users are experiencing the same thing and have been discussing it on the community but no solutions this far.

MartyG-RealSense, 2021-06-29:
Hi @tkwane  After discussing your case with Intel, they requested that you return your L515 camera to wherever you purchased it from, using the **RMA** (Return Merchandise Authorization) returns process of that supplier.

Also, could you please provide us with the serial number of your camera.  If you cannot access the camera on your computer, the serial number is printed on a small sticker on the side of the camera.

[image]

tkwane, 2021-06-30:
Good day,

Serial number of device: F0265839
The device was purchased on Amazon in January 2021. I immediately started having issues with device out the box. 
I then asked my department at university of the west indies to get in contact with the Amazon suppliers to act quickly and return the device as it should have still been under warranty. 
Due to corona virus pandemic and the university being close after a third wave of lockdown, the department was only able to get back to me in May Stating where all attempts to access warranty proved futile. 
Consequently from our location in the Caribbean we are amputated in being able to take hard action against the Amazon supplier. At this juncture we have lost  contact with the suppliers from whom the device was purchased in January 2021.
Today, the device remains in previously described death state. And I remain hopeful that I can recover the device. As we were never able to use the Cubemos Software purchased before the device, and was the reason for purchasing the device.   

RegardsT. Lowe

MartyG-RealSense, 2021-06-30:
Hi @tkwane  Thanks very much for the serial number.  I will pass the above message to Intel to seek advice about your situation.

MartyG-RealSense, 2021-06-30:
Could you provide an email address that Intel can contact you at please?

tkwane, 2021-06-30:
I can be contacted at ***@***.*** anytime convenient to you. 
Thank you.

MartyG-RealSense, 2021-06-30:
The email address was blocked out when it appeared on the forum.  Please instead send me an email and I will get your address that way.  My address is:

martyx.grover@intel.com

MartyG-RealSense, 2021-07-11:
Hi @tkwane   After Intel contacted you by email, has this issue now been resolved please?  Thanks!

tkwane, 2021-07-12:
Thank you for following up. 
I am currently in direct correspondence with someone at Intel. 
At such time we have reached a resolution. I will be sure to post and update here.

MartyG-RealSense, 2021-07-12:
Thanks very much @tkwane for the update.