realsenseai/librealsense #9194: Delay of ~60ms in point cloud calculation with multiple cameras in querying `RS2_OPTION_DEPTH_UNITS`

avikde, 2021-06-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   |5.12.11.0 | 
| Operating System & Version |   Win 10 x64, Ubuntu 18.04 (same behavior)  | 
| Kernel Version (Linux Only)    |  4.9                                         | 
| Platform                                 | PC, NVIDIA Jetson Xavier AGX |
| SDK Version                            |  2 (master)  | 
| Language                            |  C++    | 
| Segment			|  Robot   | 

### Issue Description

I am using two cameras streaming depth and using librealsense's point cloud conversion. Since the issue may (or may not) be related to the way I am accessing the two cameras, I've also included information about that below.

#### The actual issue

Occasionally, with >1 camera, the point cloud conversion will take ~60ms (usually it takes ~200-1000us). After building librealsense and putting in profiling code, I have narrowed down this "hiccup" to this block in `proc/pointcloud.cpp`
```
        if (!_depth_units)
        {
            auto sensor = ((frame_interface*)depth.get())->get_sensor().get();
            _depth_units = sensor->get_option(RS2_OPTION_DEPTH_UNITS).query();
            found_depth_units = true;
        }
```

Replacing that with `_depth_units = 0.001;` (that's what I'm setting the depth units to) "fixes" the issue. 
* Is there any API to specify to `librealsense::pointcloud` that we know what the depth units are so it doesn't query them?
* I'd also like to understand why it's having this hiccup. On the Jetson with the cameras pointed at what we want to see it seems to happen regularly. On my PC with the cameras just pointed at the ceiling, it seems to happen when I physically move one of the cameras around (?). The behavior is the same whether connected to separate USB3 ports to the CPU or through a hub.

#### How I am sampling the cameras

The way I would like to use the two cameras is to have a vision thread that (at 90hz, the selected sample rate) "pushes" new (processed) point clouds after combining the data from all cameras. The way I'm doing this now is:
```
	for (size_t i=0; i<Ndev_; ++i) {
		bool got_frame = false;
		// First camera: try_wait; other cameras: check for latest frame
		if (i > 0)
			got_frame = pipelines_[i].poll_for_frames(&framesets_[i]);
		// If i>0 cameras poll fails; also try_wait to make sure we get frames from all
		if (i == 0 || !got_frame)
			got_frame = pipelines_[i].try_wait_for_frames(&framesets_[i], 500); // timeout in ms
		
		if (!got_frame) {
			// error (camera unplugged or something)
		}
	}
```

Though this has been working as expected, I assume there may be a better way of doing this, so I included this information.

MartyG-RealSense, 2021-06-09:
Hi @avikde  The default Depth Unit Scale value of 400 Series cameras is **0.001** so this is an appropriate value to use.

Although the depth unit scale value can be retrieved in real-time from a camera using an SDK programming instruction, this is mainly useful if you expect the scale to be changed during the application's use.  The value does not change unless deliberately altered, and so in the majority of cases a fixed hard-coded value can be used for equations that use the depth scale.

For example, multiplying the raw 16-bit depth value of the camera by **0.001** (the default 400 Series depth scale value) can convert the depth into a real-world distance in meters without having to query the camera to retrieve the depth scale.

An easy way to find the default depth unit scale value of a particular RealSense camera model is to launch the **RealSense Viewer** tool and go to the **Controls** category of the Stereo Module options in the Viewer's side-panel.  Locate the **Depth Units** option and read the value that is displayed in the text box.

[image]

In regard to the issue of why the point cloud slows when the camera is moved, there is a detailed discussion of this phenomenon and potential solutions in the link below.

https://github.com/IntelRealSense/librealsense/issues/4958

avikde, 2021-06-09:
Thanks @MartyG-RealSense - since the code that retrieves the depth units is in `librealsense/src/proc/pointcloud.cpp` I was wondering if there's something I can do in my usage of librealsense that can ask it to not query for depth units. For now I can use a "patched" version but I'm assuming this is happening because I'm doing something wrong.

avikde, 2021-06-09:
Thanks @MartyG-RealSense - I did actually build with CUDA and this particular issue is there with and without. I checked on `_depth_units` - it is an "optional_value" type in the SDK which evaluates to false if it doesn't have a value, and it seems to always be going into that if even if it got the value the previous iteration. My instance of `rs2::pointcloud` is a member variable for a class so it is being re-used, and so I'm confused also about why that is.

MartyG-RealSense, 2021-06-09:
It is good practice to avoid putting the pointcloud object in a loop if you are doing that in your project, especially as doing so can cause a memory leak.

https://github.com/IntelRealSense/librealsense/issues/2860#issuecomment-445539797

MartyG-RealSense, 2021-06-15:
Hi @avikde  Do you require further assistance with this case, please?  Thanks!

avikde, 2021-06-15:
For now the workaround of modifying the librealsense sources and hardcoding the depth units in pointcloud.cpp is working for me, so I guess we can close this issue for now. However, I haven't made any progress toward an actual fix, so potentially this will come back in the future and at that point I'll have to open another issue I guess. Thanks for the help.

closed 2021-06-15 by avikde

MartyG-RealSense, 2021-06-15:
I'm pleased to hear that you found a workaround @avikde - thanks very much for the update, and please do feel free to open cases in the future when the need arises.