realsenseai/librealsense #9199: Check if realsense 435 has a custom calibration

AndyKohr, 2021-06-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Language                            |  {C++ }                          | 

### Issue Description
We produce a sensor based on a raspi 4 and a realsense 435 camera. A Manufactorer in Taiwan assembles these systems for us and is responsible to perform a calibration of the camera with the "D400 Series OEM Calibration Target" and its corresponding software from intel.

During production process we want to check if the calibration step was done and not forgotten by accident.

How can this be done? Is this possible with librealsense? Any suggestions?

MartyG-RealSense, 2021-06-10:
Hi @AndyKohr  If you are using the OEM version of the **Dynamic Calibration** tool with the OEM Calibration Target board then you should have the **CustomRW** tool installed on the computer that performs the calibration.  Is that correct, please?

If so, then you could have the manufacturer export the completed calibration as an XML text file using CustomRW.  The file date imprinted on this XML file at the time of its creation could act as proof that the calibration was performed and also the date and time that it was performed.

AndyKohr, 2021-06-10:
TY for the quick response.
Yes, we have the CustomRW tool. But we need to check the calibration at runtime on the sensor itsef.

MartyG-RealSense, 2021-06-10:
You mean perform a check with a librealsense script instead of the Dynamic Calibration tool, please?

AndyKohr, 2021-06-10:
Yes

MartyG-RealSense, 2021-06-10:
The OEM calibration tool has the ability to access a Linux server to perform calibration remotely.  So conceivably you could take control of that part of the manufacturing process yourself, with the manufacturer setting up the cameras to be tested and you conducting the calibration from your own office.  The extract below is from the OEM Target board's listing in the RealSense Store.

[image]

MartyG-RealSense, 2021-06-20:
Hi @AndyKohr  Do you require further assistance with this case, please?  Thanks!

AndyKohr, 2021-06-21:
Thank you for your help. We calibrate the Sensors on a separate Windows PC. After that the calibrated cameras go to stock.
Later, the linux applicatin software is uploaded to the system and that is the point in time where we want to check if the cameras had been calibrated.

MartyG-RealSense, 2021-06-21:
If you are calibrating the cameras after you have purchased them before putting them into stock then that process may not be necessary, as the camera's depth and RGB sensors are calibrated in the factory during the manufacturing process.  This means that RealSense cameras usually have very good calibration when taken out of the box after purchase.

AndyKohr, 2021-06-21:
Hello Marty,
I'm sorry about some misleading description. We are talking about the D430 module, not the D435 camera. That was my mistake. I changed the description and title...
The first half of our device (with the camera module) will be calibrated separately. The calibration process is stand alone. As our last step of production, we want to test our hardware and also if the device is calibrated (software installation and check is done automatically). The easiest thing would be to distinguish between an initial default calibration and our calibration by using the Realsense SDK. As alternative, we have to create some information during the calibration process as an indication, I think...

MartyG-RealSense, 2021-06-21:
If it is a D430 module then that may simplify things a lot, as the D430 does not have an RGB sensor.  Calibrating both the depth and RGB sensors is a key reason for using the Dynamic Calibration software.

But if you only need to calibrate depth then you may be able to use the other calibration tool, **On-Chip Calibration**.  This tool is based around the camera firmware driver instead of an external software package and can be controlled through the RealSense Viewer or through librealsense scripting, including C++.  Intel have published a white-paper document about it.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras#section-appendix-d-on-chip-calibration-c-api

davidcann, 2021-06-22:
We're also using the D430 module and doing the OEM calibration on a linux platform (Jetson, arm64), so we're looking to record the calibration data during assembly, as well. I just installed the librscalibrationtool package after adding the Intel repository. Running the tool appears to give us the calibration values. The only downside is that it's a 150 MB install. Here's an example:

```
double@double-desktop:~$ Intel.Realsense.CustomRW -r -sn 101622071199
CustomRW for Intel RealSense D400, Version: 2.11.0.0

  Device PID: 0AD4
  Device name: Intel RealSense D430
  Serial number: 101622071199
  Firmware version: 05.12.10.00

Calibration parameters from the device:
  resolutionLeftRight: 1280 800

  FocalLengthLeft: 632.364014 632.090942
  PrincipalPointLeft: 640.617126 402.104523
  DistortionLeft: -0.054933 0.062062 0.000376 0.001728 -0.018811

  FocalLengthRight: 628.952820 628.547241
  PrincipalPointRight: 646.969299 394.721771
  DistortionRight: -0.054970 0.063625 0.000407 -0.000496 -0.019268

  RotationLeftRight: 0.999952 -0.000702 -0.009804
                     0.000672 0.999995 -0.003028
                     0.009806 0.003021 0.999947
  TranslationLeftRight: -50.048241 -0.085826 0.068458

  HasRGB: 0
```

@MartyG-RealSense, are these calibration values from our OEM calibration?

@AndyKohr, is this what you're looking for?

AndyKohr, 2021-06-22:
@davidcann Thank's for this. We already checked that but we are interested in a calibration date or something like that, that can be read out of the device afterwards. It seems to be, that this is not possible. So, I think our only option is then to change our process because that part is stand alone at the moment. That means, only the camera part of our device is connected to a Windows machine and we are not recording calibration values. Also, there is no other software running at that time, excepted the OEM Calibration Tool.
The ticket can be closed then, I think. Thank you both for your time!

MartyG-RealSense, 2021-06-22:
@davidcann   This calibration information can be accessed through the RealSense Viewer by going to the **More** option near the top of the options side-panel and selecting the **Calibration Data** option from its menu.

[image]

[image]

From the Calibration Data interface in the Viewer, you can also save the calibration information to a **json** text file and open it to read the calibration information.

[image]

So if you require the calibration data but would prefer not to install librscalibrationtool  if you are only going to read the data then using the Viewer's calibration data interface and exporting the data as a json text file may be a viable alternative (since the Viewer tool is part of the librealsense installation and so would not be adding 150MB to the installation file-size footprint).

In regard to whether your values are from your OEM calibration: if you wrote the calibration data to the camera hardware using the OEM tool and have not written a calibration by another method since then yes, it would be reading the most recent calibration written to the internal **calibration table** of the camera.

***************

@AndyKohr  I cannot find a way to extract a date of last update from the calibration table information or whether a calibration has not yet been performed.

The Chief Technical Officer of the RealSense Group at Intel (agrunnet) provides some interesting information about On-Chip calibration in the link below though.

https://github.com/IntelRealSense/librealsense/issues/5838#issue-564388271

agrunnet states that once a calibration has been performed then new versus old calibration can be compared.  You could infer from this that if a calibration has not yet been stored then this comparison option would not be available, offering a means of determining if a camera unit has been calibrated yet.  I cannot test this myself though as I do not have a new 'fresh out of the box' Depth Module available.

MartyG-RealSense, 2021-07-04:
Hi @AndyKohr  Do you require further assistance with this case, please?

AndyKohr, 2021-07-07:
Hello @MartyG-RealSense Thank you for your help. We will change our processes during production. So far, the mentioned possibilities/ideas are not very helpful for us.

closed 2021-07-07 by AndyKohr

MartyG-RealSense, 2021-07-07:
Okay, thanks very much for the update @AndyKohr - good luck with your production design!