realsenseai/librealsense #9200: Simultaneous streaming of LiDAR and D435i in one code

AliBarzing, 2021-06-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 **&** D435i } | 
| Operating System & Version |   {Win (10)   | 
| Language                            |  {Python} |

### Issue Description
Hi,

I have been working with L515 for a while. Now I have a code and I want to test my algorithm on L515 and D435i at the same time. Therefore, I am trying to stream my LiDAR and depth cameras simultaneously. The code below is what I was using for my LiDAR:

`config = rs.config()`
`config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)`
`config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)`
`pipeline = rs.pipeline()`
`pipeline.start()`

Is it possible to start streaming both cameras in the same code while one is connected to my laptop's USBC and the other one to the USB3 port?

Thanks in advance!

RealSenseSupport, 2021-06-15:
@AliBarzing Please note that depth 1280x720 is not supported for L515. For the example of multiple cameras, please refer to https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam 

It should work to connect one to USBC and the other to USB3 port.

AliBarzing, 2021-06-15:
> @AliBarzing Please note that depth 1280x720 is not supported for L515. For the example of multiple cameras, please refer to https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam
> 
> It should work to connect one to USBC and the other to USB3 port.

Thanks for your reply. Yeah, I tried using USBC and USB3 but when I started recording both cameras using intel realsense viewer it became so laggy. Everything works fine until I start recording.  Do you have any recommendations or comments on this?

RealSenseSupport, 2021-06-18:
@AliBarzing Is there any issue if you only record one of the cameras?

RealSenseSupport, 2021-07-21:
@AliBarzing Do you still have issue? Looking forward to your update. Thanks!

AliBarzing, 2021-07-21:
Hi,

Thank you so much for asking again! Well, I solved the problem (whoops, I forgot to close the issue, my bad). I connected Lidar using my USB3 port and Depth camera using a USB-C. I could not get the highest possible quality with USB-C so I gave up on that. Then I started recording both cameras using Realsense viewer and then fed my code with the stored data. As my code had only one GUI working for a single camera I streamed just one of them each time.

closed 2021-07-21 by AliBarzing